tzapu/WiFiManager #1096: wm.startConfigPortal and wm.resetSettings cause crash

alfo, 2020-07-13:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

I'm trying to add the reset button functionality from the Advanced example sketch, but either a long press or a short press causes a crash. I've omitted some code because it's an enormous sketch, but can you see anything blatantly wrong below? I can provide the whole thing if it's needed.

### Settings in IDE

Module: D1 Mini 32

Additional libraries: 

* PubSubClient
* PxMatrix
* HTTPClient
* ArduinoJson
* SPIFFS

### Sketch

```cpp

#include <Arduino.h>

const int resetPin = 26;

void setup() {
  // WiFiManager bits and pieces
  WiFiManager wm;
  wm.setSaveConfigCallback(saveConfigCallback);
  wm.setAPCallback(configModeCallback);
  wm.setAPClientCheck(true);
  wm.setConnectTimeout(CONNECT_TIMEOUT);
  wm.setTimeout(AP_TIMEOUT);
  wm.setCountry("GB");
  wm.setClass("invert"); // Dark theme

  // Set hostname from settings
  wm.setHostname(hostname);

  // WiFiManager param stuff omitted here

  //reset settings - wipe credentials for testing, if defined
  #if defined(START_ANEW)
    Serial.println("^^^^^^^^ Clearing WiFi credentials");
    wm.resetSettings();
  #endif

  // Run the routine to connect to the network
  if (!wm.autoConnect(hostname, "glowingsquare")) {

    // If we've hit the config portal timeout, then retstart

    Serial.println("%%% Failed to connect and hit timeout, restarting");
    delay(100);
    ESP.restart();

    // Not sure if this line is necessary
    delay(5000);
  }

  // Keeping this line cos it's cute
  Serial.println("Connected ...yeey :)");

}

void checkButton(){

  WiFiManager wm;

  // check for button press
  if (digitalRead(resetPin) == LOW) {
    // poor mans debounce/press-hold, code not ideal for production
    delay(50);
    if( digitalRead(resetPin) == LOW ){
      Serial.println("Button Pressed");
      // still holding button for 3000 ms, reset settings, code not ideaa for production
      delay(3000); // reset delay hold
      if( digitalRead(resetPin) == LOW ){
        Serial.println("Button Held");
        Serial.println("Erasing Config, restarting");
        wm.resetSettings();
        ESP.restart();
      }

      // start portal w delay
      Serial.println("Starting config portal");
      wm.setConfigPortalTimeout(120);

      if (!wm.startConfigPortal("OnDemandAP","password")) {
        Serial.println("failed to connect or hit timeout");
        delay(3000);
        // ESP.restart();
      } else {
        //if you get here you have connected to the WiFi
        Serial.println("connected...yeey :)");
      }
    }
  }
}

void loop() {

  checkButton();

}
```

### Debug Messages

```
Button Pressed
Starting config portal
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OnDemandAP
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088bb7 on core 1

Backtrace: 0x4008c8d4:0x3ffbe690 0x4008cb05:0x3ffbe6b0 0x40088bb7:0x3ffbe6d0 0x400ee33a:0x3ffbe710 0x400ed649:0x3ffbe730 0x400d2ee8:0x3ffbe750 0x40080f61:0x3ffbe790 0x400815b1:0x3ffbe7b0 0x40084a85:0x3ffbe7d0 0x40086d3d:0x3ffba190 0x40082e7f:0x3ffba1b0 0x40088e91:0x3ffba1d0

Rebooting...

```

IDkonnecT, 2020-10-03:
Same issue ! Still no solution ?

tablatronix, 2020-10-03:
If you are getting issue like this it almost always an esp bug, or you are doing something wrong, or are trying to do stuff in interrupts, or using unsafe memory functions. Try to update your esplib