tzapu/WiFiManager #1098: wifiManager.addParameter not working or 6 parameter

Behzad-6-1, 2020-07-17:
Hi I am new on this and tried to use wifiManager.addParameter
it seems if I add the 4th parameter it breaks the code and it goes in to endless loop and print table of hex code

I can see why it can't work
SPIFFSEditor.h only set for 5 pre defined parameters
`#ifdef ESP32
    SPIFFSEditor(const fs::FS& fs, const String& username=String(), const String& password=String());
#else
    SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
#endif`

Can this be changed to add another one for mqtt_name
I am having issue adding mqtt_name in config page

tablatronix, 2020-07-17:
Without a logs a sketch or crashdumpnit could be anything.

The alpha version does not have a limit of params it allocates automatically So try that.

You are most likely overflowing memory or have bad code

Behzad-6-1, 2020-07-17:
this is copy of serial dump it is going around a circuit
-------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4021b398 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc20 end: 3fffffc0 offset: 0190
3ffffdb0:  3fff1184 3fff11e0 3ffeef50 402013fe  
3ffffdc0:  3fff1184 3fff11e0 3ffeef50 40202972  
3ffffdd0:  402165a8 00000000 000003e8 4020a818  
3ffffde0:  3fff1124 3fff116c 00000000 00000000  
3ffffdf0:  3ffef0fc feefeffe feefeffe feefeffe  
3ffffe00:  feefeffe feefeffe feefeffe feefeffe  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
3ffffe60:  feefeffe feefeffe feefeffe feefeffe  
3ffffe70:  feefeffe feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
3ffffeb0:  feefeffe feefeffe feefeffe 3ffeef50  
3ffffec0:  00000046 3fff11bd feefef0a feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  402163c8 feefeffe 3fff11d4 00000200  
3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
3fffff00:  feefeffe feefeffe feefeffe feefeffe  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef080  
3fffffa0:  3fffdad0 00000000 3ffef040 4020c404  
3fffffb0:  feefeffe feefeffe 3ffe8594 40100c85  
<<<stack<<<

Behzad-6-1, 2020-07-17:
could you please let me know where I can get the alpha version?

tablatronix, 2020-07-17:
Its in the arduino ide, or development branch on github

Also have you tested using one of the examples ?

Behzad-6-1, 2020-07-17:
I am trying to build the sketch again from scratch step by step. 
I can't seemed to find the build alpha