tzapu/WiFiManager #1099: Is non-blocking autoconnect with timeout on AP possible?

Hoel, 2020-07-24:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

I would like to keep the AP active for 1 hour regardless if the STA connection succeed or failed, and non blocking because the fimware has other things to do and cannot be locked for one hour until the timeout is finished.
I know the STA + AP mode is not ideal but it works well enough in our case.
IS there a way to do this with wifimanager?

tablatronix, 2020-07-24:
Yes you will have to probably setup your own timer and stopconfingportal. 

There should be an example of ondemand that runs for a bit, combine that with non blocking example 

Of course this only exists in the dev/ alpha version

sebosfato, 2020-10-15:
Hello Tablatronix, my name is fabio and im trying to develop a program with your wifimanager piece of software. I was wondering if it would be possible to simply run into access point mode after some time without choosing a network to connect, allowing to run the program and access my webserver directly connecting to the ap. i explain you why, some people would not have internet to connect to and so it should run the program and allow for user to access.. 

would it be possible? 

im using a timeout already to let the thing to try to reconect after power failure. however i would like that if no network is available or maybe even add a button to start ap mode and set it.. .

this is what i use 
#include <WiFiManager.h>
WiFiManager wifiManager;

on setup
wifiManager.setTimeout(120);
  wifiManager.setBreakAfterConfig(true);
  if (!wifiManager.autoConnect(host)) {
    delay(1000);
    ESP.reset();
 
//   
  }

thanks so much for the excellent work!

tablatronix, 2020-10-16:
Yes you can startConfigPortal or startWebPortal independantly in the non blocking version, see the DEV/ example it has examples for most use cases