tzapu/WiFiManager #1100: ESP losing connection every 10 days

jeffersonpimenta, 2020-07-26:
Hi, iam using WifiManager with one sonoff mini and I notice that ESP loses connection every 10 days or so. Iam using the code to simulate a Modbus protocol in ScadaLTS. My disconnections occur:

[image]

[image]

[image]

Iam using pihole with 24h lease time. My ESP don't change ip, even when it lost connection and i put my wifi pass again.

[image]

[image]

Any ideas? My code is:

> #include <DNSServer.h>
> #include <ESP8266WebServer.h>
> #include <WiFiManager.h>
> 
> 
> int ModbusTCP_port = 502;
> 
> int SONOFF_LED = 13; //led built in
> int SONOFF_RELAY = 12;//rele
> int SONOFF_INPUT = 4;//entrada
> int SONOFF_BOTAO = 0;//botão
> 
> //////// Required for Modbus TCP / IP /// Requerido para Modbus TCP/IP /////////
> #define maxInputRegister 20
> #define maxHoldingRegister 20
> 
> #define MB_FC_NONE 0
> #define MB_FC_READ_REGISTERS 3 //implemented
> #define MB_FC_WRITE_REGISTER 6 //implemented
> #define MB_FC_WRITE_MULTIPLE_REGISTERS 16 //implemented
> //
> // MODBUS Error Codes
> //
> #define MB_EC_NONE 0
> #define MB_EC_ILLEGAL_FUNCTION 1
> #define MB_EC_ILLEGAL_DATA_ADDRESS 2
> #define MB_EC_ILLEGAL_DATA_VALUE 3
> #define MB_EC_SLAVE_DEVICE_FAILURE 4
> //
> // MODBUS MBAP offsets
> //
> #define MB_TCP_TID 0
> #define MB_TCP_PID 2
> #define MB_TCP_LEN 4
> #define MB_TCP_UID 6
> #define MB_TCP_FUNC 7
> #define MB_TCP_REGISTER_START 8
> #define MB_TCP_REGISTER_NUMBER 10
> 
> byte ByteArray[260];
> unsigned int MBHoldingRegister[maxHoldingRegister];
> 
> //////////////////////////////////////////////////////////////////////////
> 
> WiFiServer MBServer(ModbusTCP_port);
> 
> void setup() {
> 
>   pinMode(SONOFF_LED, OUTPUT);
>   pinMode(SONOFF_RELAY, OUTPUT);
>   pinMode(SONOFF_INPUT, INPUT);
> 
>   Serial.begin(115200);
> 
>   WiFiManager wifiManager;
> 
>   for (int inicializ = 0; inicializ <= 25; inicializ++) {
>     digitalWrite(SONOFF_LED, HIGH);
>     delay(100);
>     digitalWrite(SONOFF_LED, LOW);
>     delay(100);
> 
>     if (!digitalRead(SONOFF_BOTAO)) {
>       digitalWrite(SONOFF_LED, HIGH);
>       delay(500);
>       digitalWrite(SONOFF_LED, LOW);
>       delay(500);
>       digitalWrite(SONOFF_LED, HIGH);
>       delay(500);
>       digitalWrite(SONOFF_LED, LOW);
>       delay(500);
>       digitalWrite(SONOFF_LED, HIGH);
>       delay(500);
>       digitalWrite(SONOFF_LED, LOW);
>       delay(500);
> 
>       if (digitalRead(SONOFF_BOTAO)) {
>         wifiManager.resetSettings();
>         break;
>       }
>     }
>   }
> 
> 
> 
>   wifiManager.autoConnect();
>   Serial.println("connected...yeey :)");
>   digitalWrite(SONOFF_LED, HIGH);
>   delay(2000);
>   digitalWrite(SONOFF_LED, LOW);
>   delay(2000);
>   digitalWrite(SONOFF_LED, HIGH);
> 
> 
>   MBServer.begin();
>   Serial.println("Connected ");
>   Serial.print("ESP8266 Slave Modbus TCP/IP ");
>   Serial.print(WiFi.localIP());
>   Serial.print(":");
>   Serial.println(String(ModbusTCP_port));
>   Serial.println("Modbus TCP/IP Online");
> 
> }
> 
> void loop() {
>   // put your main code here, to run repeatedly:
> 
> 
>   // Check if a client has connected // Modbus TCP/IP
>   WiFiClient client = MBServer.available();
>   if (!client) {
>     return;
>   }
> 
>   boolean flagClientConnected = 0;
>   byte byteFN = MB_FC_NONE;
>   int Start;
>   int WordDataLength;
>   int ByteDataLength;
>   int MessageLength;
> 
>   // Modbus TCP/IP
>   while (client.connected()) {
> 
>     if (client.available())
>     {
>       flagClientConnected = 1;
>       int i = 0;
>       while (client.available())
>       {
>         ByteArray[i] = client.read();
>         i++;
>       }
> 
>       client.flush();
> 
>       ///////// Holding Register [0] A [9] = 10 Holding Registers Escritura
>       ///////// Holding Register [0] A [9] = 10 Holding Registers Writing
> 
> 
>       MBHoldingRegister[0] = !digitalRead(SONOFF_INPUT);
>       // MBHoldingRegister[1] = 2;
>       // MBHoldingRegister[2] = 3;
>       // MBHoldingRegister[3] = 4;
>       // MBHoldingRegister[4] = 5;
>       // MBHoldingRegister[5] = 6;
>       // MBHoldingRegister[6] = 7;
>       // MBHoldingRegister[7] = 8;
>       // MBHoldingRegister[8] = 9;
>       // MBHoldingRegister[9] = 10;
> 
> 
> 
> 
>       ///////// Holding Register [10] A [19] = 10 Holding Registers Lectura
>       ///// Holding Register [10] A [19] = 10 Holding Registers Reading
> 
>       //int Temporal[10];
> 
>       // Temporal[0] = MBHoldingRegister[10];
>       // Temporal[1] = MBHoldingRegister[11];
>       // Temporal[2] = MBHoldingRegister[12];
>       // Temporal[3] = MBHoldingRegister[13];
>       // Temporal[4] = MBHoldingRegister[14];
>       // Temporal[5] = MBHoldingRegister[15];
>       // Temporal[6] = MBHoldingRegister[16];
>       // Temporal[7] = MBHoldingRegister[17];
>       // Temporal[8] = MBHoldingRegister[18];
>       // Temporal[9] = MBHoldingRegister[19];
> 
>       /// Enable Output 14
>       // digitalWrite(14, MBHoldingRegister[14] );
> 
>       digitalWrite(SONOFF_RELAY, MBHoldingRegister[10]);
>       //// debug
> 
>       // for (int i = 0; i < 10; i++) {
> 
>       // Serial.print("[");
>       // Serial.print(i);
>       // Serial.print("] ");
>       // Serial.print(Temporal[i]);
> 
>       // }
>       // Serial.println("");
> 
> 
>       //// rutine Modbus TCP
>       byteFN = ByteArray[MB_TCP_FUNC];
>       Start = word(ByteArray[MB_TCP_REGISTER_START], ByteArray[MB_TCP_REGISTER_START + 1]);
>       WordDataLength = word(ByteArray[MB_TCP_REGISTER_NUMBER], ByteArray[MB_TCP_REGISTER_NUMBER + 1]);
>     }
> 
>     // Handle request
> 
>     switch (byteFN) {
>       case MB_FC_NONE:
>         break;
> 
>       case MB_FC_READ_REGISTERS: // 03 Read Holding Registers
>         ByteDataLength = WordDataLength * 2;
>         ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
>         ByteArray[8] = ByteDataLength; //Number of bytes after this one (or number of bytes of data).
>         for (int i = 0; i < WordDataLength; i++)
>         {
>           ByteArray[ 9 + i * 2] = highByte(MBHoldingRegister[Start + i]);
>           ByteArray[10 + i * 2] = lowByte(MBHoldingRegister[Start + i]);
>         }
>         MessageLength = ByteDataLength + 9;
>         client.write((const uint8_t *)ByteArray, MessageLength);
> 
>         byteFN = MB_FC_NONE;
> 
>         break;
> 
> 
>       case MB_FC_WRITE_REGISTER: // 06 Write Holding Register
>         MBHoldingRegister[Start] = word(ByteArray[MB_TCP_REGISTER_NUMBER], ByteArray[MB_TCP_REGISTER_NUMBER + 1]);
>         ByteArray[5] = 6; //Number of bytes after this one.
>         MessageLength = 12;
>         client.write((const uint8_t *)ByteArray, MessageLength);
>         byteFN = MB_FC_NONE;
>         break;
> 
>       case MB_FC_WRITE_MULTIPLE_REGISTERS: //16 Write Holding Registers
>         ByteDataLength = WordDataLength * 2;
>         ByteArray[5] = ByteDataLength + 3; //Number of bytes after this one.
>         for (int i = 0; i < WordDataLength; i++)
>         {
>           MBHoldingRegister[Start + i] = word(ByteArray[ 13 + i * 2], ByteArray[14 + i * 2]);
>         }
>         MessageLength = 12;
>         client.write((const uint8_t *)ByteArray, MessageLength);
>         byteFN = MB_FC_NONE;
> 
>         break;
>     }
>   }
> 
> 
> }

dontsovcmc, 2020-08-14:
Can router be a reason of losing connection? There is an IP lifetime in every routers.

jeffersonpimenta, 2020-08-15:
I use pihole to provinde IPs, since first setup every machine got same ip and it didnt change...

uski, 2020-10-04:
What are you doing to regain connection ?
I would suggest logging the serial output of the ESP to see what happens. Without further information it is very difficult to help you

jeffersonpimenta, 2020-10-05:
To reconnect i need to log in web interface and reconfigure everything. Since I'm using an Sonoff standalone i haven't access to the terminal messages.

EgHubs, 2020-11-27:
[This ](https://internetofhomethings.com/homethings/?p=396#comment-28503) might help