tzapu/WiFiManager #1101: setConfigPortalTimeout but autoconnect still exits back to program even though not connected to any wifi

davidkjackson54, 2020-07-27:
I have setConfigPortalTimeout(180) set as follows:

wifiManager.setAPStaticIPConfig(IPAddress(172, 217, 28, 1), IPAddress(172, 217, 28, 1), IPAddress(255,255,255,0));
wifiManager.setConfigPortalTimeout(180);            // give the configuration portal 3 minutes to complete 
wifiManager.autoConnect();
Serial.println("Connected...");

I am testing with a wifi hotspot to see what happens when wifi goes offline.
WiFiManager is correctly waiting for 180 seconds, then goes into AP mode
I see the AP SSID appear on my phone but I do not want to  reconfigure - so I leave it and wait.
Then WifiManager exits from wifiManager.autoConnect and continues - without having connected to any wifi. 

How do I fix this? I don't want it to continue if it hasn't connected to any wifi.   This scenario is just to test what happens if the wifi is unavailable for any reason. so under normal circumstances, I would not wish to configure a new SSID etc.  

*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM:
*WM: Configuring access point...
*WM: ESP13649403
*WM: Custom AP IP/GW/Subnet
*WM: AP IP address: 
*WM: 172.217.28.1
*WM: HTTP server started
Connected...                                           <This Serial.print is from my progeam after returning from wifiManager.autoConnect
*WM: freeing allocated params!
WiFi connected - ESP IP address: (IP unset)

 

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

davidkjackson54, 2020-07-27:
I have resolved this.

closed 2020-07-27 by davidkjackson54

tablatronix, 2020-07-27:
Thats is exactly what that setting is for