tzapu/WiFiManager #1102: Q: Where in flash are SSID/key stored? Is this something I need to keep in mind?

noelvissers, 2020-07-31:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

### Description

In the project, I want to store some settings in flash. From what I understand the SSID and password (and maybe other settings) from this library are also stored in the flash memory. If i write some values to a specific address, are there addresses i cannot use?

example:

```c++
//EEPROM.write(address, value);
EEPROM.write(1, 0);
EEPROM.commit();
```
Will this interfere with the other stored values?

noelvissers, 2020-07-31:
From what I understand from further reading this is stored in a specific Wifi part on the ESP32. So this means the flash I use to write my own settings isn't related to this.

closed 2020-07-31 by noelvissers

tablatronix, 2020-07-31:
No esp does it