tzapu/WiFiManager #1103: calling setDebugOutput(false) causes autoconnect to fail and go straight to config (DEV BRANCH 2.0.3)

tavdog, 2020-08-02:
# Basic Infos
Using Adafruit featheresp32 target with platformio.  Problem is not present on Adafruit huzzahesp8266

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [*] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [*] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

With wifi already configured,  if device is flashed with code that executes wm.setDebugOutput(false) autoconnect fails immiediatly (does not wait for connect timeout) and goes straight to config portal.  The device connects file if calling wm.setDebugOutput(true) or line ommitted.

### Settings in IDE

Module:  			featheresp32

Additional libraries: 

### Sketch

```cpp
WiFiManagerParameter custom_brightness("brightness", "Brightness : High/Med/Low/Auto", config->brightness.c_str(), 5);
    WiFiManagerParameter custom_user("api_user", "Username : user@example.com", config->api_username.c_str(), 25);
    WiFiManagerParameter custom_password("api_password", "Password", config->api_password.c_str(), 32);

    //add all your parameters here
    wm.addParameter(&custom_brightness);
    wm.addParameter(&custom_user);
    wm.addParameter(&custom_password);

    // set the AP callback to display AP on display
    wm.setAPCallback(display_AP);

    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    //here  "AutoConnectAP" if empty will auto generate basedcon chipid, if password is blank it will be anonymous
    //and goes into a blocking loop awaiting configuration
    wm.setConfigPortalTimeout(60);
    wm.setConnectTimeout(15);
    //wm.setDebugOutput(false);
    if (force_config) {
      wm.startConfigPortal(ap_name);
    } else {
      wm.autoConnect(ap_name);
    }

    Serial.println("connected");

```

### Debug Messages

```
FAILURE OUTPUT :
MAC:2462ABDADC38
entering wifi_setup
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: brightness
*WM: [2] Added Parameter: api_user
*WM: [2] Added Parameter: api_password

SUCCESS OUTPUT:
entering wifi_setup
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: brightness
*WM: [2] Added Parameter: api_user
*WM: [2] Added Parameter: api_password
*WM: [1] Free heap:        309244
*WM: [1] ESP SDK version:  v3.2.3-14-gd3e562907
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: windytron
*WM: [3] Using Password: aaaaaaaa
*WM: [3] WiFi station enable 
*WM: [2] 15000 ms connectTimeout set
*WM: [2] 15000 ms timeout, waiting for connect...
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.43.31
```

tablatronix, 2020-08-02:
I dont understand this says

```
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.43.31
```

Which is not cp ...

tavdog, 2020-08-02:
That is output when debugLevel is at default or set to true.  The failure case is the first pasted block 
FAILURE OUTPUT :
```
MAC:2462ABDADC38
entering wifi_setup
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: brightness
*WM: [2] Added Parameter: api_user
*WM: [2] Added Parameter: api_password
```
Here output stops and it goes straight into configportal without connectiong.

tablatronix, 2020-08-02:
Then why is there no logging? Are you sure that is what is happening?

tablatronix, 2020-08-02:
Hmm ill test it here and see if I can reproduce.

tavdog, 2020-08-03:
The logging stops because it is at that point that setDebugOutput(false) is called.  My hunch is that it might just be a weird issue with esp32 and timing.  Since it takes more cycles to print out the debugging info it gives the esp32 more time to connect in the background maybe ?  I'll test by adding a delay and see if that helps.  
Here is more of the preceding code that shows the parameters added then the debug call.  
```
//add all your parameters here
    wm.addParameter(&custom_brightness);
    wm.addParameter(&custom_user);
    wm.addParameter(&custom_password);

    // set the AP callback to display AP on display
    wm.setAPCallback(display_AP);

    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    //here  "AutoConnectAP" if empty will auto generate basedcon chipid, if password is blank it will be anonymous
    //and goes into a blocking loop awaiting configuration
    wm.setConfigPortalTimeout(60);
    wm.setConnectTimeout(15);
    wm.setDebugOutput(false);
    if (force_config) {
      wm.startConfigPortal(ap_name);
    } else {
      wm.autoConnect(ap_name);
    }

    Serial.println("connected");
```

tavdog, 2020-08-03:
Found it !  removing the line : wm.setConnectTimeout(15); fixed it.  Weird.  Is the 15 in ms and not seconds maybe ?

closed 2020-08-03 by tavdog

tablatronix, 2020-08-03:
There is an ongoing ESP32 bug that several people with certain routers are having where connections take a very long time, if you remove the timeout it will use the default esp32 connect wait loop.

I think may be you experiencing this issue , where it always fails on first connection.

I find it odd that you thought debugoutput had something to do with it.. that is a bit odd if it was true.

#1067