tzapu/WiFiManager #1104: Q: Is it possible to have a selectable dropdown menu in config portal?

noelvissers, 2020-08-02:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

### Description

Since it is possible to have custom parameters in the config portal, I was wondering if there's a way to have them set as dropdown menu, so i can't get an invalid input. Example: I want to have the user select a true or false setting and not let them fill in some random word.

If this is possible, is there any example or documentation of it? Would be nice to include it in the custom parameters example since it uses 3x the same method now.

noelvissers, 2020-08-02:
Since there are three examples, it would be nice to have 1 normal one, 1 checkbox and 1 dropdown menu. I know custom html is an option in these parameters, but I'm a total noob with html, especially integrating it with c++ (and I think im not the only one).

saul-hdez, 2021-06-13:
Did you solve it? I need this exact thi g and so far u can create the drop down with custom html but hav not found a way to access the values

tablatronix, 2021-06-13:
#627 
It is possible, but you would have to handle it mostly outside of wm, I was looking into how to add the ability for wm to wrap the arguments for you via params, but nothing so far has been done, i have a test somewhere I will look at where i left off.

FLYINGSC0T, 2022-05-28:
I was also looking for a solution to add dropdown to WiFiManager. 
You might find these two links useful:

https://github.com/witnessmenow/ESP32-WiFi-Manager-Examples
https://www.youtube.com/watch?v=Errh7LEEug0

As yet I haven't tried it but hope it might help.

tablatronix, 2022-05-28:
#1365

closed 2022-05-28 by tablatronix