tzapu/WiFiManager #1111: Value for Checkboxes overwritten - "Advanced" example

Lithimlin, 2020-08-20:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other: TTGO T8

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

I've tried using the checkboxes (as in the "Advanced" example) to enable and disable things on my ESP. While doing that I figured out that once you *disable* a checkbox and exit the configuration or let the AP timeout, the value saved in the `WiFiManagerParameter` is lost and the checkbox will not work again.

#### How to reproduce

1. Open the "Advanced" example 
2. Enable the checkbox (line 30) and commented the radioboxes (lines 33f)
3. Upload the sketch
4. Connect to the AP
5. In the `Params`, click `save`*without* enabling the checkbox
6. Close the AP or let it timeout
7. Open the AP again by pressing the button
8. Try enabling the checkbox and saving

The value of the checkbox will remain empty even if it is checked

### Sketch
"Advanced" example:

```cpp

/**
 * WiFiManager advanced demo, contains advanced configurartion options
 * Implements TRIGGEN_PIN button press, press for ondemand configportal, hold for 3 seconds for reset settings.
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

#define TRIGGER_PIN 0

WiFiManager wm; // global wm instance
WiFiManagerParameter custom_field; // global param ( for non blocking w params )

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  Serial.begin(115200);
  Serial.setDebugOutput(true);  
  delay(3000);
  Serial.println("\n Starting");

  pinMode(TRIGGER_PIN, INPUT);
  
  // wm.resetSettings(); // wipe settings

  // add a custom input field
  int customFieldLength = 40;

  // new (&custom_field) WiFiManagerParameter("customfieldid", "Custom Field Label", "Custom Field Value", customFieldLength,"placeholder=\"Custom Field Placeholder\"");
  
[... 246 lines cut ...]
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] customfieldid: 
*WM: [2] -------------------- 
[CALLBACK] saveParamCallback fired
PARAM customfieldid = 
*WM: [3] Sent param save page 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] customfieldid: 
*WM: [2] -------------------- 
[CALLBACK] saveParamCallback fired
PARAM customfieldid = 
*WM: [3] Sent param save page 
```

Lithimlin, 2020-08-24:
Any idea what could be causing this? I'm trying to understand the library atm to see if I find the issue and am possibly able to fix it.

Lithimlin, 2020-08-24:
So the issue seems to lie with this method:
```cpp
void WiFiManager::doParamSave(){
   // @todo use new callback for before paramsaves, is this really needed?
  if ( _presavecallback != NULL) {
    _presavecallback();
  }

  //parameters
  if(_paramsCount > 0){
    DEBUG_WM(DEBUG_VERBOSE,F("Parameters"));
    DEBUG_WM(DEBUG_VERBOSE,FPSTR(D_HR));

    for (int i = 0; i < _paramsCount; i++) {
      if (_params[i] == NULL || _params[i]->_length == 0) {
        DEBUG_WM(DEBUG_ERROR,"[ERROR] WiFiManagerParameter is out of scope");
        break; // @todo might not be needed anymore
      }
      //read parameter from server
      String name = (String)FPSTR(S_parampre)+(String)i;
      String value;
      if(server->hasArg(name)) {
        value = server->arg(name);
      } else {
        value = server->arg(_params[i]->getID());
      }

      //store it in params array
      value.toCharArray(_params[i]->_value, _params[i]->_length+1); // length+1 null terminated
      DEBUG_WM(DEBUG_VERBOSE,(String)_params[i]->getID() + ":",value);
    }
    DEBUG_WM(DEBUG_VERBOSE,FPSTR(D_HR));
  }

   if ( _saveparamscallback != NULL) {
    _saveparamscallback();
  }

}
```
and the `saveParamCallback` in the example. I added two lines to reset the value afterwards in the example:
```cpp
void saveParamCallback(){
  Serial.println("[CALLBACK] saveParamCallback fired");
  Serial.println("PARAM customfieldid = " + getParam("customfieldid"));
  String value = "Custom Field Value";
  custom_field.setValue(value.c_str(), value.length());
}
```

tablatronix, 2021-10-07:
It is an issue with ids? There was a similar issue reported

iPhilBln, 2023-07-14:
I know this thread is old but i run into the same issue. I have some checkboxes with const hidden ip address but if the user don't enable the check box the value of this uncheck checkbox will be overwritten with an empty string. 
So I don't thing this issue is fixed. 

>// const hidden ip addresses 
>130.149.7.7
> 141.76.32.160
> 194.25.134.196
> 
> // before the data are saved from webserver
> [CALLBACK] savePreParamCallback fired
> 0 has no id
> 
> 1 was nullptr
> val: 130.149.7.7
> len: 11
> 
> 2 was nullptr
> val: 141.76.32.160
> len: 13
> 
> 3 was checked
> PARAM[t_online_ntp_server]: 194.25.134.196
> 
> 4 has no id
> 
> 5 was nullptr
> val: 
> len: 15
> 
>
>
> // after the data are saved from webserver
> [CALLBACK] saveParamCallback fired
> 0 has no id
> 
> 1 was nullptr
> val: 
> len: 11
> 
> 2 was nullptr
> val: 
> len: 13
> 
> 3 was checked
> PARAM[t_online_ntp_server]: 194.25.134.196
> 
> 4 has no id
> 
> 5 was nullptr
> val: 
> len: 15

main.cpp
```
    const char* title_str = "<br/><label style='font-weight: bold; font-size: 25px;'>NTP Server Auswahl</label><br/>";
    const char* lineBreak_str = "<br/><br />";
    const char* tu_berlin_ip    = "130.149.7.7";
    const char* tu_dresden_ip   = "141.76.32.160";
    const char* t_online_ip     = "194.25.134.196";
    const char* custom_0_ip     = "000.000.000.000";

    new (&title) WiFiManagerParameter (title_str);
    new (&tu_berlin_ntp_server) WiFiManagerParameter("tu_berlin_ntp_server", "TU Berlin", tu_berlin_ip, strlen(tu_berlin_ip), checkbox_unchecked, WFM_LABEL_AFTER);      // custom html input for tu berlin ntp server
    new (&tu_dresden_ntp_server) WiFiManagerParameter("tu_dresden_ntp_server", "TU Dresden", tu_dresden_ip, strlen(tu_dresden_ip), checkbox_unchecked, WFM_LABEL_AFTER);    // custom html input for tu dresden ntp server
    new (&t_online_ntp_server) WiFiManagerParameter("t_online_ntp_server", "T-Online", t_online_ip, strlen(t_online_ip),checkbox_checked, WFM_LABEL_AFTER);        // custom html input for t-online ntp server
    new (&lineBreak) WiFiManagerParameter (lineBreak_str);
    new (&custom_0_ntp_server) WiFiManagerParameter("custom_0_ntp_server", "NTP Server IP:", NULL, strlen(custom_0_ip), "placeholder=\"000.000.000.000\"");
    
    wm.addParameter(&title);
    wm.addParameter(&tu_berlin_ntp_server);    
    wm.addParameter(&tu_dresden_ntp_server);
    wm.addParameter(&t_online_ntp_server);
    wm.addParameter(&lineBreak);
    wm.addParameter(&custom_0_ntp_server);

    wm.setPreSaveParamsCallback(saveParamCallback);
    wm.setSaveParamsCallback(saveParamCallback);
```

callback function
```
void saveParamCallback(void) {

    static char*   lost_ips[3];
    static int     lost_len[3];
    uint8_t        lost_cnt = 0;
    
    static bool preSave = true;
    preSave ? Serial.println("[CALLBACK] savePreParamCallback fired") : Serial.println("[CALLBACK] saveParamCallback fired");
    preSave = ! preSave;
    
    for (uint8_t i = 0; i < wm.getParametersCount(); i++) {
        const char* name = wm.getParameters()[i]->getID();
        if (name) {
            const char* param = getParam(name);
        
            if (param) {
                Serial.println(String(i) + " was checked");
                Serial.print("PARAM[" + String(name) + "]: ");
                Serial.println(param);
               
                // hier dann die ntp client objecte erstellen
            }
            else {
                const char* value = wm.getParameters()[i]->getValue();
                int         length = wm.getParameters()[i]->getValueLength();
                Serial.println(String(i) + " was nullptr");
                Serial.println("val: " + String(value)); 
                Serial.println("len: " + String(length));
                //wm.getParameters()[i]->setValue(value,length);
            }
        }
        else {
            Serial.println(String(i) + " has no id");
        }
        Serial.println();
    }
}
```