tzapu/WiFiManager #1118: Problem with Autoconnect Method

raubuchs, 2020-09-01:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x ] 2.4.0
- [ ] staging (master/dev)

### Description

Hello

The autoconnect method didn't work with a Wemos D1 Mini
It didn't start

### Settings in IDE

Module:  			Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <ESP8266WiFi.h>

#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

#include <string>

void setup() {
  Serial.begin(115200);
  Serial.println("start...");
  pinMode(LED_BUILTIN, OUTPUT);
  WiFiManager wifiManager;
  /*char string[] = "Smart_Device_";
  String mac = WiFi.macAddress();
  mac = mac.substring(9);
  mac.replace(":","");
  String temp = strcat(string, mac.c_str());*/
  digitalWrite(LED_BUILTIN, HIGH);
  wifiManager.autoConnect();
  //if you get here you have connected to the WiFi
  Serial.println("connected...");
}

void loop() {
  if ((WiFi.status() == WL_CONNECTED))
  {

    // TODO Build Webpage for ACB in here
    for (int i = 0; i < 10; i++)
    {
      digitalWrite(LED_BUILTIN, HIGH);
      delay(1000);
      digitalWrite(LED_BUILTIN, LOW);
      delay(1000);
    }
    
  }
  else
  {
  }
}
```

### Debug Messages

```
-
```

raubuchs, 2020-09-01:
After a complete reset of the controller (erase flash, power down, ...) the autoconnect works now

closed 2020-09-01 by raubuchs

tablatronix, 2020-09-01:
sometimes the flash gets corrupt or changes after updated to esp lib