tzapu/WiFiManager #1119: getWiFiIsSaved() returns true if no WiFi is saved when no reset is called

Lithimlin, 2020-09-01:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other: TTGO T8

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

As the title says, when I call `getWiFiIsSaved()` it returns true if I don't call `resetSettings()` before it after startup/deep sleep.
If I reset the settings on the initial startup, the method works as expected, returning `false` if the AP is exited without configuring WiFi settings. On subsequent startups, however, it returns `true` if the settings are not reset explicitly.

I would expect it to still return `false` since no settings have been saved.

Note: Even without the deep sleep, as long as I don't call `resetSettings()`, `getWiFiIsSaved()` returns `true`.

### Sketch

```cpp
#include <WiFiManager.h>

#define uS_TO_S_FACTOR 1000000ULL

WiFiManager wm;

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:

  if(esp_sleep_get_wakeup_cause() == 0) {
    wm.resetSettings();
  }
  Serial.println("WiFi saved? " + String(wm.getWiFiIsSaved() ? "YES" : "NO"));

  if(wm.autoConnect("AutoConnectAP","password")) {
    Serial.println("Failed to connect");
    // ESP.restart();
  } else {
    //if you get here you have connected to the WiFi    
    Serial.println("connected...yeey :)");
  }
  
  Serial.println("WiFi saved? " + String(wm.getWiFiIsSaved() ? "YES" : "NO"));

  uint8_t sleepSeconds = 5;
  esp_sleep_enable_timer_wakeup(sleepSeconds * uS_TO_S_FACTOR);
  Serial.println("Sleeping for " + String(sleepSeconds) + " seconds...");
  esp_deep_sleep_start();
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages

```
*WM: [1] resetSettings 
*WM: [3] WiFi_enableSTA enable
*WM: [1] SETTINGS ERASED 
WiFi saved? NO
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WiFi station disconnect 
*WM: [3] WiFi_enableSTA enable
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AutoConnectAP
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             AutoConnectAP
*WM: [1] password:         password
*WM: [1] ssid_len:         13
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 3105 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] Scan is cached 7300 ms ago
*WM: [3] -> 192.168.4.1 
*WM: [2] <- HTTP Exit 
*WM: [3] configportal abort 
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_DISCONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
connected...yeey :)
WiFi saved? NO
Sleeping for 5 seconds...
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
WiFi saved? YES
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] No wifi save required, skipping 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WiFi station disconnect 
*WM: [3] WiFi_enableSTA enable
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AutoConnectAP
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             AutoConnectAP
*WM: [1] password:         password
*WM: [1] ssid_len:         13
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC completed in 3305 ms
*WM: [2] WiFi Scan ASYNC found: 38
*WM: [2] WiFi Scan completed in 3306 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
```

Lithimlin, 2020-09-01:
I've traveled down the rabbit hole a bit and found out that `getWiFiSSID()` returns a specific string when not reset after startup/deep sleep.

Here is the relevant line of code:
```cpp
Serial.println("WiFi SSID: " + String(wm.getWiFiSSID(true)));
```

When resetting the settings, the output is
```
WiFi SSID: 
```
So there is no SSID saved since it was reset.

Without the reset, the output always is
```
WiFi SSID: G⸮⸮⸮⸮?P⸮?
```

tablatronix, 2020-09-01:
This function returns if the esp has credentials stored in flash. Other than that I have no idea what you are doing but it has nothing to do with saving in wifimanager.

Lithimlin, 2020-09-01:
Alternatively, is there a method that connects to a WiFi network when one has been configured and returns an indicator when that fails? What I want to do is connect to a previously configured WiFi network if it is available but not start the AP when that fails.

tablatronix, 2020-09-01:
They way to use this would be to always start the CP if wifiissaved == false; meaning there are no credentials saved in this esp and it is new or erased

tablatronix, 2020-09-01:
if wifiissaved()
begin()

Lithimlin, 2020-09-01:
So have an extra `RTC_DATA_ATTR bool wifiConfigured` variable to externally keep track of whether it has been configured before?

Lithimlin, 2020-09-01:
> Or you can use autoconnect and use the disablecp settting

I'm confused what you mean.
I assume you mean captive portal when you say CP. So does that mean that if I disable the CP, the ESP will also not start a soft AP and just try to connect to the stored WiFi network?

tablatronix, 2020-09-01:
Yes

tablatronix, 2020-09-01:
https://github.com/tzapu/WiFiManager/blob/development/examples/DEV/OnDemandConfigPortal/OnDemandConfigPortal.ino#L251

tablatronix, 2020-09-01:
oops that not it

Lithimlin, 2020-09-01:
Using `wm.setCaptivePortalEnable(false);` does not actually disable the soft AP though (I just tried it and that does make sense). So what is the correct way to do it?
As it is now, the code above (`wm.autoConnect("AutoConnectAP","password")`) always starts a soft AP if there is no WiFi config saved, even when disabling the captive portal. I have not found a method to disable the config portal itself. The only thing I found was `stopConfigPortal()` but since `autoConnect(...)` is blocking, calling that won't help.

tablatronix, 2020-09-01:
// if true (default) then start the config portal from autoConnect if connection failed
    void          setEnableConfigPortal(boolean enable);

Lithimlin, 2020-09-01:
Ah! Thanks a lot! That's probably what I was looking for.
Quick question: Does it also prevent the explicit starting of a config portal via `wm.startConfigPortal`?

tablatronix, 2020-09-01:
no

Lithimlin, 2020-09-01:
Yes! This method was what I was looking for. Although, is it intended that `autoConnect` returns true when it didn't connect to anything and the config portal is disabled?

tablatronix, 2020-09-01:
hmm, i would not think so, that sounds wrong.
is that what you are seeing ?

Lithimlin, 2020-09-01:
Yes, here's the updated code:

```cpp
#include <WiFiManager.h>

#define uS_TO_S_FACTOR 1000000ULL

WiFiManager wm;

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:

  if(esp_sleep_get_wakeup_cause() == 0) {
    wm.resetSettings();
  } else {
    wm.setEnableConfigPortal(false);
  }

  if(wm.autoConnect("AutoConnectAP","password")) {
    Serial.println("Failed to connect");
  } else {
    //if you get here you have connected to the WiFi    
    Serial.println("connected...yeey :)");
  }

  uint8_t sleepSeconds = 5;
  esp_sleep_enable_timer_wakeup(sleepSeconds * uS_TO_S_FACTOR);
  Serial.println("Sleeping for " + String(sleepSeconds) + " seconds...");
  esp_deep_sleep_start();
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

I immediately exit the config portal the first time, without configuring the WiFi and get the following output:
```
*WM: [1] resetSettings 
*WM: [3] WiFi_enableSTA enable
*WM: [1] SETTINGS ERASED 
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WiFi station disconnect 
*WM: [3] WiFi_enableSTA enable
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AutoConnectAP
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             AutoConnectAP
*WM: [1] password:         password
*WM: [1] ssid_len:         13
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 3105 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] WiFi Scan completed in 3706 ms
*WM: [3] -> connectivitycheck.platform.hicloud.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Exit 
*WM: [3] configportal abort 
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_DISCONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
connected...yeey :)
Sleeping for 5 seconds...
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] No wifi save required, skipping 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
connected...yeey :)
Sleeping for 5 seconds...
```

So it actually also returns true the first time around when I simply exit the CP.

Lithimlin, 2020-09-01:
Well, nevermind that, I was missing a `!` in front of the `wm.autoConnect(...)`, so it's actually always returning false as it should

Lithimlin, 2020-09-01:
So thanks for the help!

closed 2020-09-01 by Lithimlin