tzapu/WiFiManager #1121: HELP WANTED: WM with ESPAsyncWebserver

dronecz, 2020-09-06:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [x] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I´m looking for someone, who is willing to help me with adding ESPAsyncWebserver via `#define` in sketch like [btomer](https://github.com/btomer/WiFiManager) did few years back. I did some work on [this](https://github.com/dronecz/WiFiManager/commit/f197f9d2776fbb09af980c753d5efffb396685cd) in my fork, but so far I can not compile sketch with this define.

tablatronix, 2020-09-06:
Ill take a look and add a branch, might be a few days

ortegafernando, 2020-09-07:
Hi, may be this is a good start point or a good excuse to "join" both projects:

[https://github.com/khoih-prog/ESP_WiFiManager](https://github.com/khoih-prog/ESP_WiFiManager)

tablatronix, 2020-09-07:
The async library is mostly compatible with the regualar one, so there is not much to do to add it, just some testing

tablatronix, 2020-12-09:
I am actually using it right now to see what we need to do

fkoteam, 2020-12-10:
> 
> 
> I am actually using it right now to see what we need to do

is it uploaded? what version should I use?

Thank you very much for your work!

tablatronix, 2020-12-11:
Good news, it should work and be doable
Bad news, its alot of work, lots of refactoring

dronecz, 2020-12-11:
Awesome, can we help somehow?

tablatronix, 2020-12-11:
I need to figure out what needs to change for the new ota upload, and to find the new syntax to get args.
I pushed a new branch, it is in progress, as far as I got atm. But its a start and you can see the changes in handleroot for sending headers now

tablatronix, 2020-12-14:
Its done, ready to test, there are a few things that might be broken, ota uploading has to be rewritten , esp32 not worked on yet, and url params not added back in for no scan and refresh

tablatronix, 2020-12-18:
Have you tried this yet?

dronecz, 2020-12-21:
Hi, sorry for delay, I had no time to try it until now. I implemented in my code and it works. I was able to connect to my AP via WM without problem. 

Many thanks for this. 

Is there something specific which you want me to test?

tablatronix, 2020-12-21:
There are some gotchas that I need to continue to work on, but It should work ok. Just that it works, you can save, change reset, use params whatever you were doing before.

dronecz, 2021-05-12:
Hi, sorry for long delay but I finally have time to make more work on my project with M using ESPAsyncWebserver. 

I tried [OnDemandConfigPortal](https://github.com/tzapu/WiFiManager/blob/master/examples/OnDemand/OnDemandConfigPortal/OnDemandConfigPortal.ino) sketch and with async library does not scan and show networks.

Here is log:

```

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

 Starting
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station disconnect 
*WM: [3] WiFi_enableSTA enable
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
*WM: [1] StartAP with SSID:  OnDemandAP
*WM: [2] AP has anonymous access! 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
OnDemandAP
*WM: [1] password:         
*WM: [1] ssid_len:         10
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN 

*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients...[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 16 - AP_STACONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 18 - AP_STAIPASSIGNED
*WM: [2] Portal Timeout In 103 seconds
*WM: [2] <- HTTP Root 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] WiFi Scan ASYNC started 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
*WM: [2] <- HTTP Wifi 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [1] No networks found 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [3] Sent config page 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
*WM: [2] <- HTTP Wifi 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [1] No networks found 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [3] Sent config page 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
*WM: [2] Portal Timeout In 111 seconds
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
*WM: [2] <- HTTP Root 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] WiFi Scan ASYNC started 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
*WM: [2] <- HTTP Info 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [3] Sent info page 
[W][AsyncTCP.cpp:969] _poll(): rx timeout 4
*WM: [2] <- HTTP Exit 
*WM: [3] configportal abort 
*WM: [2] disconnect configportal 
*WM: [0] [ERROR] disconnect configportal - softAPdisconnect FAILED 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_DISCONNECTED
*WM: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 15 - AP_STOP
[2] wifi mode: STA
*WM: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 15 - AP_STOP
failed to connect and hit timeout

```

Thanks for help.

dronecz, 2021-08-16:
@tablatronix any hint on this one? I looked on this again and difference is somewhere in `autoConnect()` function. If I comment out call to this function and start configportal manually by pressing button, I do not see any networks. Also value of `_lastscan` variable is not reset when reach 60 seconds.

tablatronix, 2022-02-23:
I merged master back into this and resolved conflicts, ota seems all jacked up, code needs to be updated.

Backport the following methods to abstact master to prepare for cross compile..

- [ ] setupHTTPServer
- [ ] teardownHTTPServer
- [ ] setupDNSD
- [ ] HTTPSend

- [ ] template all web handlers with arguments to allow both or other webservers objects or globals

tablatronix, 2023-05-11:
My idea was to make WM compatible with both or any webserver, but I have no idea how to get this working ideally.

I want to keep route callbacks as seperate methods, which means lots of functions to deal with.
therefore I want to avoid having to have overloads for these callbacks..

My idea is this

```c++
template <typename T>
void          callbacknofmany(T *obj, bool argx);

std::unique_ptr<choiceserver> server;

server->on("", HTTP_ANY, std::bind(&myclass::callbacknofmany, this, null , true)); // sync - global or server pointer
server->on("", HTTP_ANY, std::bind(&myclass::callbacknofmany, this, std::placeholders::_1, true)); // async  - local response obj

template <typename T>
void myClass::callbacknofmany(T *serverobj, bool argx) {
    if(serverobj) serverobj->sendresponse();
    else server->sendresponse()
}
```

lucaoliano, 2024-03-08:
Hi @tablatronix,
on my [fork](https://github.com/lucaoliano/WiFiManager/tree/async_webserver) I have an implementation that supports both WebServer and AsyncWebServer. Can you take a look at it? If you think it would be good, I'm happy to contribute with a PR.

tablatronix, 2024-03-08:
Thanks, I was hoping to find a way to do it much cleaner than having all those blocks, makes maintenance a nightmare and code legibility is a mess

alexhopeoconnor, 2026-09-09:
Following the earlier fork reference here, I have opened #1873 to propose a realistic path for addressing this work as a new major version. The proposal is not a request to merge the fork unchanged: it asks whether there is support for an async/chunked portal direction, restored multi-language support, and an explicit compatibility/migration boundary rather than trying to keep the sync and async implementations indefinitely in one code path.

tablatronix, 2026-09-12:
Yeah I have gave up trying to write portable methods with callback references and without, and forking both into 1 branch is dumb.

I would have to merge chunking into the async branch and it should be mostly complete.

I think there was an outstanding issue with OTA, have you tested that branch

alexhopeoconnor, 2026-09-28:
> Yeah I have gave up trying to write portable methods with callback references and without, and forking both into 1 branch is dumb.
> 
> I would have to merge chunking into the async branch and it should be mostly complete.
> 
> I think there was an outstanding issue with OTA, have you tested that branch

Replied over in the issue I created but was more talking about dropping support for the legacy release, tidying up the fork and adding back stuff I removed during the process and then releasing as a new major version.

I added tests for OTA now as to be honest had never really used it and fixed up a few issues but have not really had issues with available flash size, but have only really tested on Wemos D1 Mini boards for esp8266, so I am fairly sure any OTA issue has been fixed.