tzapu/WiFiManager #1122: Autoconnect problem after network loss - need some help

FilipDem, 2020-09-10:
#### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
WifiManager 2.0.3-alpha from the Arduino Library Manager

**Esp8266/Esp32: ESP8266**
**Hardware:			ESP-12F**
**ESP Core Version:	2.7.4**

### Description
I am using the WifiManager... Now it happens that the network is interrupted. I see then that the AP is activated (from LED behavior and network scan). The network comes up again (because my PC has again a WIFI connection. Afterwards the AP is quit (probably after timeout), however there is no network access. After reboot all comes back OK.

Who could help me to the behavior of the WifiManager to detect that the network is up again?
Or somebody has some ideas what could be the problem, always welcome.

### Concept Sketch

```
void setup() {
...
		WiFiManager wifiManager;
		wifiManager.setDebugOutput(DEBUG_WIFIMANAGER);
		wifiManager.setShowStaticFields(true);
		static const char* menu[] PROGMEM = {"wifi", "wifinoscan", "restart", "exit"};
		wifiManager.setMenu(menu, 4);
		wifiManager.setAPCallback(APconfigModeCallback);
		wifiManager.setAPStaticIPConfig(AP_IP_ADDRESS, AP_GATEWAY, AP_SUBNET_MASK);
		wifiManager.setConfigPortalTimeout(TIMEOUT_AP_PORTAL);

		// Connect to network or active AP if no connection can be made
		if (wifiManager.autoConnect(APname, APpassword)) {
			// Send the IP address of the ESP8266 to the computer
			DEBUG(Serial.print(F("\nConnected to: "));Serial.println(WiFi.SSID());Serial.print(F("IP address: "));Serial.println(WiFi.localIP());)
	
			// Update status of the connection
			LastConnectionStatus = true;
			LastConnectionDateTime = CurrentDateTime;
	
			// Led on
			ticker.detach();
			digitalWrite(LED, LOW);
	
			return true;
		}
		// Active AP has quit after timeout; no network connection
		else {
			DEBUG(Serial.print(F("\nNo network connection!"));)
	
			// Led slowly blinking if no network
			ticker.attach(1, toggle_led);
			
			return false;
		}

	}
...

}

```
void loop() {
```
...
		if (WiFi.status() == WL_CONNECTED) {
			// Connection is just restored
			if (LastConnectionStatus == false) {
				// Toggle status
				LastConnectionStatus = true;
				LastConnectionDateTime = CurrentDateTime;
				// Led on
				ticker.detach();
				digitalWrite(LED, LOW);
				DEBUG(Serial.println(F("WIFI connected again!"));)
			}
			return true;
		}
		// Not connected to network
		else {
			// Connection is just lost
			if (LastConnectionStatus == true) {
				// Toggle status and count disconnections
				LastConnectionStatus = false;
				NbrDeconnections++;
				DEBUG(Serial.println(F("WIFI disconnected!"));)
				// Led slowly blinking if no network
				ticker.attach(1, toggle_led);
			}
			// Connection stay lost: reboot after a certain time
			else {
				if ( CurrentDateTime-LastConnectionDateTime > MAX_TIME_WITHOUT_INTERNET ) {
					ESP.restart();
				}
			}
			return false;
		}
...

```
}
```

tablatronix, 2020-09-10:
autoconnect autostarts configportal on fail, it should only start on reboot, unless you disable that (its optional)

esp should auto reconnect, so why it is not is the question.

It sounds like your esp is having a problem reconnecting, it could be another bug or unrelated to wm.

ddweber456, 2020-10-28:
I have had the same issue with a power outages.  Router takes 30 - 40 sec to connect to the web, mean time the ESP wifiManager.autoConnect is already in AP mode waiting for SSID/Password.  

I fixed the problem with wifiManager.setTimeout and a ESP.restart.

   wifiManager.setTimeout(120);   
   wifiManager.setConfigPortalTimeout(120);
   
  if(!wifiManager.autoConnect("APSetup WIFI", "password")) {
    Serial.println("failed to connect 1st time after BOOT and hit timeout");
    delay(100);
    ESP.restart();

tablatronix, 2020-10-29:
Yeah I would just add a little loop and deal with it in your own code.

tablatronix, 2020-10-29:
if you wifiManager.setTimeout(120); then it should keep checking before starting the cp, but maybe there is something else going on. There are documented issues of connects failing first time, I would reproduce and get some logs to see why the 120S is not enough

tarontop, 2021-03-16:
> 我曾经遇到过停电的问题。路由器需要30到40秒才能连接到网络，这意味着ESP wifiManager.autoConnect已处于AP模式，正在等待SSID /密码。
> 
> 我解决了wifiManager.setTimeout和ESP.restart的问题。
> 
> wifiManager.setTimeout（120）;
> wifiManager.setConfigPortalTimeout（120）;
> 
> if（！wifiManager.autoConnect（“ APSetup WIFI”，“ password”））{
> Serial.println（“启动后第一次连接失败，命中超时”）;
> 延迟（100）;
> ESP.restart（）;
> 
> 我还在安装程序中添加了以下内容：
> 
> ```
> WiFi.persistent(true);
> WiFi.setAutoConnect(true);
> WiFi.setAutoReconnect(true);
> ```

The problem I encountered is the same as you. I used non-blocking. How can I reconnect to Wi-Fi without restarting ESP8266?

tablatronix, 2021-03-16:
esp8266 is supposed to autoreconnect

tarontop, 2021-03-17:
> esp8266应该自动重新连接

When I test, it will reconnect, but this is only when the ESP8266 is not powered off, only the router is powered off, and then the router ESP8266 will reconnect, if the ESP8266 and the router are both powered off, they will not reconnect , Because it takes 2 minutes for my router to start up, ESP8266 automatically enters the AP mode.

tablatronix, 2021-03-17:
wm.setConfigportalTimeout()

There are about 5 solutions to this discussed all over these issues

tarontop, 2021-07-09:
> wm.setConfigportalTimeout()
> 
> 在这些问题上讨论了大约 5 个解决方案

I tried non-blocking, `wm.setConfigportalTimeout()` does not seem to work

tablatronix, 2021-07-10:
Hmm have log output?

tarontop, 2021-07-10:
[image]

tarontop, 2021-07-10:
After 30 seconds it was still in AP mode, it did not close the captive portal (I used non-blocking)

tarontop, 2021-08-20:
Hello, will the poor signal quality cause the device to restart? I have a device whose signal is not very good, and the device will restart from time to time

tablatronix, 2021-08-21:
That sounds like bad power, unless you have some code to do a restart no

tablatronix, 2021-08-21:
@tarontop can you make a new issue for this, it seems to be a valid bug. If not I will try to sometime soon.