tzapu/WiFiManager #1123: Spiffs gets erased after ssid and password save

Qbunjo, 2020-09-10:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ x] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x ] 2.4.0
- [ ] staging (master/dev)

### Description

Hi there, 
I Just added your manager to my project. 
https://github.com/The-Coobaz/hunter-hack 
 I use spiffs to keep web page and css. But When I added the manager, after saving ssid and pass, The spiffs gets cleared. I don't know why id does happen, since I can't find in your library place where the credentials gets saved. 

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2020-09-10:
The esp saves the credentials on its own. So I suspect you have a esp lib issue or corrupt memory, do a full erase.

Qbunjo, 2020-09-10:
Whoah! What a prompt answer! I will examine that.

tablatronix, 2020-09-10:
also may sound dumb, but make sure you are not calling spiff.format.... lol

Qbunjo, 2020-09-11:
Noo, I am not that dumb :)

Qbunjo, 2020-09-11:
Hi. I did as you advised. Did full erase and updated library to current dev version. Everything works now. Thanks for your hint.

closed 2020-09-11 by Qbunjo