tzapu/WiFiManager #1124: Help with AutoConnectWithFSParameters example with simple modification.

Shalabyer, 2020-09-13:
**Esp8266/Esp32:**
- [ ] ESP8266
**Hardware:
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
### Description
In the great example AutoConnectWithFSParameters it tries to save SSID, Password, localIP, gatewayIP and subnetMask of a WI-FI network for the ESP8266 by giving these data via web while the ESP acts as a hotspot.
Well that's all okay but what i need help with and i need to do is to give these data to the ESP to save via code and not via link,
here is an example of what i want to do :
```

#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#ifdef ESP32
  #include <SPIFFS.h>
#endif

void(* resetFunc) (void) = 0; //declare reset function @ address 0
void setup()
{
String SSIDName="test";
String Password="test";
String localIPs="192.168.1.10";
String gatewayIPs="192.168.1.1";
String subnetMasks="255.255.255.0";

}
void loop()
{
if(true)
{
   WiFiManager wifiManager;    
    wifiManager.resetSettings();
    //HERE save new SSIDName, Password, localIPs, gatewayIPs and subnetMasks to the ESP8266
    //the above line is the code that i want please if you may help
  resetFunc();
}
}

```

tablatronix, 2020-09-13:
Then just use the esp functions for this

Shalabyer, 2020-09-13:
Thank for your help,
i searched throught all the public function on the library and found non to set SSID name and Password of the Wi-Fi
But i have made it to save a new IP address  here is the new code

```
#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

#ifdef ESP32
  #include <SPIFFS.h>
#endif

void(* resetFunc) (void) = 0; //declare reset function @ address 0
void setup()
{
String SSIDName="test";
String Password="test";
String localIPs="192.168.1.10";
String gatewayIPs="192.168.1.1";
String subnetMasks="255.255.255.0";

}
void loop()
{
if(true)
{
 
         WiFiManager wifiManager;    
    wifiManager.resetSettings();
    
     #ifdef ARDUINOJSON_VERSION_MAJOR >= 6
    DynamicJsonDocument json(1024);
  #else
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
  #endif

    json["ip"] ="192.168.43.99";
    json["gateway"] = "192.168.43.1";
    json["subnet"] = "255.255.255.0";

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }
  #ifdef ARDUINOJSON_VERSION_MAJOR >= 6
    serializeJson(json, Serial);
    serializeJson(json, configFile);
  #else
    json.printTo(Serial);
    json.printTo(configFile);
  #endif
    configFile.close();
//    wifiManager.wifiConnectNew("testSSID","testPassword");  
//this is a private function that i found but i can't use it as It's Private
  resetFunc();
}
}
```
**What** i want here is to save new SSID name and Password to the ESP8266 file manually without the need to enter the data through ESP Access Point.
If you could please help me with function names that i can use to save these two paramters.

tablatronix, 2020-09-13:
Esp.begin()

Shalabyer, 2020-09-13:
For WiFi.setConfig() function, i didn't found any info, hope you mean WiFi.config() which only do config the IP address.
now for the WiFi.begin() function,
i added it to my code like the following:

  ```
  configFile.close();
   WiFi.begin("testSSID","testPass");
    Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
  resetFunc();
```
It prints that it connected successfully which is good but the bad thing here that it doesn't save these two parameters as when it resets it enters AP mode as it didn't save the SSID and PASS but it saved All the new IP Addresses .
This is what i get when it resets to make it more clear:

> reading config file
> opened config file
> {"ip":"192.168.43.99","gateway":"192.168.43.1","subnet":"255.255.255.0"}
> parsed json
> setting custom ip from config
> 192.168.43.99
> *WM: [1] AutoConnect 
> *WM: [1] No Credentials are Saved, skipping connect 
> *WM: [2] Starting Config Portal 
> *WM: [2] AccessPoint set password is VALID 
> *WM: [1] password 
> *WM: [3] WIFI station disconnect 
> *WM: [3] WiFi station enable 
> *WM: [2] Disabling STA 
> *WM: [2] Enabling AP 
> 

**NOTE**: "192.168.43.99" is the new IP address

tablatronix, 2020-09-14:
Sorry I meant WiFi.config it takes all addresses

Shalabyer, 2020-09-14:
Iam sorry man but what about saving wifi data like ssid and password, Wifi.begin doesn't save these data it just starts the connection for one time.

tablatronix, 2020-09-14:
No it saves it

Shalabyer, 2020-09-14:
Allrigth here is my full code if you please may help me with that if iam getting it wrong

```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <SPI.h> //What is used to communicate witht he WiFi chip
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

//default custom static IP
char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";
//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}
void(* resetFunc) (void) = 0; //declare reset function @ address 0
int status = WL_IDLE_STATUS; //status of wifi
WiFiServer server(80); //declare server object and spedify port, 80 is port used for internet
void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  

  Serial.begin(115200);
  delay(10);
if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);

        #ifdef ARDUINOJSON_VERSION_MAJOR >= 6
          DynamicJsonDocument json(1024);
          DeserializationError deserializeError = deserializeJson(json, buf.get());
          serializeJson(json, Serial);
          if (!deserializeError) {
        #else
          DynamicJsonBuffer jsonBuffer;
          JsonObject& json = jsonBuffer.parseObject(buf.get());
          json.printTo(Serial);
          if (json.success()) {
        #endif
          Serial.println("\nparsed json");
        if (json["ip"]) {
            Serial.println("setting custom ip from config");
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);
            Serial.println(static_ip);
          } else {
            Serial.println("no custom ip in config");
          }
        } else {
[... 273 lines cut ...]
> *WM: [1] authmode:         3
> *WM: [1] ssid_hidden:     
> *WM: [1] max_connection:   4
> *WM: [1] country:          CN
> *WM: [1] beacon_interval:  100(ms)
> *WM: [1] -------------------- 
> *WM: [1] AP IP address: 192.168.4.1
> *WM: [3] setupConfigPortal 
> *WM: [1] Starting Web Portal 
> *WM: [3] dns server started with ip:  192.168.4.1
> *WM: [2] HTTP server started 
> mode : sta(84:0d:8e:8d:ed:80) + softAP(86:0d:8e:8d:ed:80)
> add if0
> scandone
> *WM: [2] WiFi Scan completed in 2187 ms
> 

Also i tried to put this line:
` WiFi.begin("Xperia","88888888m");
`
after the this line:
`    configFile.close();
`
and still the same response , No save for the SSID and PASS.
I would be pleased for help,Thanks anyway :)

Shalabyer, 2020-09-25:
Any help :)

tablatronix, 2020-09-25:
alot going on here, first off why are you parsing the web server, you can just add a route and a callback and then check if the arguments exist for querystrings. There are web server functions to do this.

It looks like you are still connected after resetsettings, that is odd, maybe it is ok after reboot.

Not sure when you say not saved, `WiFi.begin("Xperia","88888888m");` is saved by the esp

closed 2020-10-18 by Shalabyer