tzapu/WiFiManager #1125: Nodemcu Wifi Manger Country code error

Lavanyarajan, 2020-09-14:
Hello All,

I want to connect nodemcu to required wifi network without hard coding the SSID and pw details. For that Im trying to use WiFimanager and I downloaded it in arduino libraries.

When I try to compile the code the following error occurs:

C:\Users\xxxxx\Arduino\libraries\WiFiManager/strings_en.h:358:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_US{"US",1,11,WIFI_COUNTRY_POLICY_AUTO};

       ^

C:\Users\xxxxxx\Arduino\libraries\WiFiManager/strings_en.h:359:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,WIFI_COUNTRY_POLICY_AUTO};

       ^

C:\Users\xxxxx\Arduino\libraries\WiFiManager/strings_en.h:360:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,WIFI_COUNTRY_POLICY_AUTO};

       ^

How to resolve this error. Pls help me out

tablatronix, 2020-09-14:
What board

Lavanyarajan, 2020-09-14:
LoLin nodemcu board

tablatronix, 2020-09-14:
Sounds like your boards might be out of date or old, what esp lib are you on ?

Lavanyarajan, 2020-09-14:
Esp core library v2.3.0

tablatronix, 2020-09-14:
you arduino or pio ? can you enable verbose compile logging and post it?

Lavanyarajan, 2020-09-14:
I'm using arduino ide, how to enable verbose compile

tablatronix, 2020-09-14:
its in settings
[image]

Lavanyarajan, 2020-09-15:
verbose logging.. pls check it

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\SUKSHM_3D\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\SUKSHM~1\AppData\Local\Temp\arduino_build_954805 -warnings=none -build-cache C:\Users\SUKSHM~1\AppData\Local\Temp\arduino_cache_693921 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Users\SUKSHM_3D\Desktop\DESKTOP FILES\nodemcu autoconnect\autoconnect-1\autoconnect-1.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\SUKSHM_3D\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadSpeed=115200,FlashSize=4M3M -ide-version=10805 -build-path C:\Users\SUKSHM~1\AppData\Local\Temp\arduino_build_954805 -warnings=none -build-cache C:\Users\SUKSHM~1\AppData\Local\Temp\arduino_cache_693921 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9 -verbose C:\Users\SUKSHM_3D\Desktop\DESKTOP FILES\nodemcu autoconnect\autoconnect-1\autoconnect-1.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Using core 'esp8266' from platform in folder: C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0
Detecting libraries used...
"C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\SUKSHM~1\AppData\Local\Temp\arduino_build_954805/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "C:\Users\SUKSHM~1\AppData\Local\Temp\arduino_build_954805\sketch\autoconnect-1.ino.cpp" -o "nul"
"C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\SUKSHM~1\AppData\Local\Temp\arduino_build_954805/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266" "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\variants\nodemcu" "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src" "C:\Users\SUKSHM~1\AppData\Local\Temp\arduino_build_954805\sketch\autoconnect-1.ino.cpp" -o "nul"
"C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/include" "-IC:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include" "-IC:\Users\SUKSHM~1\AppData [... cut ...]
Using library DNSServer at version 1.1.0 in folder: C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\DNSServer 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\SUKSHM_3D\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer 
Using library WiFiManager at version 2.0.3-alpha in folder: C:\Users\SUKSHM_3D\Documents\Arduino\libraries\WiFiManager 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

tablatronix, 2020-09-15:
Ok , I see, It looks like I was not testing with 2.3.0

Why are you using such an old version of esp ?

tablatronix, 2020-09-15:
I fixed this in dev branch, country settings are removed now in 2.3.0

closed 2020-09-15 by tablatronix

Lavanyarajan, 2020-09-16:
Means I need to reinstall the Wifi Manager library or esp version?

tablatronix, 2020-09-16:
Well its only on github, not sure how you install dev branches on arduino.
If you just use a newer version of esp lib you should be okm Is there a reason you are using such an old outdated version ?

tablatronix, 2020-09-16:
https://www.baldengineer.com/installing-arduino-library-from-github.html

Lavanyarajan, 2020-09-16:
actually I reinstalled both wifi manager and esp8266 libraries but still im getting same error

Lavanyarajan, 2020-09-16:
ESP version 2.3.0 is the latest I guess.. and thats what I have in my boards manager list