tzapu/WiFiManager #1126: Change wifi network or password

pilere, 2020-09-17:
Hi,
I would love to offer the ability to change the wifi network or password while already connected, is it possible ? I am using autoconnect at the beginning of my code and it's just perfect but after that I do setup a webserver so the AP one need to be added in noAP mode .... not sure it's possible ... How do you manage password change ... 

Stephane

tablatronix, 2020-09-17:
If you mean via the configportal, then use the ondemand examples to manually start the configportal

pilere, 2020-11-26:
> If you mean via the configportal, then use the ondemand examples to manually start the configportal, however you cannot run the wm webserver and your own at the same time, unless you use the same instance ( is is public in the dev version)

Thanks !!

closed 2020-11-26 by pilere