tzapu/WiFiManager #1127: Store parameters even when WiFi cannot connect

Jeppedy, 2020-09-17:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ X] Master

**Esp8266/Esp32:**
- [X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
Adafruit Feather Huzzah - ESP8266-based

### Description

Feature Request:
Today, the additional parameters are not being stored if a connection to the Wifi cannot be established.
As I'm sure many are, I'm using your library both to manage the Wifi config and as an easy way to allow me to adjust parameters on the device (even when it can connect to WiFi).
All works fine.  No bugs...
But, you've made the design decision to NOT call the "SaveParams" callback if you cannot get a connection, or if the new SSID/PSWD are not supplied.

Preferred:
If the SSID or PSWD are null, fall back to the previously-configured params,
then call the "SaveParams" callback anyway.
In fact, the first thing you should do is call the Save callback, then worry about WiFi.
Or, have two callbacks: "Wifi Connection related", called as it is today, and "Unrelated to Wifi", which is always called when the config portal calls for a save.

After all, the params I have are unrelated to the WiFi connection.

tablatronix, 2020-09-17:
already possible
```C++
    //if this is set, it will exit after config, even if connection is unsuccessful.
    void          setBreakAfterConfig(boolean shouldBreak);
```

In alpha/dev version you can move params to a seperate page and have seperate callbacks.

Jeppedy, 2020-09-17:
Can you point me to an example sketch?

Jeppedy, 2020-09-17:
So setBreakAfterConfig means "save params even if connection didn't
succeed"?

J

tablatronix, 2020-09-17:
Yeah its an old throwback thing, it should be noted in the readme