tzapu/WiFiManager #1128: CANNOT SET STATIC IP

birrex, 2020-09-18:
Hello, I have many days looking for a solution to my problem, I am trying to connect to the wifimanager with a fixed IP, but I cannot.

My case:
I want to connect my nodemcu to a Wi-Fi network through the wifimanager library, then I want to put the fixed IP on the device, to make url requests to do various tasks but I can't.

This is my code

I have searched a lot and I can not find a solution.

`/*
 * Sketch: ESP8266_LED_Control_06_Station_Mode_with_mDNS_and_wifiManager
 * Control an LED from a web browser
 * Intended to be run on an ESP8266
 */

#include <ESP8266WiFi.h>
//#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>   

 
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>

 
// change these values to match your network
//char ssid[] = "MyNetwork_SSID";       //  your network SSID (name)
//char pass[] = "Newtwork_Password";    // your network password
 
WiFiServer server(80);
 
String header = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
String html_1 = "<!DOCTYPE html><html><head><meta name='viewport' content='width=device-width, initial-scale=1.0'/><meta charset='utf-8'><style>body {font-size:140%;} #main {display: table; margin: auto;  padding: 0 10px 0 10px; } h2,{text-align:center; } .button { padding:10px 10px 10px 10px; width:100%;  background-color: #4CAF50; font-size: 120%;}</style><title>test</title></head><body><div id='main'><h2>test</h2>";
String html_2 = "";
String html_4 = "</div></body></html>";
 
String request = "";
int LED_Pin = D1;

String onlyURL;
int index1;
int index2;
String NoUrl;
String NoUrl3;

int Index1 ;
int Index2; 
String NoUrl1;
String onlyURL1;
WiFiManager wifiManager;

void setup() 
{
      pinMode(LED_Pin, OUTPUT); 
 
      Serial.begin(115200);
      delay(500);
      Serial.println(F("Serial started at 115200"));
      Serial.println();
 
      // We start by connecting to a WiFi network
      //Serial.print(F("Connecting to "));  Serial.println(ssid);
      //WiFi.begin(ssid, pass);
 
      //while (WiFi.status() != WL_CONNECTED) 
      //{
      //    Serial.print(".");    delay(500);
      //}
      //Serial.println("");
      //Serial.println(F("[CONNECTED]"));
      //Serial.print("[IP ");              
      //Serial.print(WiFi.localIP()); 
      //Serial.println("]");

     // WiFiManager

      //wifiManager.resetSettings();
      wifiManager.autoConnect("ESP8266","password");
    
      
      // or use this for auto generated name ESP + ChipID
      //wifiManager.autoConnect();
      
      // if you get here you have connected to the WiFi
      Serial.println("Connected.");
    
    
      if (!MDNS.begin("esp8266"))   {  Serial.println("Error setting up MDNS responder!");  }
      else                          {  Serial.println("mDNS responder started");  }
    
      // start a server
      server.begin();
      Serial.println("Server started");
 
} // void setup()
 
 
 
void loop() 
{
 
    // Check if a client has connected
    WiFiClient client = server.available();
    //if (!client)  {  return;  }

    if (client)
    {
          // Read the first line of the request
          request = client.readStringUntil('\r');
          Serial.println("request:");
          Serial.println(request);

            index1 = request.indexOf('/?');  //finds location of first ,
            NoUrl = request.substring(0, index1);   //captures first data String
            index2 = request.indexOf(' ', index1+1 );   //finds location of second ,
            onlyURL = request.substring(index1+1, index2+1);
            Serial.println("La URl SEPARADA ES:"+onlyURL+"txt: "+onlyURL.indexOf("CHANGEIP"));

            Index1 = onlyURL.indexOf(':'); // Si es que se encuentran ,
            Index2 = onlyURL.length();   //finds location of second ,
            onlyURL1 = onlyURL.substring(Index1+1, Index2);
            Serial.println("ip solicitada ");
            Serial.println(onlyURL1);
            
        if(onlyURL.indexOf("CHANGEIP") == 0){
          Serial.print("CHANGEIP...");
          html_2 = "peticion de cambio de ip "+onlyURL1;
         wifiManager.setSTAStaticIPConfig(IPAddress(192,168,1,98), IPAddress(192,168,1,98), IPAddress(255,255,255,0));
//here not work!!!!!! 
        }
      
        if(request.indexOf("/?X=2") > 0){
       
          html_2 = "<form id='F1' action='LEDOFF'><input class='button' type='submit' value='Turn off the LED' ></form><br>";

        }

       if(request.indexOf("/?X=1") > 0){
        html_2 = "<form id='F1' action='LEDOFF'><input class='button' type='submit' value='Turn off the LED' ></form><br>";

        }
   
       
       
          client.flush();
       
          client.print( header );
          client.print( html_1 );    
          client.print( html_2 );
          client.print( html_4);
       
          delay(5);
          
         // The client will actually be disconnected when the function returns and 'client' object is detroyed
           
    }

 
} // void loop()`

EgHubs, 2020-12-11:
Just try [this ](https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParametersAndCustomIP/AutoConnectWithFSParametersAndCustomIP.ino)example (AutoConnectWithFSParametersAndCustomIP).
But first, you may need to reset your wifi settings which can be done by uncommenting this line
`  wifiManager.resetSettings();
`
and then reupload the code with that line commented.

jordanadania, 2021-04-15:
Bad Link

tablatronix, 2021-04-15:
setSTAStaticIPConfig only works when you are using autoconnect

EgHubs, 2021-04-16:
> Bad Link

My bad, link updated