tzapu/WiFiManager #1132: After Timeout, Retry Connecting to Last Known SSID/Password

bwjohns4, 2020-10-05:
This is just a question, is there a built-in way to retry connecting to the last known SSID/Password without a hard reset? I've done this and it works with the reset

if (!wifiManager.autoConnect("MY SSID")){
    BWJ_DEBUG_PRINTLN("failed to connect and hit timeout, now resetting");
    delay(3000);
    ESP.restart();
}

but if I try:

while (!wifiManager.autoConnect("MY SSID")){
    BWJ_DEBUG_PRINTLN("failed to connect and hit timeout, now retrying");
    delay(3000);
}

It does not work. Is there an easy way to fix this so that a retry with the last successful credentials is possible without actually resetting and toggling relay pin states?

tablatronix, 2020-10-05:
Esp8266 automatically autoreconnects, you can try WiFi.begin()

vginside, 2021-05-23:
Hi @tablatronix !
  I have similar problem i tried using simplified code as below:
#include <ESP8266WiFi.h>        //For WiFi Connection
#include <WiFiManager.h>        //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFiManager wifiManager;
  wifiManager.setConfigPortalTimeout(10);    //Use this function to timeout AP portal page; Parameter time in seconds
  if(!(wifiManager.autoConnect()))         // check if wifi is connected within configportaltimeout if not restart
  {
    Serial.println("Failed to connect");
    WiFi.mode(WIFI_OFF);
    WiFi.mode(WIFI_STA);                  // explicitly set mode, esp defaults to STA+AP
    WiFi.begin();
    //ESP.restart();
  } 
  else                              
  {
    //if you get here you have connected to the WiFi
    Serial.println((String)"Connected to Wifi SSID: "+WiFi.SSID());
    Serial.println((String)"Wifi password: "+WiFi.psk());
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
  }
}

void loop()
{
  if (WiFi.status() != WL_CONNECTED)      // Keep checking if wifi Connection is connected or lost
  {
    Serial.print(".");
    delay(500);
  }
  else
  {
    Serial.println("Connected!");
    delay(500);
  }
}

  If configuration timesout then device wont reconnect to saved credentials this is problem for powerfail ofc. Any helps?

vginside, 2021-05-24:
> Hi @tablatronix !
> I have similar problem i tried using simplified code as below:
> #include <ESP8266WiFi.h> //For WiFi Connection
> #include <WiFiManager.h> //https://github.com/tzapu/WiFiManager WiFi [...]

Never mind I was using an older version of 0.14.0... Updating it to 0.16.0 resolved it! Not even need to write WiFi.begin() cheers! :)