tzapu/WiFiManager #1133: Crash when using startConfigPortal after setAPStaticIPConfig

IDkonnecT, 2020-10-06:
Hello, I have a crash issue that I really don't find how to fix using startConfigPortal after setAPStaticIPConfig

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [X ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ X] Other : ESP32 WROVER Module

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [X ] staging (master/dev)

### Description

I have a crash when using startConfigPortal after setAPStaticIPConfig.
The exact same code works properly if I use autoConnect instead startConfigPortal, or if I suppress setAPStaticIPConfig line.

### Sketch

#include <Arduino.h>
#include <WiFiManager.h>

// Device
const char* DeviceName = "Module";

// WiFiManager AP
const char* WiFiManager_SSID = DeviceName;
IPAddress WiFiManager_IP = IPAddress(192,168,1,1);
IPAddress WiFiManager_Gateway = IPAddress(192,168,1,1);
IPAddress WiFiManager_SubNet = IPAddress(255,255,255,0);

WiFiManager wifiManager;
boolean SaveParameters=false;

void Debug(String Texte="", boolean EndLine=true)
{
 if (EndLine) Serial.println(Texte);
 else Serial.print(Texte);
}

void WiFiManagerWiFiFail(WiFiManager *myWiFiManager)
{
  Debug();
  Debug("********************************");
  Debug("Usual WiFi network not available");
  Debug("Reconfigure by connecting to WiFi : '", false);
  Debug(myWiFiManager->getConfigPortalSSID()+"' on page 'http://"+WiFi.softAPIP().toString()+"'");
  Debug("********************************");
  Debug();
}

void WiFiManagerConfigOK()
{
  Debug("Wifi Configuration saved !");
}

void WiFiManagerParamOK()
{
  SaveParameters = true;
}

void SetUpWiFiManager()
{
  Debug();
  Debug("WiFi connexion ongoing...");
 
  wifiManager.setDebugOutput(true);
  wifiManager.setCountry("US");
  wifiManager.setWiFiAPChannel(7);

  wifiManager.setScanDispPerc(false);

  std::vector<const char *> menu = {"wifi"};
  wifiManager.setMenu(menu);
  
  wifiManager.setAPStaticIPConfig(WiFiManager_IP, WiFiManager_Gateway, WiFiManager_SubNet); // IP, Gateway, SubNet
  wifiManager.setConfigPortalTimeout(300);
  wifiManager.setMinimumSignalQuality(5);
  wifiManager.setAPCallback(WiFiManagerWiFiFail);
  wifiManager.setSaveConfigCallback(WiFiManagerConfigOK);
  wifiManager.setSaveParamsCallback(WiFiManagerParamOK);
 
  if (!wifiManager.startConfigPortal(WiFiManager_SSID, "")) Debug("Erreur WiFi !");
}

void setup()
{
  Serial.begin(57600);
  while (!Serial);
  SetUpWiFiManager();
}

void loop()
{
}

### Debug Messages

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
*WM: [2] WiFi Scan completed in 1848 ms
*WM: [2] Config Portal Running, blocking, waiting for clients...
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 16 - AP_STADISCONNECTED
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x8011c499  A1      : 0x3ffb3d30
A2      : 0x3ffcdf14  A3      : 0x3ffce148  A4      : 0x3ffcd520  A5      : 0x3ffcd500
A6      : 0x0201a8c0  A7      : 0x0c01a8c0  A8      : 0x8011c33c  A9      : 0x3ffb3cf0
A10     : 0x3ffcdf24  A11     : 0x3ffce148  A12     : 0x3ffb3d3c  A13     : 0x00000044
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

Backtrace: 0x00000000:0x3ffb3d30 0x4011c496:0x3ffb3d70 0x40126981:0x3ffb3d90 0x4012b989:0x3ffb3dd0 0x40130c26:0x3ffb3df0 0x4011c663:0x3ffb3e10 0x40088b7d:0x3ffb3e40

Rebooting...

Gerry33, 2020-12-03:
Same here. 

Occurs if all flash or WIFI settings cleared .

tablatronix, 2020-12-03:
Someone will have to decode stack traces but sounds like bad flash or esp lib bug