tzapu/WiFiManager #1139: Stay in AP mode after

JonasVorwerk, 2020-11-01:
Great library, love it. 
2.0 has great new features especially setConfigPortalBlocking, m much appreciated!

There is a difference in behaviour with the 0.15 and still in alpha 2.0x versions of WM. 
0.15: after setConfigPortalTimeout, WM stays in AP mode and it's possible to access html served by the esp

`
  WiFiManager wifiManager;
  wifiManager.setConfigPortalTimeout(60);
  wifiManager.setDebugOutput(true);
  wifiManager.autoConnect(HOSTNAME, PASSWORD);
`

2.0: WM unloads itself and puts the esp in STA mode. No option to open files on the webserver / spiffs 
- if we don't use setConfigPortalTimeout WM continues to server it's webportal. 
- This also happens in nonblocking mode using setConfigPortalBlocking(false);
Both desired behaviours but it would be really helpfull if there would be a way to continue to the web server if there might be, in some ocassions, no wifi. 

Why not use the exit button to stay in AP mode but continue to serve html like 0.15 does? Just to have the option to "not connect" it to any AP but run as a stand alone device? 

Maybe this is already possible but it's not documented? Would love to know how ;)

Thanks!!!

Reni0013, 2020-11-04:
This is what i am also looking for. Can anyone has any idea or any hint will be enough too for starting AP mode when ESP is not connected to the WiFi and that too in non blocking mode. @tablatronix Please give some light on this. BTW greatest lib you guys have. Thanks a lot for this lib.

tablatronix, 2020-11-04:
Sorry I thought I responded to this..

It is a bit pointless since you can just use non blocking mode now.. Just do whatever you want in code.
You can also use the webserver as its a shared pointer now.

Both the webserver and softap, and captive portal can be run on demand and non blocking.

Also wm should not stay in ap mode after timing out, so that is not expected behavior.

Reni0013, 2020-11-06:
AutoConnectNonBlocking is the best to way to fulfil this requirement but the thing is that  AutoConnectNonBlocking  is not being connected automatically after i turn off and then turn on my Wifi. Its stuck in WL_DISCONNECTED status.

```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    wm.setConfigPortalBlocking(false);

    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect("AutoConnectAP")){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }
}

void loop() {
    wm.process();
    Serial.println(wl_status_to_string(WiFi.status()));
    Serial.print(wm.getWiFiSSID(true).c_str());
    Serial.print(wm.getWiFiPass(true).c_str());
    Serial.println();

    if (WiFi.status() == WL_CONNECTED) {
      Serial.println("I am Connected yea!!");
    }
    else {
      // ESP is now offline.
      WiFi.begin(wm.getWiFiSSID(true).c_str(), wm.getWiFiPass(true).c_str());
    }
}

const char* wl_status_to_string(wl_status_t status) {
  switch (status) {
    case WL_NO_SHIELD: return "WL_NO_SHIELD";
    case WL_IDLE_STATUS: return "WL_IDLE_STATUS";
    case WL_NO_SSID_AVAIL: return "WL_NO_SSID_AVAIL";
    case WL_SCAN_COMPLETED: return "WL_SCAN_COMPLETED";
    case WL_CONNECTED: return "WL_CONNECTED";
    case WL_CONNECT_FAILED: return "WL_CONNECT_FAILED";
    case WL_CONNECTION_LOST: return "WL_CONNECTION_LOST";
    case WL_DISCONNECTED: return "WL_DISCONNECTED";
  }
}
```

Any idea why it is happening? am i doing something wrong or it is a bug of **wm.setConfigPortalBlocking(false);** 

I am Using ESP8266
WiFiManager Version is 2.0.3 alpha Development branch.

If possible please check above code @tablatronix  is it capable of Reestablish connection with Wifi in void loop?

tablatronix, 2020-11-06:
no idea what do your logs say?

esp should auto connect by itself, if that is now happening then sounds like a esp bug, try autoconnect twice

Reni0013, 2020-11-07:
> esp should auto connect by itself, if that is now happening then sounds like a esp bug, try autoconnect twice

Autoconnect twice means? Can you please explain? You mean just put autoconnect(); Right?

Reni0013, 2020-11-07:
And also **wm.setEnableConfigPortal(true);** gives more control over WifiManager we can call it for starting Configportal but how can we set AP name and password of the AP which is created via **setEnableConfigPortal(true);** It is always shows name as ESP-and any random numbers and alphabets

tablatronix, 2020-11-07:
There is a bug in esp that some routers fail to connect on first try, it says wl disconnect, but the second time it works.

Setenableconfigportal is for adjusting autoconect.

But for on demand you can just startconfigportal whenever you wabt

Reni0013, 2020-11-08:
> But for on demand you can just startconfigportal whenever you wabt

I do the same in loop with if WL_DISCONNECTED condition insted of Trigger button HIGH & LOW. The thing is that under the loop it's starts configportal again and again. I thought maybe this is because of i am calling it in loop.

Reni0013, 2020-11-08:
> There is a bug in esp that some routers fail to connect on first try, it says wl disconnect, but the second time it works.

Oh okay got it. Is it bug of specially esp8266 or issue with all esp boards? Does esp32 also have this bug? If not then i purchase it and start learning on it. Thanks alot for your support @tablatronix

tablatronix, 2020-11-09:
Nobody knows, we just have found that some configurations do not reconnect.

If you can give me a rundown of exactly what behavior you are looking for in steps, and i see if we can make sure there is an example for it.

I just am not really clear on what you are doing.

tablatronix, 2020-11-09:
see #1137

Reni0013, 2020-11-12:
Flow of my code is very simple. 

I just want to connect my esp8266 with Wifi which is already done in void setup() with the help of this awesome library. I am using nonblocking example. 

So if esp restarts and doesn't get wifi ssid & password then only it starts config portal and then it gose to loop part ( Because it is non blocking example & this is how it should be)

But once if i restarts esp and if it gets wifi ssid and the password then it will not start any config portal and access point and gose to void loop and after code reach to the void loop and then if wifi disconnects then i have to put IF STATEMENT like 

```
If (WiFi.status() == WL_DISCONNECTED) {
// Then start config portal ( in case user want to change his network credentials instead of starting wifi with the already available credentials )
}
```

I know that wifi.begin() can help us to reconnect as shown in the many of the suggestion in the issue section of this repository but don't know somehow it is not working for me in Esp8266.

This is what i am doing but still it is being automatically reconnect to the wifi.

```
Void loop()
{
  if ( WiFi.status() == WL_CONNECTED ) {
   // Do something
   Serial.println("Connected to the internet");
  }
  else {
  Serial.println("Disconnected from internet");
  WiFi.begin();
  }
}

```

Still it is not automatically reconnect when wifi get back and online.

tablatronix, 2020-11-12:
That should totally work, esp8266 or esp32? 8266 will in fact automatically reconnect.

The only thing I can think is to turn on esp debugging and check serial log and see what it says when you call begin(), maybe its hung or your ap is still running.

Add WiFi.mode(WiFi_STA); to the start of setup

Reni0013, 2020-11-14:
> Add WiFi.mode(WiFi_STA); to the start of setup

Thanks a lot @tablatronix This seems to work excellent. Thank you very much for this awesome library and for such quick support.

tablatronix, 2020-11-14:
ok So I am assuming that was happening, the esp can default to softap mode, and it stays on, wm does not explicitly turn it off. I need to reenable that since its causing some confusion and issues like this.

Softap can cause issues with sta connections, something to do with the channel not being able to change I think.

tablatronix, 2020-11-14:
Are you using Github dev branch or the arduino alpha release?
I think I might have actually addresses this in dec

tablatronix, 2020-11-14:
I think I did fix this, so let me know if you are not using cutting edge
https://github.com/tzapu/WiFiManager/commit/e9d75a08b8ad59508a70441fad1cbba90ec0dcaf

JonasVorwerk, 2020-11-24:
Thanks everybody for all insights. 

Using the DEV/OnDemandConfigPortal example makes the use of new functions so much clearer since most options are presented in a clear and useful way. 

The latest version works perfectly on a Huzzah Feather ESP8266 in my setups. Thanks again!

tablatronix, 2020-11-24:
Yeah since I have not had the time to work on the docs nor help, I kind of user that as the main example by adding everything in it