tzapu/WiFiManager #1142: Hard reset required after autoConnect before fauxmoESP/Alexa can discover devices.

ddweber456, 2020-11-10:
Question: I thought this issue had been address or did I just miss something?

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ x] Development      2.0.3-alpha

**Esp8266/Esp32:**
- [ x] ESP8266    2.7.4
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

 LittleFS at version 0.1.0 
 ArduinoJson at version 5.13.5 
 ESP8266WiFi at version 1.0 
 WiFiManager at version 2.0.3-alpha 
 ESP8266WebServer at version 1.0 
 DNSServer at version 1.1.1 
 fauxmoESP at version 3.1.1 
 ESPAsyncTCP at version 1.2.2 

### Description

After successful autoConnect and the saving of the parameters, I have to do a hard reset before the ESP is in the correct configuration Alexa can discover devices with fauxmoESP.  I have WiFi.mode(WIFI_STA); in the setup().

### Settings in IDE

Module:  		 Wemos D1,    v1.4 Higher Bandwidth

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
14:55:47.642 -> SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:1.4.0rc2/BearSSL:5c771be
14:55:47.642 -> 
14:55:47.642 -> bcn 0
14:55:47.642 -> del if1
14:55:47.642 -> usl
14:55:47.642 -> mode : sta(98:f4:ab:da:62:f0)
14:55:47.642 -> add if0
14:55:47.689 -> Mounting FS...
14:55:47.689 -> wifi evt: 8
14:55:47.829 -> 
14:55:47.829 -> Mounted LittleFS file system
14:55:47.829 -> *WM: [3] allocating params bytes: 20
[... 713 lines cut ...]
14:57:27.234 -> 68.4.1,mask:255.255.255.0,gw:192.168.4.1)
14:57:27.234 -> bcn 100
14:57:28.217 -> *WM: [2] restoring usermode STA
14:57:28.217 -> bcn 0
14:57:28.217 -> del if1
14:57:28.217 -> pm open,type:2 0
14:57:28.217 -> mode : sta(98:f4:ab:da:62:f0)
14:57:28.217 -> *WM: [2] wifi status: WL_CONNECTED
14:57:28.217 -> *WM: [2] wifi mode: STA
14:57:28.217 -> *WM: [1] config portal exiting 
14:57:28.217 -> WiFi Connected.  yepee!!
14:57:28.217 -> 
14:57:28.217 -> Saving JSON Config
14:57:28.264 -> wifi evt: 8
14:57:28.264 -> please start sntp first !
14:57:28.264 -> Here is the SAVED JSON File:
14:57:28.264 -> {"smart_home_phrase":"Paper","double_press":"Crayon "}lfs_file_close: fd=0x3fff0a6c
14:57:28.264 -> please start sntp first !
14:57:28.264 -> 
14:57:28.264 -> Smart Home Phrases: 
14:57:28.264 -> Paper  Crayon 
14:57:40.599 -> :urn 265
14:57:40.599 -> :urd 265, 265, 0
14:57:40.694 -> :urn 274
```

tablatronix, 2020-11-11:
Make sure your esp is still in sta mode after wm exits. Not sure what else it could be

ddweber456, 2020-11-11:
What am I missing.  At the very end of the degug it looks to me like it's in STA mode.
```14:57:28.217 -> *WM: [2] restoring usermode STA
14:57:28.217 -> bcn 0
14:57:28.217 -> del if1
14:57:28.217 -> pm open,type:2 0
14:57:28.217 -> mode : sta(98:f4:ab:da:62:f0)
14:57:28.217 -> *WM: [2] wifi status: WL_CONNECTED
14:57:28.217 -> *WM: [2] wifi mode: STA
14:57:28.217 -> *WM: [1] config portal exiting 
14:57:28.217 -> WiFi Connected.  yepee!!
```
Where do I put the WiFi.mode(WIFI_STA); if not at the beginning of setup()  ?

tablatronix, 2020-11-11:
Yeah that should be good,

where in this log is the problem? How are you checking that your connectivity is working etc

ddweber456, 2020-11-11:
I really appreciate your help and time on this, Thanks.

OK I'm back on 0.15.0, I had this problem from the very start, so it's not anything new with the alpha release.  I added a couple of WiFi.printDiag(Serial); commands after the autoConnect to check the status.  Is there something else I should do?

Here are the serial print logs from the autoConnect and then from the hard reset.

autoConnect:

```22:37:29.519 -> *WM: AP IP address: 
22:37:29.519 -> *WM: 192.168.4.1
22:37:29.519 -> *WM: HTTP server started
22:38:32.462 -> *WM: Request redirected to captive portal
22:38:35.367 -> *WM: Request redirected to captive portal
22:38:35.367 -> *WM: Handle root
22:38:36.446 -> *WM: Request redirected to captive portal
22:38:36.493 -> *WM: Request redirected to captive portal
22:38:40.571 -> *WM: Scan done
22:38:40.571 -> *WM: DUP AP: Roosevelt
22:38:40.571 -> *WM: NachoLINK
22:38:40.571 -> *WM: -43
22:38:40.571 -> *WM: iConnect
22:38:40.571 -> *WM: -63
22:38:40.571 -> *WM: iConnect Guest
22:38:40.571 -> *WM: -63
22:38:40.571 -> *WM: DIRECT-02MFC-L8610CDW_BRbb03
22:38:40.571 -> *WM: -64
22:38:40.571 -> *WM: Dattebayo
22:38:40.571 -> *WM: -64
22:38:40.571 -> *WM: HOME-CAD2
22:38:40.618 -> *WM: -76
22:38:40.849 -> *WM: Sent config page
22:38:41.926 -> *WM: Request redirected to captive portal
22:38:41.973 -> *WM: Request redirected to captive portal
22:38:45.026 -> *WM: Request redirected to captive portal
22:39:20.201 -> *WM: WiFi save
22:39:20.201 -> *WM: Parameter
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: Parameter
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: Parameter
22:39:20.201 -> *WM: smart_home_phrase
22:39:20.201 -> *WM: Silver 
22:39:20.201 -> *WM: Parameter
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: Parameter
22:39:20.201 -> *WM: double_press
22:39:20.201 -> *WM: Gold 
22:39:20.201 -> *WM: Parameter
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: 
22:39:20.201 -> *WM: Sent wifi save page
22:39:21.186 -> *WM: Connecting to new AP
22:39:21.186 -> *WM: Connecting as wifi client...
22:39:21.233 -> *WM: Status:
22:39:21.233 -> *WM: 0
22:39:21.279 -> *WM: [ERROR] WiFi.begin res:
22:39:21.279 -> *WM: 6
22:39:36.979 -> *WM: Connection result: 
22:39:36.979 -> *WM: 4
22:39:38.995 -> *WM: Failed to connect.
22:39:38.995 -> *WM: Request redirected to captive portal
22:39:38.995 -> *WM: Request redirected to captive portal
22:39:48.801 -> *WM: Request redirected to captive portal
22:39:48.801 -> *WM: Request redirected to captive portal
22:40:02.969 -> *WM: WiFi save
22:40:02.969 -> *WM: Parameter
22:40:02.969 -> *WM: 
[... 100 lines cut ...]
22:48:03.413 -> SSID (9): NachoLINK
22:48:03.413 -> Passphrase (12): FreidA18556a
22:48:03.413 -> BSSID set: 0
22:48:03.413 -> 
22:48:03.413 -> 
22:48:03.413 -> After Fauxmo.createServer
22:48:03.413 -> 
22:48:03.413 -> Mode: STA
22:48:03.413 -> PHY mode: N
22:48:03.413 -> Channel: 9
22:48:03.413 -> AP id: 0
22:48:03.413 -> Status: 5
22:48:03.413 -> Auto connect: 1
22:48:03.413 -> SSID (9): NachoLINK
22:48:03.413 -> Passphrase (12): FreidA18556a
22:48:03.413 -> BSSID set: 0
22:48:03.413 -> 
22:48:03.413 -> WiFi Connected.  yepee!!
22:48:03.413 -> 
22:48:03.413 -> Smart Home Phrases: 
22:48:03.413 -> Silver   Gold 
```

In the autoConnect serial print log I'm getting " 22:40:04.036 -> *WM: [ERROR] WiFi.begin res: "
Let me know what you think.

tablatronix, 2020-11-11:
Thats odd, the first log says it connected, this one says it doesnt.

I wonder why, definitely a connection issue, which esp lib version are you using ?

ddweber456, 2020-11-11:
ESP 2.5.0
ESP8266 board mgr 2.7.4

tablatronix, 2020-11-11:
Not a fritzbox router is it ?

ddweber456, 2020-11-11:
Tp-Link.  Actually just install a new Tp-Link.  Haven't done anything with ESP yet.

tablatronix, 2020-11-11:
Try a full flash erase yet?

ddweber456, 2020-11-12:
I have.  The reset is needed on all the devices.  About 30 or 40 8266 so far.  It's become standard operating procedure to do a hard reset, but I know that's not right.

tablatronix, 2020-11-12:
It must have something to do with fauxmo, maybe mdns interferes with the udp ?

Have you tried a basic ping example ? Or some other network check ?

tablatronix, 2020-11-12:
Which fauxmo library are you using ?

ddweber456, 2020-11-12:
I'm using the current 3.1.1

Let me see if I can Ping it after the autoConnect.

tablatronix, 2020-11-12:
I mean which one, there seems to be a bunch xoseperez/fauxmoESP ?

ddweber456, 2020-11-12:
I'm sorry.  Support for the fauxmoESP and been transferred to Paul, @pvint [https://github.com/vintlabs/fauxmoESP](https://github.com/vintlabs/fauxmoESP)

tablatronix, 2020-11-12:
ok they should fix that in platform io

ddweber456, 2020-11-12:
Agreed. Maybe someone with the expertise could assist @pvint in making platform io happen.

pvint, 2020-11-12:
> Agreed. Maybe someone with the expertise could assist @pvint in making the platform io happen.

I hadn't thought of the lib in platformio, thanks for mentioning it. I took a look and I see how it works now. I'll try to get it and Arduino updated later today.

tablatronix, 2020-11-12:
I think the old one would have to be removed first to avoid confusion, unless the name changes. i assume you are maintaining a fork, and the source is abandoned?

pvint, 2020-11-12:
Xose, the original author, has handed it over to me to maintain. I've updated the library.json file on the original repo, which I believe should update it.

pvint, 2020-11-22:
I've looked at this a bit more and took some packet captures when it fails and when it works. For some reason the ESP8266 is sending a RST, ACK packet back to the Amazon echo . Everything seems the same up until that packet, and in the "good" scenario, it sens a SYN ACK and everything continues.

I have a bit more info here: https://github.com/vintlabs/fauxmoESP/issues/123#issuecomment-731847618

ddweber456, 2020-11-24:
Just curious if anyone has had a chance to investigate the findings that Paul, FauxmoESP, was able to see with the packet capture.  I'm still having the issue of the required hard reset and would really like to get that resolved if possible.  Let me know if there is anything I can do to help.

tablatronix, 2020-11-24:
you would have to eliminate WM and see if it can be reproduced with just esp examples, and then reach out to espressif, does not sound like sdk issue.

ddweber456, 2020-11-24:
It seems like I would have to do the esp.reset() as part of the autoConnect?  Unless I can conditionally esp.reset() coming out of autoConnect.  Ideas?

tablatronix, 2020-11-24:
You only have this issue when saving new wifi ?
You can use the saveconfig callback to flag a reset maybe?

ddweber456, 2020-11-24:
Yes, the only time I see the issue is when saving new WiFi credentials.

Several months back, I did see the comments regarding the corrupt flash, I was hopeful, but that didn't help.

tablatronix, 2020-11-24:
It hardly even does anything but maybe try adding a WiFi.reconnect() after ?

ddweber456, 2020-11-25:
The WiFi.reconnect() did not do the trick.  I think we maybe lookin in the wrong place.

Shawn, I'm not the most experienced coder, but it look like to me that autoConnect() call to connectWiFi() when saving credentials it does not return to the calling function cleanly.   Otherwise, I think I should be seeing the *WM: IP Address ... in the first serial print log below:
```
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.102
 WiFi Connected.  yepee!!
```

Serial print log when saving wifi credentials via autoConnect(): 

```
17:31:33.854 -> *WM: Sent wifi save page
17:31:34.839 -> *WM: Connecting to new AP
17:31:34.839 -> *WM: Connecting as wifi client...
17:31:34.839 -> *WM: Status:
17:31:34.884 -> *WM: 0
17:31:34.931 -> *WM: [ERROR] WiFi.begin res:
17:31:34.931 -> *WM: 6
17:31:39.284 -> *WM: Connection result: 
17:31:39.284 -> *WM: 3
17:31:39.331 -> Should save config
17:31:42.271 -> WiFi Connected.  yepee!!
17:31:42.271 -> 
17:31:42.271 -> Saving JSON Config
17:31:42.271 -> Here is the SAVED JSON File:
17:31:42.271 -> {"smart_home_phrase":"Black","double_press":"White "}
17:31:42.271 -> Smart Home Phrases: 
17:31:42.271 -> Black  White
 ```
Serial print log after doing the hard reset:

```
17:33:51.497 -> *WM: AutoConnect
17:33:51.543 -> *WM: Connecting as wifi client...
17:33:51.543 -> *WM: Status:
17:33:51.543 -> *WM: 6
17:33:51.543 -> *WM: Using last saved values, should be faster
17:33:55.091 -> *WM: Connection result: 
17:33:55.091 -> *WM: 3
17:33:55.091 -> *WM: IP Address:
17:33:55.091 -> *WM: 192.168.0.102
17:33:58.695 -> WiFi Connected.  yepee!!
17:33:58.695 -> 
17:33:58.695 -> Smart Home Phrases: 
17:33:58.695 -> Black  White 
```

tablatronix, 2020-11-25:
Yes your saves are failing,  your code is checking the return true? Not sure what is failing or if begin is failing, not sure why, have you tried a different esp a different lib version?

ddweber456, 2020-11-25:
I have only tried this on the 8266.  But on many 8266s.  All have the same issue and want a hard reset before they will communicate with Alexa.  After that they are very, very stable.  Multiple power re-boots, power outages, wifi going away for period of time.  No Issues.  

What lib version would you recommend?

ddweber456, 2020-11-25:
I just modified the example AutoConnect sketch from the 0.15.0 WiFiManager lib.  Modified it just enough to blink the on-board LED and I'm getting the same serial print log as before:

Serial print log from autoConnect() saving wifi credentials:
```
21:50:11.333 -> *WM: WiFi save
21:50:11.333 -> *WM: Sent wifi save page
21:50:12.359 -> *WM: Connecting to new AP
21:50:12.359 -> *WM: Connecting as wifi client...
21:50:12.359 -> *WM: Status:
21:50:12.359 -> *WM: 0
21:50:12.405 -> *WM: [ERROR] WiFi.begin res:
21:50:12.405 -> *WM: 6
21:50:16.704 -> *WM: Connection result: 
21:50:16.704 -> *WM: 3
21:50:16.752 -> connected...yeey :)
21:50:16.752 -> IP Address: 192.168.0.102
21:50:16.752 -> *WM: freeing allocated params!
```

After ESP8266 hard reset:

```
21:55:19.368 -> *WM: AutoConnect
21:55:19.368 -> *WM: Connecting as wifi client...
21:55:19.368 -> *WM: Status:
21:55:19.368 -> *WM: 6
21:55:19.368 -> *WM: Using last saved values, should be faster
21:55:22.958 -> *WM: Connection result: 
21:55:22.958 -> *WM: 3
21:55:22.958 -> *WM: IP Address:
21:55:22.958 -> *WM: 192.168.0.102
21:55:23.004 -> connected...yeey :)
21:55:23.004 -> IP Address: 192.168.0.102
21:55:23.004 -> *WM: freeing allocated params!
```

Here is the code:
```
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

unsigned long lastLEDMillis = 0;      // will store the last time the LED was updated
unsigned long interval = 3000;        // place holder, will be updated in the LED blink

int ledState=HIGH;
int blinkTime =             1;       // blink on duration

void setup() {
    // put your setup code here, to run once:
    
    // LEDs and PINs used for logic
    pinMode(2, OUTPUT);                // on-board LED
    digitalWrite(2, LOW);              // LOW = LED on

    Serial.begin(115200);

IPAddress(255,255,255,0));

    wifiManager.autoConnect("AutoConnectAP", "gaslights");
    
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
}

void loop() {
    // put your main code here, to run repeatedly:

    unsigned long currentMillis = millis();

    // >>>  Simple code to flash the on-board LED every 3 seconds, no delay
    if (currentMillis - lastLEDMillis > interval) {

    lastLEDMillis = currentMillis;
    ledState = !ledState;

    // if the LED is off, turn it on and vice-versa:
    if (ledState == HIGH)
    {
      digitalWrite(2, LOW);
      interval = blinkTime;
    }
    else
    {
      digitalWrite(2, HIGH);
      interval = 3000;
    }
  }
    
}
```
The {ERROR} *WM : WiFi.begin res:  that we're seeing in the first Serial print log is un-expected, I'm starting to think this is NOT the problem.  If everyone else is able to use the 8266 as a STA right after the autoConnect() without doing a hard reset, then something else is going on.  I'm going to share this with @pvint - vintlabs/FauxmoESP.  See if he can shed his intelligence on the subject.  Thanks Paul for your help.

David

tablatronix, 2020-11-25:
Ok so I am going to say use the dev/alpha version, there was some fixes for this kind of stuff added and better logging. I cannot say for sure but there was some clean connect stuff I worked out to fix odd issues like this

tablatronix, 2020-11-25:
There is a clean connect, let me find it

Try these

  // set wifi connect retries
  // wm.setConnectRetries(2);

  // This is sometimes necessary, it is still unknown when and why this is needed but it may solve some race condition or bug in esp SDK/lib
  // wm.setCleanConnect(true); // disconnect before connect, clean connect

pvint, 2020-11-25:
I just tested with the development branch, both with and without `.setCleanConnect(true)` and the issue remains. I'll try to look a bit more later on.

ddweber456, 2020-11-29:
@pvint were you able to do any additional testing?  Could this be because FauxmoESP is using UDP?

Even though we are seeing an ERROR from WM: connectWiFi() returning to the calling WM: autoConnect() (see comment of 5 days ago https://github.com/tzapu/WiFiManager/issues/1142#issuecomment-733396464 ) it appears most everyone else is able use the ESP as a STA without doing a hard reset.   It not just my code that is producing this WM: ERROR, it is also happening in the example sketches WM 0.15.0.  https://github.com/tzapu/WiFiManager/issues/1142#issuecomment-733456344

tablatronix, 2020-11-30:
It could be router specific. I will test this on some of my boards when I have time

ddweber456, 2020-12-01:
> It could be router specific. I will test this on some of my boards when I have time.

I've taken the unit to 8 to 10 client locations, there's no difference.  I've always had to do a hard reset. 

Ant progress in finding a fix?  Is there anything I can help you with?
david

ddweber456, 2020-12-03:
@tablatronix Hey, just checking in to see if you had made any progress.  I hope the silence is not an indication of bad news.

tablatronix, 2020-12-03:
Sorry I will try to test here tomorrow

pvint, 2020-12-04:
It would be interesting to try to reproduce the issue in a simpler manner - get rid of all the fauxmo stuff and just insert a `sendUDPPacket(data, dst, port)` sort of thing and see if we can narrow it down. (I made up that function name...)

I'll try to have a look in the morning (in 12 hours or so)

tablatronix, 2020-12-04:
I am not sure what to expect here, it seems to work, not sure how I test it, alexa needs wemo skill ? My homeassistant isnt finding a new device but I do have these `[FAUXMO] Responding to M-SEARCH request` and
```
[FAUXMO] Handling devicetype request
[FAUXMO] Client #0 disconnected
[FAUXMO] Client #0 connected
[FAUXMO] Handling list request
[FAUXMO] Sending device info for "yellow lamp", uniqueID = "1a6083b4d603"
[FAUXMO] Sending device info for "green lamp", uniqueID = "ea4c8bf1b2e4"
```

tablatronix, 2020-12-04:
fauxmo runs a webserver, I would suspect that before anything, make sure wm is destroyed before starting the fauxmo one

ddweber456, 2020-12-04:
That sounds good...sooo how do you stop a server?

tablatronix, 2020-12-04:
I have seen issues with running another webserver even though wm stops it
Maybe since we are using a shared pointer for the webserver now, or maybe stop() is not enough

```
bool WiFiManager::shutdownConfigPortal(){
  if(webPortalActive) return false;

  //DNS handler
  dnsServer->processNextRequest();
  //HTTP handler
  server->handleClient();

  // @todo what is the proper way to shutdown and free the server up
  server->stop();
  server.reset();
  dnsServer->stop(); //  free heap ?
  dnsServer.reset();
```

ddweber456, 2020-12-04:
What if I did all the WiFiManager stuff, including the WiFiManager MW; in it own function, called from Setup().  When focus is lost, that should destroy WM and the Server. Right?

tablatronix, 2020-12-04:
It should, if you take it out of global scope.

ddweber456, 2020-12-07:
The ESP SDK is very persistent in keeping the server, once established, connected and running.  Moving the creation of the server out of scope in a separate function did not work.  Server remained regardless of scope.

ddweber456, 2020-12-08:
@tablatronix How do you change the port from 80 to something else?  This may be the problem with having to do a hard reset before FauxmoESP can use port80 for UDP.  #148

ddweber456, 2020-12-08:
@tablatronix Could it be that somehow WM is effecting Port 80, blocking FauxmoESP from being able to use 80 to receive UDP messages?

tablatronix, 2020-12-08:
Yes, that is what I think, if you use an alt port to test it will be evident.
I meant to say that the other day

    // set port of webserver, 80
    void          setHttpPort(uint16_t port);

ddweber456, 2020-12-08:
Can you be a little more specific how and where to implement is for testing?  thanks

tablatronix, 2020-12-08:
wm.setHttpPort(); you can set the port in the dev branch

pvint, 2020-12-08:
Just to interject - I think this is moot... port 80 is for TCP on the webserver. When we see the issue with UDP on fauxmoESP it's on totally different ports.

I got looking at this more yesterday, but I'm still a bit stumped. Might look more tomorrow.

tablatronix, 2020-12-08:
I saw port 80 in the example, so did not want to assume if it was the default or not, was not sure what v3 was or if I needed which for alexa, also was not sure if the webserver was necessary

tablatronix, 2020-12-09:
mdns uses udp no ?

tablatronix, 2020-12-09:
well that didn't work

tried adding      MDNS.end(); no change

tablatronix, 2020-12-09:
I can confirm using non 80 port on wm webserver works, so not sure if you are using udp on port 80, but it seems to not matter it phy is using it its using it. I will ask around and look at the esp sdk and see if we can close it better or proper of if its a bug.

Maybe when add async compate we can just use the same server as we can now.

ddweber456, 2020-12-10:
> wm.setHttpPort(); you can set the port in the dev branch

Preliminary tests show that using the above with a port other than 80, Alexa is able to Discover the devices without doing a hard reset.

tablatronix, 2020-12-10:
I have debugged everything I could tcp_close does not return any errors, did not see any asynctcp begin bind errors

ddweber456, 2020-12-10:
Unfortunately, I could not get the fauxmo.server(80) and Alexa Discover Devices it to work together without doing the hard reset after saving WiFi credentials via autoConnect().

Maybe? Someone else has a solution?
 
Like you, I can't spend more time of this issue right now.  A better solution if just an OTA away.  
Shawn and Paul, Thank you very much for help.

Here is the minimum code for WifiManager and FauxmoESP:
```
#include <Arduino.h>
#ifdef ESP32
    #include <WiFi.h>
#else
    #include <ESP8266WiFi.h>
#endif

#include <fauxmoESP.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

// Rename the credentials.sample.h file to credentials.h and 
// edit it according to your router configuration
//#include "credentials.h"

fauxmoESP fauxmo;

#define SERIAL_BAUDRATE     115200

#define DEVICE_NAME "Marco Polo"

// -----------------------------------------------------------------------------
// Wifi
// -----------------------------------------------------------------------------

void wifiSetup() {

    WiFiManager wifiManager;
    //reset saved settings
    //wifiManager.resetSettings();

        
    wifiManager.autoConnect("AutoConnectAP");

         
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeehaa :)");    
    
    Serial.printf("\n[WIFI] STATION Mode, SSID: %s, IP address: %s\n", WiFi.SSID().c_str(), WiFi.localIP().toString().c_str());
}

// -----------------------------------------------------------------------------
// SETUP
// -----------------------------------------------------------------------------

void setup() {

    // Init serial port and clean garbage
    Serial.begin(SERIAL_BAUDRATE);
    Serial.println();
    Serial.println();

    // Wifi
    wifiSetup();

    fauxmo.createServer(true); // not needed, this is the default value
    fauxmo.setPort(80); // This is required for gen3 devices

    fauxmo.enable(true);

    fauxmo.addDevice(DEVICE_NAME);     
}

// -----------------------------------------------------------------------------
// Main Loop
// -----------------------------------------------------------------------------

void loop() {

    // fauxmoESP uses an async TCP server but a sync UDP server
    // Therefore, we have to manually poll for UDP packets
    fauxmo.handle();
}
```

[21 more replies]