tzapu/WiFiManager #1143: After connection via Wifimanager how do I pass the SSID and Pass for use with Thinger.io

waketech03, 2020-11-15:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ X] Master
- [ ] Development

**Esp8266/Esp32:**
- [ X] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ X] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Need to pass SSID and Password to (thing.add_wifi(SSID, SSID_PASSWORD);) how do I retrieve it and pass it over ?

### Settings in IDE

Module:  			Wemos D1

#include<Wire.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ThingSpeak.h>
#include <ThingerWifi.h>

 
// BME280 I2C address is 0x76(108)
#define Addr 0x76

#define USERNAME "TBA"           //thingerio username
#define DEVICE_ID "TBA"           //thingerio device name
#define DEVICE_CREDENTIAL "TBA"   //thingerio pass
ThingerWifi thing(USERNAME, DEVICE_ID, DEVICE_CREDENTIAL); 

void setup()
{
Serial.begin(115200);
WiFiManager wifiManager;

wifiManager.autoConnect("AutoConnectAP");
Serial.println("connected...yeey :)");
Wire.begin();

thing.add_wifi(SSID, SSID_PASSWORD);

tablatronix, 2020-11-15:
You should check to see if there is a way to use the non wifi classes, it does not require them since you are already handling wifi..

waketech03, 2020-11-15:
ok I will have to research this as I am a noob, so I guess the answer to my original question is there is not a way ?