tzapu/WiFiManager #1144: Unable to access web page...!!!

joy-sen, 2020-11-16:
Hi,

The intension of the below program is to connect  a WIFI network dynamically and to access a simple web page.

**Issue :-  Able to connect a WIFI network successfully, but not able to access any web page though the ESP32 module is connected in the STA Mode.** 

can you please help me or correct me if i am making any mistake in program. Your quick response will be highly appreciable.  

**The Arduino program is below:**

#include <WiFi.h>
#include <WebServer.h>
#include <WiFiManager.h>
#include <DNSServer.h>

WebServer Server;

void handle_root() 
{
  String HTML = "<!DOCTYPE html>\
<html>\
  <body>\
    <h1>Welcome</h1>\
    <p>Your first Iot Project made with ESP32</p>\
    <p>&#128522;</p>\
  </body>\
</html>";

  Server.send(200, "text/html", HTML);
}

void setup()
{
  Serial.begin(115200);
  WiFiManager wm;

  /************************************ Configuration of Dynamic IP Generation ********************************/

  /*if(wm.autoConnect("ESP32AP","12345678"))
  {
    Serial.println("WiFi Connected...");
    Serial.println("local ip");
    Serial.println(WiFi.localIP());

    Server.on("/", handle_root);
    Server.begin();
    Serial.println("HTTP server started");
  }
  else
  {
    Serial.println("Failed to connect");
  }*/

  /************************************************************************************************************/

  /********************************* Configuration for Static IP Generation ***********************************/

  IPAddress _ip = IPAddress(192, 168, 1, 250);
  IPAddress _gw = IPAddress(192, 168, 1, 1);
  IPAddress _sn = IPAddress(255, 255, 255, 0);

  wm.setSTAStaticIPConfig(_ip, _gw, _sn);

  if (!wm.autoConnect("ESP32AP", "12345678")) 
  {
      Serial.println("failed to connect, we should reset as see if it connects");
  }
  else
  {
      
    Serial.println("WiFi Connected...");
    Serial.println("local ip");
    Serial.println(WiFi.localIP());

    Server.on("/", handle_root);
    Server.begin();
    Serial.println("HTTP server started");
  }
  /***********************************************************************************************************/

}

void loop()
{
  Server.handleClient();
}

**Here is the below Serial Monitor output.**

*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP: 192.168.1.250
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom STA IP/GW/Subnet 
*WM: [1] STA IP set: 192.168.1.250
*WM: [1] No wifi save required, skipping 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] 12345678 
*WM: [3] WiFi station disconnect 
*WM: [3] WiFi_enableSTA enable
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  ESP32AP
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             ESP32AP
*WM: [1] password:         12345678
*WM: [1] ssid_len:         7
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC completed in 2501 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] WiFi Scan completed in 2502 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
dhcps: send_offer>>udp_sendto result 0
*WM: [2] NUM CLIENTS: 1 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> c.whatsapp.net 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> android.bugly.qq.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> portal.fb.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] WiFi Scan ASYNC completed in 2701 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] WiFi Scan completed in 2702 ms
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] Scan is cached 10213 ms ago
*WM: [3] -> 192.168.4.1 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 20415 ms ago
*WM: [1] 5 networks found
*WM: [2] AP:  -51 Hathway_JOY KUMAR
*WM: [2] AP:  -83 ACT_2.4G
*WM: [2] AP:  -90 dlink-842E
*WM: [2] AP:  -95 CISCO 36.G
*WM: [2] AP:  -96 veenas
*WM: [3] _staShowStaticFields 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [3] Sent config page 
*WM: [2] NUM CLIENTS: 1 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [3] static ip: 192.168.1.250
*WM: [3] static gateway: 192.168.1.1
*WM: [3] static netmask: 255.255.255.0
*WM: [3] Sent wifi save page 
*WM: [2] processing save 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP: 192.168.1.250
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom STA IP/GW/Subnet 
*WM: [1] STA IP set: 192.168.1.250
*WM: [1] CONNECTED:
*WM: [1] Connecting to NEW AP: Hathway_JOY KUMAR
*WM: [3] Using Password: 8210683020
*WM: [3] WiFi_enableSTA enable
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.1.250 
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
WiFi Connected...
local ip[ERROR] wm not ready

192.168.1.250
HTTP server started
*WM: [3] unloading 

Thank You.

tablatronix, 2020-11-16:
probably because wm already runs a webserver

joy-sen, 2020-11-16:
Sir, 
Is there a way to soft reset the board? 
Because i observed that after pressing the rest button available in board, i can able to access the web page.

tablatronix, 2020-11-16:
esp.reset

but you can either destroy the wm instance, or just use the wm webserver instance. wm.server is a shared pointer

see the dev/ example

void bindServerCallback(){
  wm.server->on("/custom",handleRoute);
  // wm.server->on("/info",handleRoute); // you can override wm!
}

joy-sen, 2020-11-17:
Thanks.
The way you suggested i can able to access the webpage only when the board is in AP mode. But as per my requirement i want to access it when the ESP32 module is in STA mode.
 
Is there a way to access the webpage when the board will connected to WIFI network and work as a STA?
Your suggestion will help me to move ahead in my project.

tablatronix, 2020-11-17:
Yes see the ondemand/non blocking webportal exampples.

The webportal can now be started by itself without configportal

joy-sen, 2020-11-18:
First of all thank you so much for giving your valuable time and guidance.
With the help of webportal, webserver can be accessible now. But it is possible only when i am resetting the module once.

Is it necessary or mandatory to reset the ESP32 module to access the webportal using local IP address...? Though the module is already in STA mode.

If reset is not required, i request you to have a look into the program (wrote just for trial) mentioned below. May be i am missing something and you can correct me.

I tried to soft reset the board using ESP.restart(), but it is running infinite times and not allowing to execute the remaining source code.  

#include <WiFiManager.h>
WiFiManager wm;

void handle_root() 
{
  String HTML = "<!DOCTYPE html>\
<html>\
  <body>\
    <h1>Welcome</h1>\
    <p>Your first Iot Project made with ESP32</p>\
    <p>&#128522;</p>\
  </body>\
</html>";

  wm.server->send(200, "text/html", HTML);
}

void handleRoute(){
  Serial.println("[HTTP] handle route");
  wm.server->send(200, "text/plain", "hello from user code");
}

void bindServerCallback()
{
  wm.server->on("/webpage",handle_root);
  wm.server->on("/custom",handleRoute);
}

void setup() 
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);

  IPAddress _ip = IPAddress(192, 168, 1, 120);
  IPAddress _gw = IPAddress(192, 168, 1, 1);
  IPAddress _sn = IPAddress(255, 255, 255, 0);

  wm.setSTAStaticIPConfig(_ip, _gw, _sn);
  wm.setWebServerCallback(bindServerCallback);
  //wm.setHostname("WIFIMANAGERTESTING");
  

  if (!wm.autoConnect("ESP32AP", "12345678")) 
  {
    Serial.println("failed to connect, we should reset as see if it connects");
    delay(3000);
    //ESP.reset();
    delay(5000);
  }
  else
  {
   //if you get here you have connected to the WiFi
   Serial.println("WiFi Connected...");
   Serial.println("local ip");
   Serial.println(WiFi.localIP());
   //wm.stopConfigPortal();
   wm.startWebPortal();   
  }
}

void loop() 
{
 wm.process();
}

tablatronix, 2020-11-18:
hmm it should work, let me test.
It sounds like you are having a connection issue to wifi

petermagdy94, 2021-01-02:
i finally find a way for that and hope it's correct and works with you.

**Problem i faced:**
when i enter the wifi credentials in wifi manager and after the esp32 is connected to the wifi network, then i can't access my custom webserver unless i make a manual hard reset.

**Solution:**
after wifimanager is finished, call server.close() then server.begin() for my webserver object.

**Code:**
```
#include <WebServer.h>
WebServer server(80);
void setup()
{
    /* start wifimanager autoConnect or startConfigPortal and
        enter the network configuration in the webpage */
    .
    .
    .
    .
    /* add handlers for my custom server routes */
    server.close();
    server.begin();
}
void loop()
{
    server.handleClient();
    .
    .
    .
}
```

tablatronix, 2021-01-03:
#1065

tablatronix, 2021-01-03:
#1142

petermagdy94, 2021-01-03:
ok i tried again today and **it didn't work** :(
so i make a wifi event on SYSTEM_EVENT_AP_STOP and in the callback function i made a soft reset ESP.reset();
it's working till now using esp32 and development branch.

tablatronix, 2021-01-04:
Yeah, I am investigating why the ports are not closing, pretty sure this is a esp bug

microicRI, 2022-02-16:
you should set dns1 and dns 2 IP address, looks like both are set to 0.0.0.0

after I set IPAddress dns1(192, 168, 1, 1);  //Google dns ,  all works ok

closed 2023-09-08 by tablatronix