tzapu/WiFiManager #1145: Unable to retrieve saved password from WifiManager while working with ESP32 Cam from Ai Thinker

TheBluePhoenix10, 2020-11-19:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##
I can retrieve the password but cannot retrieve the SSID of my wifi
## POST SERIAL OUTPUT !
**Serial output for wifi connection:**
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
SPIFFS mounted successfully
Connecting to WiFi*WM: [1] resetSettings 
*WM: [3] WiFi station enable 
*WM: [1] SETTINGS ERASED 
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] 12345678 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AIB-Cam
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             AIB-Cam
*WM: [1] password:         12345678
*WM: [1] ssid_len:         7
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2505 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
*WM: [2] NUM CLIENTS: 0 
dhcps: send_offer>>udp_sendto result 0
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] WiFi Scan completed in 2103 ms
*WM: [3] -> globalapi.ad.xiaomi.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> clients3.google.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] Scan is cached 82 ms ago
*WM: [3] -> connect.rom.miui.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] Scan is cached 3048 ms ago
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> 192.168.4.1 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Wifi 
*WM: [2] Scan is cached 6461 ms ago
*WM: [1] 2 networks found
*WM: [2] AP:  -50 TheBluePhoenix
*WM: [2] AP:  -84 Angel
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [3] Sent config page 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> resolver.msg.xiaomi.net 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] Scan is cached 9112 ms ago
*WM: [3] -> connectivitycheck.gstatic.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] NUM CLIENTS: 1 
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [3] Sent wifi save page 
*WM: [2] processing save 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] CONNECTED:
*WM: [1] Connecting to NEW AP: TheBluePhoenix
*WM: [3] Using Password: TheBluePhoenix@7666133300
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 192.168.29.184 
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
connected :)
Stored SSID: TheBluePhoenix
Stored passphrase: TheBluePhoenix@76
WiFi connected

**Serial Output while retrieving wifi information saved above**
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
SPIFFS mounted successfully
Connecting to WiFi*WM: [1] Reading SSID 
*WM: [1] SSID:  
*WM: [1]  
*WM: [1] Reading Password 
*WM: [1] Password: TheBluePhoenix@76
Stored SSID: *WM: [1] Reading SSID 
*WM: [1] SSID:  
*WM: [1]  

Stored passphrase: TheBluePhoenix@76
.................

#### Hardware ESP32Cam from Ai Thinker
**WiFimanager Branch/Release:**
- Master

**Esp8266/Esp32:**
- [ ] ESP32

**Wifi manager code block**
pinMode(wifi,INPUT);
  Serial.begin(115200);
  
  if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    ESP.restart();
  }
  else {
    delay(500);
    Serial.println("SPIFFS mounted successfully");
  }
  //  Configure and connect to WiFi
  IPAddress ip;
 
  WiFi.mode(WIFI_STA);
  //WiFi.begin(SSID1, PWD1);
  Serial.print("Connecting to WiFi");
  
  WiFiManager wifiManager;
  if(digitalRead(wifi) == HIGH){
    wifiManager.resetSettings();
    wifiManager.autoConnect("AIB-Cam","12345678");
    Serial.println("connected :)");
  } 
  String ssid = wifiManager.getSSID();
  String pass = wifiManager.getPassword();
  Serial.print("Stored SSID: ");
  Serial.println(wifiManager.getSSID());
  Serial.print("Stored passphrase: ");
  Serial.println(wifiManager.getPassword());
  
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(F("."));
  }
  
  ip = WiFi.localIP();
  Serial.println(F("WiFi connected"));
  Serial.println("");
  Serial.print("Stream Link: http://");
  Serial.print(ip);
  Serial.println("/mjpeg/1");

tablatronix, 2020-11-19:
There is no such thing what are you using? Its not this library

 String ssid = wifiManager.getSSID();
  String pass = wifiManager.getPassword();

TheBluePhoenix10, 2020-11-19:
There is, its just commented in the WiFiManager.cpp on line 2911-2922

Regards,
Dishant Shah

TheBluePhoenix10, 2020-11-19:
There is, its just commented in the WiFiManager.cpp on line 2911-2922

here are the two blocks:

```
String WiFiManager::getSSID() {
  if (_ssid == "") {
    DEBUG_WM(F("Reading SSID"));
    _ssid = WiFi.SSID();
    DEBUG_WM(F("SSID: "));
    DEBUG_WM(_ssid);
  }
  return _ssid;
  }

  String WiFiManager::getPassword() {
  if (_pass == "") {
    DEBUG_WM(F("Reading Password"));
    _pass = WiFi.psk();
    DEBUG_WM("Password: " + _pass);
    //DEBUG_WM(_pass);
  }
  return _pass;
  }

```

tablatronix, 2020-11-19:
ok well master branch does not support esp32 so I have no idea how you are using this code..

The dev branch / alpha version does support esp32.

Also the esp32 does not provide this info if the wifi has not been initialized

There are wrappers for this in dev

```C++
// HELPERS

/**
 * getWiFiSSID
 * @since $dev
 * @param bool persistent
 * @return String
 */
String WiFiManager::getWiFiSSID(bool persistent){
  return WiFi_SSID(persistent);
}

/**
 * getWiFiPass
 * @since $dev
 * @param bool persistent
 * @return String
 */
String WiFiManager::getWiFiPass(bool persistent){
  return WiFi_psk(persistent);
} 
```