tzapu/WiFiManager #1148: Can't reconnect after disconnect

Primus007, 2020-11-20:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos
Arduino IDE 1.8.12 

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ x] Development

**Esp8266/Esp32:**
- [ x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [x] v2.7.4
- [ ] staging (master/dev)

### Description

I use your WifiManager to connect to my home Network. It works fine. But after i get some infos from the internet i disconnect the wifi in my Code. it should reconnect if i press a button. but if i do that, always the AP Config mode fires up and i get the error: [ERROR] wifi begin failed

### Settings in IDE

Module:  			Wemos D1 & R2 mini

Additional libraries: 

### Sketch
I disconnect with this code:

   WiFi.persistent(false);
    WiFi.disconnect();
    WiFi.persistent(true);

and i try to reconnect with this code:
wm.autoConnect("Clock_Config");

### Debug Messages

```
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: pLhxxxxx
*WM: [3] Using Password: *****
*WM: [3] WiFi_enableSTA enable
*WM: [3] enableSTA PERSISTENT ON 
*WM: [3] Mode after delay: STA 
*WM: [0] [ERROR] wifi begin failed 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station disconnect 
*WM: [3] WiFi_enableSTA enable
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  Clock_Config
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             Clock_Config
*WM: [1] password:         
*WM: [1] ssid_len:         15
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
Entered config mode
192.168.4.1
MS_Clock_Config
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan completed in 2265 ms
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0 
```

tablatronix, 2020-11-22:
*WM: [0] [ERROR] wifi begin failed 
hmm this is something in the esp lib failing

Can you try using wm.disconnect() instead ?

Primus007, 2020-11-24:
wm.disconnect()  works fine. Thank you.

tablatronix, 2020-11-24:
Thanks, something must be left hanging with the esp method. Glad it helped

Primus007, 2021-02-12:
I have a similar problem.

Here my code:

```
#include <WiFiManager.h> 
WiFiManager wm;
void setup() {
  Serial.begin(115200);
  startWiFi();
}
void loop() {
   delay(3000);
   Serial.println("Wifi Disconnect");
   wm.disconnect();
   delay(3000);
   int conn_result = WiFi.waitForConnectResult(10000);    
   Serial.print("Connection Result: ");
   Serial.println(conn_result);
}
// -------------------------------------------------------------------------
//                               startWiFi
// -------------------------------------------------------------------------
   //for reference:
    //typedef enum {
    //    WL_NO_SHIELD = 255,   // for compatibility with WiFi Shield library
    //    WL_IDLE_STATUS = 0,
    //    WL_NO_SSID_AVAIL = 1,
    //    WL_SCAN_COMPLETED = 2,
    //    WL_CONNECTED = 3,
    //    WL_CONNECT_FAILED = 4,
    //    WL_CONNECTION_LOST = 5,
    //    WL_DISCONNECTED = 6
    //} wl_status_t;
void startWiFi() {
    WiFi.mode(WIFI_STA); //  Station Mode 
    WiFi.setSleepMode(WIFI_NONE_SLEEP);
    Serial.println("");
    //wm.resetSettings(); //reset wifi settings
    Serial.print("WiFi saved?: ");
    Serial.println((String)wm.getWiFiIsSaved() ? "YES" : "NO");
    if (!wm.getWiFiIsSaved()) {
        Serial.println("Starte Konfigurationsportal!");
        if (!wm.startConfigPortal("MS_Clock_Config")) {
            Serial.println("failed to connect and hit timeout");
            delay(3000);
            ESP.reset();
            delay(5000);
        }
    }
    if (!WiFi.isConnected()) { // if not connected then connect
        Serial.println("SSID: " + WiFi.SSID());
        int conn_result = WiFi.waitForConnectResult(10000); // connect to wlan, try it max 10 Seconds
        Serial.print("conn_result: ");
        Serial.println(conn_result, DEC);
        if (conn_result == 1) {
            Serial.println("FEHLER: Wlan nicht erreichbar! SSID nicht gefunden.");
        }
        if (conn_result == -1) {
            Serial.println("FEHLER: Wlan nicht erreichbar! Timeout.");
        }
        if (conn_result != -1 && conn_result != 3 && conn_result != 1 ) {
            Serial.println("FEHLER: Wlan nicht erreichbar!");
        }
    }
    if (WiFi.isConnected()) {
        Serial.println("Wifi is connected!");
    }
    else {
        Serial.println("Wifi is not connected!");
    }
    Serial.println("");
}
```

Here is the Result:
WiFi saved?: YES
SSID: hotspot_test
conn_result: 3
Wifi is connected!

Wifi Disconnect
*WM: [1] Disconnecting 
*WM: [3] WiFi station disconnect 
Connection Result: 0
Wifi Disconnect
*WM: [2] Disconnecting: Not connected 
Connection Result: 0
Wifi Disconnect
*WM: [2] Disconnecting: Not connected 
Connection Result: 0

The first connect works fine conn_result: 3
Then i disconnect and after a delay i try to connect with the same method as before and the connect result is always 0

Why? what is wrong?