tzapu/WiFiManager #1149: AP & STA MODE TOGETHER

Abedi98, 2020-11-20:
hi

How to access ap mode when esp connected to wifi
I want to access AP mode forever.

thanks

Reni0013, 2020-11-24:
Developers already have an example called **nonblocking** which may fulfil you recruitment. Thanks to @tablatronix & @tzapu for such awesome lib.
.

tablatronix, 2020-11-24:
I would run the webserver only on sta mode, running sta+ap is not very stable

ldijkman, 2020-12-07:
is there  webserver example for AP 
and or Ap+STA 
for my tft touch thermostat
https://www.youtube.com/watch?v=qNYT64CaLEI
https://oshwlab.com/l.dijkman/esp32-dev-kit-38-pin-to-spi-touch-tft

Abedi98, 2021-03-05:
> I would run the webserver only on sta mode, running sta+ap is not very stable

```
WiFi.mode(WIFI_AP_STA);
WiFi.enableAP(true);
    Serial.print("Setting soft-AP configuration ... ");
    Serial.println(WiFi.softAPConfig(IPAddress(172,217,1,99), IPAddress(172,217,1,99), IPAddress(255,255,255,0)) ? "Ready" : "Failed!");

    Serial.print("Setting soft-AP ... ");
    Serial.println(WiFi.softAP("MYAP","@app99") ? "Ready" : "Failed!");
    WiFi.begin();
```

When I connect to a WiFi via wifimanager, I run this code in the setup and an access point is permanently active, but after 10 to 15 minutes the wifi is disconnected and it tries to connect to the WiFi again.

How can this problem be solved?

Abedi98, 2021-03-28:
Is there no solution to this problem?

tablatronix, 2021-03-29:
What problem? This is easily doable

tablatronix, 2021-03-29:
You did not provide any info in why your wifi disconnects after 10 minutes

Abedi98, 2021-03-30:
```
void setup()
{

  Serial.begin(9600);

  if (WiFi.SSID() == "") {

    WiFi.mode(WIFI_AP);

    configtone();

    delay(1000);
    initialConfig = true;

  } else {
    if (mobilemode_flag == 1) {
      WiFi.mode(WIFI_AP_STA);
      Serial.print("Setting soft-AP configuration ... ");
      Serial.println(WiFi.softAPConfig(IPAddress(192, 168, 4, 202), IPAddress(192, 168, 4, 202), IPAddress(255, 255, 255, 0)) ? "Ready" : "Failed!");
      Serial.print("Setting soft-AP ... ");
      Serial.println(WiFi.softAP("LOCALHOSTMob") ? "Ready" : "Failed!");
      //      WiFi.begin();
    } else {
      WiFi.mode(WIFI_STA);
    }

    WiFi.hostname("LOCALHOST_WIFI");
    int connRes = WiFi.waitForConnectResult();

    server.begin();
    

    wifiAPPrint();

  

  }

}

void loop()
{

  if (initialConfig) {
    WiFiManagerParameter custom_api("api", "API KEY", token, 65);

    wm.setSaveConfigCallback(saveConfigCallback);
    wm.addParameter(&custom_api);
    wm.setAPStaticIPConfig(IPAddress(192, 168, 4, 1), IPAddress(192, 168, 4, 1), IPAddress(255, 255, 255, 0));
    if (!wm.startConfigPortal("LOCALHOST")) {
      Serial.println("Not connected to WiFi but continuing anyway.");
    } else {
      strcpy(token, custom_api.getValue());
      if (shouldSaveConfig) {
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.createObject();
        json["api"] = token;
        File configFile = SPIFFS.open("/config.json", "w");
        if (!configFile) {
          Serial.println("failed to open config file for writing");
        }

        json.printTo(Serial);
        json.printTo(configFile);
        configFile.close();
        //end save
      }
      Serial.println("web server started:" + WiFi.localIP().toString());
      wdt_reset(); ESP.reset(); ESP.restart(); while (1)wdt_reset();

    }
  } else {
    server.handleClient();
  }

}
```
In the setup method, if the value of mobilemode_flag is equal to 1, Wi-Fi will run in AP_STA mode.
In this case, the station is active and an access point called LOCALHOSTMob is created next to the station and disconnects the station.

tablatronix, 2021-03-31:
Well yeah you call startconfigportal, why?

you want configportal to run always in sta ap mode? Then you have to setup wifi on your own and use startwebportal. There IS a way to set mode for captive portal not sure if its exposed, let me check

Abedi98, 2021-04-01:
> Well yeah you call startconfigportal, why?
> 
> you want configportal to run always in sta ap mode? Then you have to setup wifi on your own and use startwebportal. There IS a way to set mode for captive portal not sure if its exposed, let me check

Yes I want it to be enabled in AP_STA mode and connected to home WiFi in station mode and an access point always enabled next to station mode

tablatronix, 2021-04-01:
These are not exposed you will have to change them in .h file to test if this works.

set mode you want STA+AP

then start cp

```
    bool          _disableSTA             = false; // disable sta when starting ap, always
    bool          _disableSTAConn         = true;  // disable sta when starting ap, if sta is not connected ( stability )
```

You can try changing those to false and test. I will check later

Abedi98, 2021-04-03:
> These are not exposed you will have to change them in .h file to test if this works.
>
> set mode you want STA+AP
>
> then start cp
>
> ```
>     bool          _disableSTA             = false; // disable sta [...]

```
WifiManager.h
 bool          _disableSTA             = false; 
 bool          _disableSTAConn         = false;
```

Hi. I applied these changes to the .h file but disconnected the station again

tablatronix, 2021-04-03:
esp32 or 8266?

tablatronix, 2021-04-03:
Working fine for me so far, I will see if something disconnects

```C++

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

WiFiManager wm;    

void setup() {
  WiFi.mode(WIFI_AP_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");

  wm.setEnableConfigPortal(false);
  wm.autoConnect();
  delay(1000);
  wm.startConfigPortal("WIFIAPSTA");
}

void loop() {
 
}
```

Abedi98, 2021-04-03:
> esp32 or 8266?

Wemos d1 mini (8266)

tablatronix, 2021-04-03:
Still connected to both..

Abedi98, 2021-04-04:
> Still connected to both.. Works fine
> 
> Something wrong with your code
> 
> you are using a server, wm already runs a server
> what is all this ?
> wdt_reset(); ESP.reset(); ESP.restart(); while (1)wdt_reset();

void setup()
{

  Serial.begin(9600);

  if (WiFi.SSID() == "") {

    WiFi.mode(WIFI_AP);

    configtone();

    delay(1000);
    initialConfig = true;

  } else {
    **if(mobilemode_flag == 1){
      WiFi.mode(WIFI_AP_STA);
      wm.setEnableConfigPortal(false);
      wm.autoConnect();
      delay(1000);
      wm.startConfigPortal("WIFIAPSTA");
    }else{
      WiFi.mode(WIFI_STA);
    }**

    WiFi.hostname("LOCALHOST_WIFI");
    int connRes = WiFi.waitForConnectResult();

    server.begin();
    

    wifiAPPrint();

  

  }

}

void loop()
{

  if (initialConfig) {
    WiFiManagerParameter custom_api("api", "API KEY", token, 65);

    wm.setSaveConfigCallback(saveConfigCallback);
    wm.addParameter(&custom_api);
    wm.setAPStaticIPConfig(IPAddress(192, 168, 4, 1), IPAddress(192, 168, 4, 1), IPAddress(255, 255, 255, 0));
    if (!wm.startConfigPortal("LOCALHOST")) {
      Serial.println("Not connected to WiFi but continuing anyway.");
    } else {
      strcpy(token, custom_api.getValue());
      if (shouldSaveConfig) {
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.createObject();
        json["api"] = token;
        File configFile = SPIFFS.open("/config.json", "w");
        if (!configFile) {
          Serial.println("failed to open config file for writing");
        }

        json.printTo(Serial);
        json.printTo(configFile);
        configFile.close();
        //end save
      }
      Serial.println("web server started:" + WiFi.localIP().toString());
      ESP.restart(); 

    }
  } else {
    server.handleClient();
  }

}

See the part where I bolded.
When wm.startConfigPortal ("WIFIAPSTA"); I run it, the station no longer connects and the loop does not run.

tablatronix, 2021-04-04:
What does your serial say ? Mine works just like that, do you get an error?

Abedi98, 2021-04-04:
> What does your serial say ? Mine works just like that, do you get an error?

no
There is no error. Only after 10 minutes the station is disconnected and reconnected again

tablatronix, 2021-04-04:
If you enable esp debugging it should show more info about what the radio is doing and clients.
Not sure what else could be going on, is your power supply good ?

tablatronix, 2021-04-04:
Also you might want to try using the non blocking code, since you will want your code to keep processing past setup

Abedi98, 2021-04-05:
> Also you might want to try using the non blocking code, since you will want your code to keep processing past setup

There is no problem when I test the same code with the following library,
**https://github.com/Hieromon/AutoConnect**

but I have this problem with the wifimanager library.
What is the problem?

yukimach, 2023-06-16:
> 

I'm trying to keep config portal open even after connect on a esp32s2. 

but once the esp32 connects to a wifi router the portal automatically closes. Is it possible to keep it open? I've read through a few posts but it's not very clear. This sort of user experience by tonyp7 is what I'm looking to do without having to use esp idf. https://www.youtube.com/watch?v=hxlZi15bym4&source_ve_path=MjM4NTE&feature=emb_title&ab_channel=idyl.io 
 
```
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address:
*wm:192.168.1.23
*wm:config portal exiting 
```

I've set 
```
WifiManager.h
 bool          _disableSTA             = false; 
 bool          _disableSTAConn         = false;
```

and use
```
`void setup() {
  WiFi.mode(WIFI_AP_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");

  wm.setEnableConfigPortal(false);
  wm.autoConnect();
  delay(1000);
  wm.startConfigPortal("WIFIAPSTA");
}
````