tzapu/WiFiManager #1150: Set html page for startConfigPortal

Abedi98, 2020-11-21:
This is the code I wrote for the dedicated page for startConfigPortal. 
But when I connect to the Device-AP, the information of the default portal page is displayed.

```
if(WiFi.SSID() == ""){
    wm.setAPStaticIPConfig(IPAddress(172,217,1,28), IPAddress(172,217,1,28), IPAddress(255,255,255,0));
    configtone();
    digitalWrite(D2, HIGH);
    digitalWrite(D3, HIGH);
    wm.autoConnect("ESP8266","1234567890xa");
    server.on("/", handleRoot);
    
  }else{ 
     
    Serial.println("after wireless availability : "+WiFi.SSID());
    wm.setAPStaticIPConfig(IPAddress(172,217,1,22), IPAddress(172,217,1,22), IPAddress(255,255,255,0));
    wm.setConfigPortalBlocking(false);
    wm.startConfigPortal("Device-AP","hhh1110098");
    server.on("/", handlePage);
  }

  server.begin();
```

hanndlePage() not displayed in 172,217,1,22

tablatronix, 2020-11-21:
You might have to use the web start callback, see the dev example

Abedi98, 2020-11-21:
> You might have to use the web start callback, see the dev example

Hello
I do not understand what you mean, please explain more

Thanks

Abedi98, 2020-11-21:
> web start callback,
```

void handleRoot()
{
  wm.startWebPortal();
}

void handlePage() {
  //digitalWrite(BUILTIN_LED, 1);
  server.send(200, "text/html", INDEX_HTML); //"text/plain"
  //digitalWrite(BUILTIN_LED, 1);
}

void bindServerCallback()
{
  wm.server->on("/",handleRoot);
}

void bindServerCallback2()
{
  wm.server->on("/custom",handlePage);
}

void setup()
{
  Serial.println("SSID : "+WiFi.SSID());
  delay(1000);
  Serial.begin(115200);
 
  pinMode(D2, OUTPUT); //Green LED
  pinMode(D3, OUTPUT); //Red LED
  
  digitalWrite(D2, LOW);
  digitalWrite(D3, LOW);
  digitalWrite(D8, LOW);

  if(WiFi.SSID() == ""){
    wm.setAPStaticIPConfig(IPAddress(172,217,1,28), IPAddress(172,217,1,28), IPAddress(255,255,255,0));
    wm.setWebServerCallback(bindServerCallback);
    configtone();
    digitalWrite(D2, HIGH);
    digitalWrite(D3, HIGH);
    wm.autoConnect("ESP8266","1234567890xa");    
  }else{ 
     
    Serial.println("after wireless availability : "+WiFi.SSID());
    wm.setAPStaticIPConfig(IPAddress(172,217,1,22), IPAddress(172,217,1,22), IPAddress(255,255,255,0));
    wm.setWebServerCallback(bindServerCallback2);
    wireless_availability();
    wm.setConfigPortalBlocking(false);
    wm.startConfigPortal("Device-AP","password");
  }

  server.begin();
}
```

not working ...

Abedi98, 2020-11-21:
> > web start callback,
> 
> ```
> 
> void handleRoot()
> {
>   wm.startWebPortal();
> }
> 
> void handlePage() {
>   server.send(200, "text/html", INDEX_HTML); //"text/plain"
> }
> 
> void bindServerCallback()
> {
>   wm.server->on("/",handleRoot);
> }
> 
> void bindServerCallback2()
> {
>   wm.server->on("/custom",handlePage);
> }
> 
> void setup()
> {
>   Serial.println("SSID : "+WiFi.SSID());
>   delay(1000);
>   Serial.begin(115200);
>  
>   pinMode(D2, OUTPUT); //Green LED
>   pinMode(D3, OUTPUT); //Red LED
>   
>   digitalWrite(D2, LOW);
>   digitalWrite(D3, LOW);
>   digitalWrite(D8, LOW);
> 
>   if(WiFi.SSID() == ""){
>     wm.setAPStaticIPConfig(IPAddress(172,217,1,28), IPAddress(172,217,1,28), IPAddress(255,255,255,0));
>     wm.setWebServerCallback(bindServerCallback);
>     configtone();
>     digitalWrite(D2, HIGH);
>     digitalWrite(D3, HIGH);
>     wm.autoConnect("ESP8266","1234567890xa");    
>   }else{ 
>      
>     Serial.println("after wireless availability : "+WiFi.SSID());
>     wm.setAPStaticIPConfig(IPAddress(172,217,1,22), IPAddress(172,217,1,22), IPAddress(255,255,255,0));
>     wm.setWebServerCallback(bindServerCallback2);
>     wireless_availability();
>     wm.setConfigPortalBlocking(false);
>     wm.startConfigPortal("Device-AP","password");
>   }
> 
>   server.begin();
> }
> ```
> 
> not working ...

http://connectivitycheck.gstatic.com/generate_204

tablatronix, 2020-11-21:
hmm

void handleRoot()
{
  wm.startWebPortal();
} 

why are you doing this?