tzapu/WiFiManager #1152: Can't use wifiManager.resetSettings(); inside loop

ITstreet1, 2020-11-27:
### Basic Infos
Arduino IDE 1.8.12

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

In example sketch AutoConnectWithFSPParameters.ino, there is a wifiManager.resetSettings(); function, which can reset settings of SPIFFS data. As wifiManager is declared inside setup, I can not use this function inside a loop with a button. Or maybe there is a way, that I don't know.

### Settings in IDE

Module:  			Generic ESP8266 Module

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {
// didn't change a thing from an example
}

void loop() {
wifiManager.resetSettings();
}
```

### Debug Messages

```
exit status 1
'wifiManager' was not declared in this scope
```

EgHubs, 2020-12-11:
Just redeclare it!
 ```
    if ( digitalRead(TRIGGER_PIN) == LOW) {
        WiFiManager wifiManager;
        wifiManager.resetSettings();
        delay(500);
        resetFunc();
      }
```
and resetFunc is declared before the void setup() function
`void(* resetFunc) (void) = 0; //declare reset function @ address 0
`

tablatronix, 2020-12-11:
Yeah this is a scope issue, basic C++, not a wm issue

closed 2020-12-11 by tablatronix

ChernikovS1971, 2022-04-24:
To use wifiManager.resetSettings() see the OnDemand example. 
Well described here: https://dronebotworkshop.com/wifimanager/

ChernikovS1971, 2022-04-25:
[EgHubs](https://github.com/EgHubs) 
thank! your code works!