tzapu/WiFiManager #1153: [QUESTION] ControlManager Existing after connecting to network from captive portal

Karmakstylez, 2020-11-29:
Hello,

Is there a way to keep the ControlManager accessible instead of wiping the settings and start over again so you can change the WiFi or reset the info via de captive portal?

Please let me know.

~Karmakstylez

andrezanna, 2020-12-20:
Hello, i'm interested in this too.
Thanks

flomerz95, 2020-12-20:
Hi I think that's not possible with the current version. The problem is that the controller must change from ap mode to sta mode to check your inserted WiFi credentials. During the start of the sta mode the ap mode will be disabled which will close the config portal...

tablatronix, 2020-12-20:
Yes in the dev and alpha version, you can ondemand and non blocking start the webserver on sta mode also

tablatronix, 2020-12-20:
#500

closed 2020-12-20 by tablatronix