tzapu/WiFiManager #1154: size of the lib

sebosfato, 2020-11-30:
Hello i was using old version of the lib and it was only for esp8266 i guess and it consumed 40kb less memory from my esp01 very small memory available.. upon moving to new updated lib with non blocking function it dont fit within the memory with enough space left for httpupdate... 

my question is there a way to edit the lib for use separete for esp8266 from esp32 or perhaps reduce the size of the wifi signal streght image to fit ? 

anybody help please?

Daemach, 2020-11-30:
FWIW - just noticed this on gitter:  https://www.screencast.com/t/nlFyjjfUX

sebosfato, 2020-12-01:
> FWIW - just noticed this on gitter: https://www.screencast.com/t/nlFyjjfUX
> […](#)

you mean could be possible to update from a compressed file? 

i tried to delete all the esp32 ifdef elif etc from the cpp file but didnt reduced much, it seems to me that the main alternatives would be to reduce the size of the png file or maybe delete the debug options... those seems to be related to this great increase in size of the lib... when i tried to delete the debug lines it broke the code..

tablatronix, 2020-12-01:
Nice I have been meaning to look into gzip, at least for the css.

I can see what I can remove, I am going to make the info details optional or reduce them back down, and fix up debug environment to not always compile, some of the css can be reduced and you can remove the icons

sebosfato, 2020-12-01:
Would be lovely to use it with the 01...  the older version even run but if i get over 406kb of sketch space the http update wont run... im using it at the limit of space... counting variables to try reduce it... 

i think that wifi signal strenght icon could be reduced with no loss in apearance maybe

tablatronix, 2020-12-01:
Nah I already optimized the sprite png, svg was larger. But I am open to optimizations, or ascii?

Daemach, 2020-12-01:
Should try that gzip method for the best across-the-board results...  It
looks like the simplest solution based on that forum post.  And it will get
lost soon as gitter is an active board.

tablatronix, 2020-12-01:
The main issue is that alot of the code is templated and dynamic, not much is actually static

sebosfato, 2020-12-01:
in the mean while i found some few special kilobytes reduction by setting the iwipvariants to high bandwidth no features.. do you know guys what features are this? could not find documentation about 

my sketch varies from 412k with to 402k with no features... 

I think most of the increase of the lib size may be the debug function  that is being compiled together... if we could simply deactivate the debug stuff from being compiled would turn out very much smaller... 

the original wifimanager cpp i was using had 22kb the new had 90kb 
taking all the esp32 configurations only released 10kb... so i think the diference may be the debug... 

just my guess

tablatronix, 2020-12-01:
That is a good question, I am not sure about the new lwip variants, quick search found not much info

tablatronix, 2020-12-01:
I moved all debug strings to flash
And added some testing for no debug
Testing basic.ino, not sure a 1-2%, I will look for another 1-2% by removing helper functions that are not internally used

RAM:   [===       ]  34.7% (used 28464 bytes from 81920 bytes)
Flash: [===       ]  31.8% (used 332580 bytes from 1044464 bytes)

WM_NODEBUG
RAM:   [===       ]  34.7% (used 28416 bytes from 81920 bytes)
Flash: [===       ]  30.9% (used 322744 bytes from 1044464 bytes)

tablatronix, 2020-12-01:
Master

RAM:   [====      ]  35.5% (used 29052 bytes from 81920 bytes)
Flash: [===       ]  28.1% (used 293260 bytes from 1044464 bytes)

Where are you getting your numbers from ?

sebosfato, 2020-12-01:
when i compile with the new alpha version it come out with 438kb and with the old one 410..

something is going wrong with the updates... thats why i was trying to reduce things without loosing functionality 

my 01 is 1mb so it should be able to get updates up to 470kb doesnt it? 

what could be going wrong?

tablatronix, 2020-12-01:
What all are you loading ? Additional libraries?

sebosfato, 2020-12-01:
Im using lots of libraries, let me show you

#include <FS.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <ESP_EEPROM.h>
#include <WiFiUdp.h>  
#include <WiFiManager.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <NTPClient.h>
#include <TaskScheduler.h>
#include <RTCVars.h>
#include "SAVE.h"
#include "SERVERON.h"
//#define ATOMIC_FS_UPDATE
RTCVars state; 

each is needed by a function called in the messy code... 

and the code is more than 1000 lines long...

tablatronix, 2020-12-01:
28K more ?
I have no idea , are you using params?

sebosfato, 2020-12-02:
from the wifimanager lib im using 
the wm.autoconnect on demand non blocking and set timeout 
and there is a function to be called to call  the wifi portal by the user 
also the acess point have names on it... is this the parameters?

wm gives a lot of debuging info on the serial monitor... maybe thats whats calling so much memory on the new version 
it may be interesting to know whats is going but perhaps would work even better lighter =) 

i found that taking printf comands out takes some bits out...

tablatronix, 2020-12-02:
I already removed the debug as I stated above, it should save you like 11k

tablatronix, 2020-12-02:
2k if I remove vector

sebosfato, 2020-12-02:
with the new version you changed i tried and it reduced from 430 to 427 kb size of the ino.bin file its great already because it dont show those parameters at startup...

tablatronix, 2020-12-02:
Is that with or without debugging ?

WM_NODEBUG

tablatronix, 2020-12-02:
I am about to merge in OTA, I hope it doesn't make it that much bigger... lol

sebosfato, 2020-12-02:
> Is that with or without debugging ?
> 
> WM_NODEBUG

i tried this  wm.setDebugOutput(false); how do you set this wm_nodebug?

tablatronix, 2020-12-02:
Are you using arduino? You might have to edit the .h file and add it. In pio you can add it to build flags. For now debug is enabled for dev reasons

sebosfato, 2020-12-03:
im using the last arduino ide 

esp 01  im trying to add to see what happens  ii try #define  #define WM_NODEBUG

tablatronix, 2020-12-03:
Nah if you do it in your sketch it wont work, Ill cut a new branch with no debug for you to try

sebosfato, 2020-12-03:
> Nah if you do it in your sketch it wont work, Ill cut a new branch with no debug for you to try

great..thanks a million  where can we find?

tablatronix, 2020-12-03:
branch dev_nodebug

sebosfato, 2020-12-03:
Hey i just tested it worked now it takes 13kb less memory 

thanks a milion times saved the day.... 

i have a question dont know if is related but my http updates are working well but i get no response for the sketck ok updated 

i tried all here and there but it seems that is interfering with something on memory because with nodemcu that has 4mspiff it works.. 

when the sketch finish uploading the serial monitor prints some blank spaces... 

do you know what could it be?: i googled it but didnt found solution 

problem is that i would like to update eprom before reset and is reseting before the switch ret so i have no chance to save..

tablatronix, 2020-12-03:
I get caught by this now and then, it might be the same issue,sometimes you have to restart the esp after ota as it does not do a real restart, only a soft.

tablatronix, 2020-12-03:
ALso try clean erasing flash if you have been switching esp lib versions

sebosfato, 2020-12-03:
i tried everything you can imagine even changing the order of libs and  how the things are called in setup i suspect it may have to to with the conversion toFloat()  or something because the led build in is one of the serial gpio1 or is because task manager 
the update is going on so im able to change  the file version for the update to validade but i have to change in the sketch every time
its strange because on the nodemcu with 4mb works no problem.. therefore i only need to change the version on a php file 

void UPDATE(){
//check version of update
HTTPClient http;

http.begin(client, versaoatualphp);
http.addHeader("Content-Type", "application/x-www-form-urlencoded");
 int httpCode = http.GET();            //Send the request

 Serial.println(httpCode);   //Print HTTP return code
if (httpCode==200 && (http.getString()!="") ){
versaoatual = http.getString();
 Serial.println("versao atual " + versaoatual);
 Serial.print("versao ");
 Serial.println(versao); 
 http.end();
 
if (versao==versaofs){
 Serial.println("versao atual " + versaoatual);
 UPDATEFS();
}

else if (versao != versaoatual.toFloat()) {

  
 Serial.println("versao da atualização ");
 Serial.println(versaoatual);
 Serial.print("checando update versao instalada ");
 Serial.println(versao); 
   Serial.println("Update Sketch...");
   
 // Serial.end();
    
    ESPhttpUpdate.setLedPin(LED, LOW);
    t_httpUpdate_return ret = ESPhttpUpdate.update(client, updatesketch);
   if (ret == HTTP_UPDATE_OK) {
     EEPROM.put(102, versaofs);  EEPROM.commit();
//     Serial.println("Updated spiff okkkk...");
   }
    switch(ret) {
                 case HTTP_UPDATE_FAILED:
                    Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
                    break;
            
                  case HTTP_UPDATE_NO_UPDATES:
                    Serial.println("HTTP_UPDATE_NO_UPDATES");
                    break;
                  
                  case HTTP_UPDATE_OK:
                      //versao=versaofs;
                      EEPROM.put(102, versaofs);  EEPROM.commit();
                      Serial.println("HTTP_UPDATE_OK");
                    break;
                    }
                    http.end();
                    }
                      Serial.println("A versao era igual a fonte");
                    }else{
                  Serial.println("conexao falhou");
 }}

here in this part works nice it prints the output 

void  UPDATEFS (){
   if (versao == versaofs ) {
//check version of update
//HTTPClient http;
Serial.println("Update SPIFFS..."); 

   HTTPClient http;
    ESPhttpUpdate.setLedPin(LED, LOW);
    t_httpUpdate_return ret = ESPhttpUpdate.updateSpiffs(spiffslink);
    if (ret == HTTP_UPDATE_OK) {
     Serial.println("Updated spiff okkkk...");

      switch (ret) {
        case HTTP_UPDATE_FAILED:
          Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
          break;

        case HTTP_UPDATE_NO_UPDATES:
          Serial.println("HTTP_UPDATE_NO_UPDATES");
          break;

        case HTTP_UPDATE_OK:
          versao=versaoatual.toFloat();
          EEPROM.put(102, versao);  // long - so 8 bytes (next address would be '12')
          //digitalWrite(LED, LOW);
          // commit (write) the data to EEPROM - only actually writes if there has been a change
          EEPROM.commit(); //Serial.println((ok43) ? "43 commit OK" : "Commit 43 failed");
          Serial.println("HTTP_UPDATE_OK");
           variavel="versao";
              valor=versao;
         // POSTDATA();
         break;
          }
          http.end();
          } 
         }else{
       
        }
         
          ESP.restart();
        }

here is the serial output for the part that fails 

07:38:22.676 -> 200
07:38:22.676 -> versao atual 1.01    
07:38:22.676 ->     
07:38:22.676 ->     
07:38:22.676 ->     
07:38:22.676 ->     
07:38:22.676 -> versao 1.00
07:38:22.676 -> versao da atualização 
07:38:22.676 -> 1.01    
07:38:22.676 ->     
07:38:22.676 ->     
07:38:22.676 ->     
07:38:22.676 ->     
07:38:22.676 -> checando update versao instalada 1.00
07:38:22.676 -> Update Sketch...
07:38:23.998 ->

sebosfato, 2020-12-03:
> ALso try clean erasing flash if you have been switching esp lib versions

what exactly you mean here? by the way great job on the lib!!! thanks again its opening the doors for most people working with those chips

tablatronix, 2020-12-03:
Full flash erase from esptool or arduino, sometimes the flash gets corrupted

sebosfato, 2020-12-04:
was working flawless on node mcu now is not working...not sure what is going... same code same problem... 

i tried switching back to the standard wifimanager to see if this was the diference ... not a clue... 

=(

sebosfato, 2020-12-04:
i just managed it to work by seting this flag  ESPhttpUpdate.rebootOnUpdate(false);

 ESPhttpUpdate.rebootOnUpdate(false);

but i would like to understand why it stoped working ... not sure if they changed the lib... i think i updated my arduino ide few days ago...

sebosfato, 2020-12-04:
i found it because i found how to turn on the debug from httpupdate on the preferences and it was returning the ok but not printing because maybe some function changed 

so i found this boolean function ESPhttpUpdate.rebootOnUpdate(false); inside the .h file of this lib
the code end up like this : 

 ESPhttpUpdate.rebootOnUpdate(false);
   Serial.println("Update Sketch...");
   ESPhttpUpdate.setLedPin(LED, LOW);
   t_httpUpdate_return Ret = ESPhttpUpdate.update(client, "http://www.update.....");
    switch (Ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
          versao=versaofs;
          variavel="versao";
          valor=versao;
          POSTDATA();
          digitalWrite(LED, LOW);
          Serial.println("HTTP_UPDATE_OK");
          Serial.println("Updated Sketch ooooookkk...");
          EEPROM.put(198, versao);  // long - so 8 bytes (next address would be '12')
          digitalWrite(LED, LOW);
          // commit (write) the data to EEPROM - only actually writes if there has been a change
          boolean ok43 = EEPROM.commit(); Serial.println((ok43) ? "43 commit OK" : "Commit 43 failed");
           EEPROM.put(170, userid);  // long - so 8 bytes (next address would be '12')
          boolean ok49 = EEPROM.commit(); Serial.println((ok49) ? "43 commit OK" : "Commit 43 failed");
          Serial.println("HTTP_UPDATE_OK");
          Serial.println("versao atualizada"+ String(versao));
          ESP.restart();
          break;
          
          } } Serial.println("A versao era igual a fonte");
    }else{
      Serial.println("falhou");
 }
  //http.end();

tablatronix, 2020-12-04:
Yeah I am not entirely sure how ESPhttpUpdate relates to this issue , but you should check that out with the esp guys, ask on gitter

sebosfato, 2020-12-05:
i tried to ask them there lets see what comes out... thanks a lot bro!

tablatronix, 2021-04-15:
What board are you using ?