tzapu/WiFiManager #1159: Custom CSS and Html

rohoop, 2020-12-06:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I need to put custom html and css code but i dont know in which code and which part of the code... Help! Friends! 
### Settings in IDE

Module:  			NodeMcu

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

charitha95, 2020-12-09:
You could simply add a style tag to the head like this. 
`wifiManager.setCustomHeadElement("<style>body {background-color: powderblue;}</style>");`

IcaroCreativa, 2021-02-27:
Hi!

I would like to know how to changes text in buttons and the colors of the buttons. 

Thanks!

IcaroCreativa, 2021-02-27:
Hi,

just modify the string_en file in the WiFimanager library in the arduino libraries file.
Attention better to make a copy of the original file.
you can change button colors, text etc

ppytlarz, 2024-02-11:
Hi,

i want to delete 2 buttons info and update where i can delete it?

[image]

gokhanmeteerturk, 2024-07-21:
@ppytlarz 
> i want to delete 2 buttons info and update where i can delete it?

That's more of a css question really. You can see the html template here: 
[https://github.com/tzapu/WiFiManager/blob/master/extras/WiFiManager.template.html](https://github.com/tzapu/WiFiManager/blob/master/extras/WiFiManager.template.html)

https://github.com/tzapu/WiFiManager/blob/e978bc059c522404c01e06cd136fcf23234eb784/extras/WiFiManager.template.html#L221-L231

And use the code @charitha95 already provided, but with your custom css to hide the buttons.

In your case, this would be:
```c
wifiManager.setCustomHeadElement("<style>form[action='/info'],form[action='/exit']:nth-of-type(2){display:none}</style>");
```