tzapu/WiFiManager #1160: is there a forum about tzapu wifimanager

ldijkman, 2020-12-07:
sorry not an issue
is there a forum about tzapu wifimanager
i have some questions that does not fit the isseus here

for my tft touch thermostat
https://www.youtube.com/watch?v=qNYT64CaLEI
https://oshwlab.com/l.dijkman/esp32-dev-kit-38-pin-to-spi-touch-tft

tablatronix, 2020-12-08:
I can throw a discord together or a gitter

ramanraja, 2020-12-08:
@tablatronix: Please do. It would be greatly helpful for many of us. Thanks.

ldijkman, 2020-12-08:
i see you can add a discussion link / tab / button to github
i think its pritty new

https://github.com/ldijkman/ART-ESP32-Touch-TFT-Thermostat/discussions

tablatronix, 2020-12-08:
Hmm ill look into it but this is not my repo so I might not have access to manage it

tablatronix, 2020-12-11:
I added a gitter to the readme for now

closed 2020-12-18 by tablatronix