tzapu/WiFiManager #1163: Notify the user when user is not successfully connected to the local network (after providing the credentials) / invalid password scenario

charitha95, 2020-12-09:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I would like to notify the user when the user is not connected successfully to the local network after providing the credentials. Is there a way to redirect the user to a page or tell the user that the given credentials are invalid?

charitha95, 2020-12-09:
I think dev branch addresses this issue already.

closed 2020-12-09 by charitha95

tablatronix, 2020-12-09:
It will say invalid password after it starts the ap back up, but it will not automatically redirect you to it after saving, the reason is that the ap is shutdown when testing new connections, and captive portals will close, we may add some js checking, but it is later on the todo