tzapu/WiFiManager #1166: ESP8266-12E Device is connected to the WiFi but with no internet connection.

EgHubs, 2020-12-11:
**WiFimanager Branch/Release:**
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266

**ESP Core Version:**
- [ ] 2.7.4

### Description

i tried this [code ](https://github.com/3zuli/esp_internet_alarm/blob/master/esp_internet_alarm.ino) which connects to the internet with `  WiFi.begin(ssid, password); 
` and it's ok and has an internet connection, but when I try to connect to the WiFi with WifiManger it joins the WiFi but with no internet connection.
### Settings in IDE

Module:  			NodeMcu, 

### Sketch

```cpp

#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h> // https://github.com/bblanchon/ArduinoJson

char api_token[32] = "YOUR_API_TOKEN";

//default custom static IP
char static_ip[16] = "10.0.1.56";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";
bool shouldSaveConfig = false;
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}
// Select mode of operation by uncommenting one of the following lines
// Default: DETECT_NO_INTERNET
//#define DETECT_INTERNET    // Turn on the light when there IS internet connnection
#define DETECT_NO_INTERNET // Turn on the light when there is NO internet connection

// The URL we will use to test our connection
// We'll be connecting to http://httpbin.org/get , the "get" part is specified later in the code
// httpbin returns a very small amount of data, therefore should work
// even on the slowest connection (EDGE/2G)
const char* host = "httpbin.org";

// Global variable to track our connection status
bool hasInternet = false;

// The ESP pin on which the strobe light is connected
const int strobePin = 02;

// Set to true to enable debug printouts
const bool debug = false;

void led(int pin) {
#if defined(DETECT_INTERNET)
  digitalWrite(pin, hasInternet);
  if (debug) {
    Serial.print("Setting LED to ");
    Serial.println(hasInternet);
  }
#elif defined(DETECT_NO_INTERNET)
  digitalWrite(pin, !hasInternet);
  if (debug) {
    Serial.print("Setting LED to ");
    Serial.println(!hasInternet);
  }
[... 290 lines cut ...]
connected...yeey :)
custom_api_token hereeeeeeeeeeeeeeeee

local ip
192.168.1.210
192.168.1.1
255.255.255.0

Operation mode: DETECT_NO_INTERNET
WiFi connected
IP address: 192.168.1.210
*WM: [3] freeing allocated params! 
*WM: [3] unloading 
WiFi OK
Connecting to httpbin.org
Connection failed!!!
WiFi OK
Connecting to httpbin.org
Connection failed!!!
WiFi OK
Connecting to httpbin.org
Connection failed!!!
WiFi OK

```

tablatronix, 2020-12-11:
It must have to do with the static ip what happens with dhcp ?

EgHubs, 2020-12-11:
Yeah, it's now connected, but I need it to be with a static IP, it's a must for me, how can I achieve that, please.
Also for the DHCP, I tried to use AutoConnectWithFSParameters example but it seems that there is a static IP option on the web configuration page which I think that it shouldn't be there, so I tried AutoConnectWithFeedbackLED .

tablatronix, 2020-12-11:
you can turn the static fields off, not sure why its not working, are you using the same ip, gateway is correct?

EgHubs, 2020-12-12:
Didn't try the same ip that dynamically assigned but i will today, but iam very sure that the ip that i used is available on the network. Also the gateway and subnet mask are correct.
Iam using the latest version of arduino, esp board core and the latest alpha version of the library.
My sdk version is 2.2 i guess which is old i know but i tried many times to upgrade it without success, dose it even affect?

tablatronix, 2020-12-12:
I will give it a try today and see if there is an unfound bug
Are you using alpha or github branch ?

EgHubs, 2020-12-12:
> Are you using alpha or github branch ?

I tried the latest versions on both with no success, also tried joining with the same IP that was dynamically assigned with no success, tried the code with another ESP8266-12E and the problem stays the same.
Tried joining with a static IP Address with my android phone to see if the problem was related to my router and the phone connected to the internet successfully.
I really really hope you come up with something with your research, I'll be waiting for you.

tablatronix, 2020-12-12:
I assume you tested this with a small static example, not using all the json stuff, just setstaconfig static

EgHubs, 2020-12-12:
:(
Sorry for that, I didn't test it, but I just did now with a simple code without the use of WM library, and the same problem exists!.
I am really sorry for wasting your time :( ...
in the code that I tested right now when I just removed the line `WiFi.config(local_IP, gateway, subnet)` boom there is an internet connection with the line in the code no internet connection.
it just uses ESP8266WiFi library.
any recommendations, please?
the code I used for explanation:
```cpp
#include <ESP8266WiFi.h>
// Replace with your network credentials
const char* ssid     = "Shalaby";
const char* password = "01008711248.Sh.";
#define DETECT_NO_INTERNET // Turn on the light when there is NO internet connection

// Set your Static IP address
IPAddress local_IP(192, 168, 1, 215);
// Set your Gateway IP address
IPAddress gateway(192, 168, 1, 1);

IPAddress subnet(255, 255, 255, 0);
//IPAddress primaryDNS(8, 8, 8, 8);   //optional
//IPAddress secondaryDNS(8, 8, 4, 4); //optional
// even on the slowest connection (EDGE/2G)
const char* host = "httpbin.org";

// Global variable to track our connection status
bool hasInternet = false; 

// The ESP pin on which the strobe light is connected
const int strobePin = 02;

// Set to true to enable debug printouts
const bool debug = false;

void led(int pin){
#if defined(DETECT_INTERNET)
    digitalWrite(pin,hasInternet);
    if(debug){Serial.print("Setting LED to "); Serial.println(hasInternet);}
#elif defined(DETECT_NO_INTERNET)
    digitalWrite(pin,!hasInternet);
    if(debug){Serial.print("Setting LED to "); Serial.println(!hasInternet);}
#else
    digitalWrite(pin,!hasInternet);
    if(debug){Serial.print("Setting LED to "); Serial.println(!hasInternet);}
#endif
}
//// led(strobePin);

int connectWifi(int retryDelay=500){
  hasInternet=false; 
  led(strobePin);
  
  //Start connecting, the connection process is done by the ESP in the background
  WiFi.begin(ssid, password); 
  
  // values of WiFi.status() (wl_status_t) defined in wl_definitions.h located in (Windows):
  // %AppData%\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.5-947-g39819f0\libraries\ESP8266WiFi\src\include
  // Mac & Linux will vary
  
  wl_status_t wifiStatus = WL_IDLE_STATUS; //Assume nothing
  while (wifiStatus != WL_CONNECTED) {
    // While the ESP is connecting, we can periodicaly check the connection status using WiFi.status()
    // We keep checking until ESP has successfuly connected
    wifiStatus = WiFi.status();
    switch(wifiStatus){
      // Print the error status we are getting
      case WL_NO_SSID_AVAIL:
          Serial.println("SSID not available");
          hasInternet=false;
          break;
[... 102 lines cut ...]
  String url = "/get";
  
  Serial.print("Requesting URL: ");
  Serial.println(host+url); 
  
  // We need to manually create the HTTP GET message
  // client.print() will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(10); //Wait for it to send the message
  
  // Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  // We don't check if we actually received something from server
  // TODO: measure ping and turn on the strobe, when it's too high
  
  Serial.println("closing connection");
  Serial.println();
}
```

EgHubs, 2020-12-12:
Damn it now works !!
i just set the primary and secondrey dns server with the following line `WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)`
which used 
```cpp
IPAddress primaryDNS(8, 8, 8, 8); 
IPAddress secondaryDNS(8, 8, 4, 4);
```
Now, how can I achieve that in WM library?
can I set those settings from the config web page?
and if I cant, how can I config just these two parameters in the configurations as a default, as I will need to keep the other values like static IP and default gateway as variables.

tablatronix, 2020-12-12:
ahh, interesting I am testing right now and yes the esp does not get dns from dhcp, i also have fail with static ip

tablatronix, 2020-12-12:
```
  IPAddress _ip = IPAddress(192, 168, 20, 5);
  IPAddress _gw = IPAddress(192,168,20,1);
  IPAddress _sn = IPAddress(255, 255, 255, 0);
  IPAddress _dns = IPAddress(8,8,8,8);
  //end-block2
  
  wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn, _dns);
```

EgHubs, 2020-12-12:
I was about to comment that i solved it, but you got me first :).
Will I solved it by two solutions, the first is yours after I read the readme file about that function I finally connected to the internet, but I got confused because my router has a different DNS server, will that be a problem !?.
Then after some search I got another solution by the help of [this ](https://github.com/esp8266/Arduino/issues/369#issuecomment-108089750)man comment, and achieved it by the following:

```cpp
  if (!wifiManager.autoConnect("xxxxxxx", "xxxxxxx")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  } 
  Serial.println("Dynamic Ip IS");
  Serial.println(WiFi.localIP());
  IPAddress _ip, _gw, _sn;
  _ip.fromString(static_ip);
  _gw.fromString(static_gw);
  _sn.fromString(static_sn);
  //wifiManager.setSTAStaticIPConfig(local_IP, _gw, _sn); //WHICH DIDN'T WORK I DON'T KNOW WHY
  WiFi.config(_ip, _gw, _sn);
  Serial.println("Static Ip Is");
  Serial.println(WiFi.localIP());
  Serial.println("connected...yeey :)");
```
And I can see that this will be better as the ESP will connect to the internet with the same DNS server as the router, and also will solve the problem you stated about the secondary DNS server.
Anyway, I am really thankful for your help, time, and effort :)

EgHubs, 2020-12-12:
Finnally, after so many tries I accomplished it,
Here is my final tested code i think it's understandable by the comments :
```cpp
WiFiManagerParameter custom_api_token("api", "api token", "", 32);
  wifiManager.addParameter(&custom_api_token);
  bool FirstConfig = false;
  if (WiFi.SSID() == "") {//first configuration then we assign a static ip--- We just enter here once in life at the configuration
    FirstConfig = true;
    IPAddress _ip, _gw, _sn;
    _ip.fromString(static_ip);
    _gw.fromString(static_gw);
    _sn.fromString(static_sn);
    wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);
  }
  wifiManager.setMinimumSignalQuality();
  Serial.println("connectingggggg)");
 wifiManager.setBreakAfterConfig(true);
  if (!wifiManager.autoConnect("xxxxxxxx", "xxxxx")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
  }
  if (!FirstConfig) //we don't enter here in the first configuration--- We Only Enter it after
  { Serial.println("FirstConfig"); // if we are here then the ESP IP is assigned automatically which means we now have the dns server :)
IPAddress _ip, _gw, _sn;
_ip.fromString(static_ip);
_gw.fromString(static_gw);
_sn.fromString(static_sn);
WiFi.config(_ip, _gw, _sn);    //now we assign our static IP 
  }
Serial.println("connected...yeey :)");// At the First configuration we now have no Internet as we have a static IP so we will restart the esp after saving the configurations.

if (shouldSaveConfig) {
  Serial.println("saving config");

#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
  DynamicJsonDocument json(1024);
#else
  DynamicJsonBuffer jsonBuffer;
  JsonObject& json = jsonBuffer.createObject();
#endif

  json["ip"] = WiFi.localIP().toString();
  json["gateway"] = WiFi.gatewayIP().toString();
  json["subnet"] = WiFi.subnetMask().toString();
  json["api_token"]   = api_token;
  userIDString = api_token;
  path = "/usersData/" + userIDString;
  File configFile = SPIFFS.open("/config.json", "w");
  if (!configFile) {
    //Serial.println("failed to open config file for writing");
  }

#ifdef ARDUINOJSON_VERSION_MAJOR >= 6
  serializeJsonPretty(json, Serial);
  serializeJson(json, configFile);
#else
  json.printTo(Serial);
  json.printTo(configFile);
#endif

configFile.close();
ESP.restart();// restart as at this point we have a static IP which means no internet.
}
```
**I would like to know what you think about it, is it a good try or something is wrong with it?**

tablatronix, 2020-12-13:
Good ideas here thanks

EgHubs, 2020-12-13:
Glad you like it 🤩

closed 2020-12-13 by EgHubs