tzapu/WiFiManager #1167: WiFiManager inside of Fast_LED-Webserver not wortking

fretsbiz, 2020-12-11:
Board: Lolin D1 Mini
App: Fast_LED-Webserver
Issue: WiFiManager not visible and unable to connect using 192.168.4.1

Okay, I 've been pulling my eyeballs out of their sockets.  Can you guys see anything in the following code that is preventing WiFimanager from starting up in Access Point mode?  I finally solved the issue of SPIFFS puking all over me by installing ESP8266FS-0.5.0.zip located at https://github.com/esp8266/arduino-esp8266fs-plugin/releases  It's been one thing or another followed by something else, and I really need a hand.  Thanks.

/*
   ESP8266 FastLED WebServer: https://github.com/jasoncoon/esp8266-fastled-webserver
   Copyright (C) 2015-2018 Jason Coon

   This program is free software: you can redistribute it and/or modify
   it under the terms of the GNU General Public License as published by
   the Free Software Foundation, either version 3 of the License, or
   (at your option) any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

//#define FASTLED_ALLOW_INTERRUPTS 1
//#define INTERRUPT_THRESHOLD 1
#define FASTLED_INTERRUPT_RETRY_COUNT 0

#include <FastLED.h>
FASTLED_USING_NAMESPACE

extern "C" {
#include "user_interface.h"
}

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266HTTPClient.h>
//#include <WebSocketsServer.h>
#include <FS.h>
#include <EEPROM.h>
//#include <IRremoteESP8266.h>
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager/tree/development
#include "GradientPalettes.h"

#define ARRAY_SIZE(A) (sizeof(A) / sizeof((A)[0]))

#include "Field.h"

//#define RECV_PIN D4
//IRrecv irReceiver(RECV_PIN);

//#include "Commands.h"

WiFiManager wifiManager;
ESP8266WebServer webServer(80);
//WebSocketsServer webSocketsServer = WebSocketsServer(81);
ESP8266HTTPUpdateServer httpUpdateServer;

#include "FSBrowser.h"

#define DATA_PIN      D8
#define LED_TYPE      WS2812
#define COLOR_ORDER   GRB  //manufacturers use different data ordering - some use RGB, some use GRB, some use BGR.
#define NUM_LEDS      314

#define MILLI_AMPS         2000 // IMPORTANT: set the max milli-Amps of your power supply (4A = 4000mA)
#define FRAMES_PER_SECOND  120  // here you can control the speed. With the Access Point / Web Server the animations run a bit slower.
[... 1234 lines cut ...]
    uint16_t bri16 = (uint32_t)((uint32_t)b16 * (uint32_t)b16) / 65536;
    uint8_t bri8 = (uint32_t)(((uint32_t)bri16) * brightdepth) / 65536;
    bri8 += (255 - brightdepth);

    uint8_t index = hue8;
    //index = triwave8( index);
    index = scale8( index, 240);

    CRGB newcolor = ColorFromPalette( palette, index, bri8);

    uint16_t pixelnumber = i;
    pixelnumber = (numleds - 1) - pixelnumber;

    nblend( ledarray[pixelnumber], newcolor, 128);
  }
}

// Alternate rendering function just scrolls the current palette
// across the defined LED strip.
void palettetest( CRGB* ledarray, uint16_t numleds, const CRGBPalette16& gCurrentPalette)
{
  static uint8_t startindex = 0;
  startindex--;
  fill_palette( ledarray, numleds, startindex, (256 / NUM_LEDS) + 1, gCurrentPalette, 255, LINEARBLEND);
}

EgHubs, 2020-12-12:
Maybe you can post your code in a nicer way by selecting it and clicking on "insert code"  from the typing tab bar.
Also maybe you can post your serial monitor output so that we can see what is happening.
But as a start, you can try to update your library version, ESP board core version, and Arduino software version, also erasing your flash might help.

fretsbiz, 2020-12-12:
True & absolutely agree. I finally figured it out. Ultimately it was a bad
Wemos D1 Mini.  I also sent you the info for updating the link to the
broken version of esp8266fs on your Fast-LED-Webserver GitHub page for your
reference this morning. All in all you guys are a brilliant group and I
truly appreciate your patience & thank you for the light toasting rather
than a total flame-out! .....plus I took a look at the logic for using a
#if define () || define() as a high level filter this morning, so I
understand a little more of what's going on at the beginning of Wifi
manager.cpp

EgHubs, 2020-12-12:
Great to hear. 
I think it's time for closing the issue now :).

fretsbiz, 2020-12-12:
Agreed!

tablatronix, 2020-12-13:
Let me know if I need to do anything, I did not read that at all.. lol

closed 2020-12-13 by tablatronix