tzapu/WiFiManager #1171: After configuration completed  Connection result: WL_DISCONNECTED

Behzad-6-1, 2020-12-13:
I have been using AsyncWiFiManager for a while. It was working fine for past 4 months. 
It was connecting to any network with ease. Now it is connecting to my Mobile Hotspot however it results in WL_DISCOUNCTED when I connect to my router.
SSID and password are correct, after configuration completed I get the followings
*WM: [1] Connecting to saved AP: WiFi-F4B0
*WM: [3] Using password: XXXXXXXX
*WM: [3] WiFi_enableSTA:  enabled.
*WM: [3] Mode after delay: STA 
E (1000) wifi:sta is connecting, return error
*WM: [1] connectTimeout not set, ESP waitForConnectResult. 

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2

*WM: [2] [EVENT] WIFI_REASON: 2
*WM: [2] Connection result: WL_DISCONNECTED
*WM: [3] lastconxresulttmp: WL_STATION_WRONG_PASSWORD
*WM: [3] lastconxresult: WL_STATION_WRONG_PASSWORD
*WM: [1] AutoConnect: FAILED 
*WM: [2] Starting config portal. 
*WM: [2] AccessPoint set password is valid. 
*WM: [1] @password 
*WM: [3] WIFI station disconnect. 
*WM: [3] WiFi_enableSTA: 
*WM: [2] [EVENT] WIFI_REASON: 202
 enabled.
*WM: [2] Disabling STA. 
*WM: [2] Enabling AP. 
*WM: [1] StartAP with SSID: FE-IOTR
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [1] SoftAP Configuration: 
*WM: [1] -------------------- 
*WM: [1] ssid:             FE-IOTR
*WM: [1] password:         @password
*WM: [1] ssid_len:         7
*WM: [1] channel:          1
*WM: [1] authmode:         3
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.43.178
*WM: [3] setupConfigPortal() 
*WM: [1] Starting web portal. 
*WM: [3] DNS server started with IP: 192.168.43.178
*WM: [2] HTTP server started. 
*WM: [2] Async WiFi scan completed in  in 2624 ms.
*WM: [2] Async WiFi scan found: 2
*WM: [2] WiFi scan completed in 2625 ms.
*WM: [2] Config portal running in blocking mode, waiting for clients. 
*WM: [2] Num clients: 0 

Could you please help here?

Code is fairly standard
`void setup() {
  pinMode(LED, OUTPUT);
  pinMode(Relay1, OUTPUT);
  pinMode(Relay2, OUTPUT);
  pinMode(DI1, INPUT);
  pinMode(DI2, INPUT);
  pinMode(RestInput, INPUT);

  Serial.println("Configuring WDT...");
  //esp_task_wdt_init(WDT_TIMEOUT, true); //enable panic so ESP32 restarts
  //esp_task_wdt_add(NULL); //add current thread to WDT watch

  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  WiFi.begin();
  // start ticker with 0.5 because we start in AP mode and try to connect
  ticker.attach(0.6, tick);
  // put your setup code here, to run once:
  Serial.begin(115200);
  
 
  setupSpiffs();
 
  // AsyncWiFiManager, Local intialization. Once its business is done, there is no need to keep it around
  AsyncWiFiManager wm;

  //set config save notify callback
  wm.setSaveConfigCallback(saveConfigCallback);
  // setup custom parameters
  //
  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  AsyncWiFiManagerParameter custom_GOOGLE_SCRIPT_ID("Google", "GOOGLE SCRIPT ID", GOOGLE_SCRIPT_ID, 100);
  AsyncWiFiManagerParameter custom_LAST_RECORD("Record", "LAST RECORD", LAST_RECORD, 8);
  AsyncWiFiManagerParameter custom_Log_Interval("api", "Log Interval in Sec", Log_Interval, 5);
  AsyncWiFiManagerParameter custom_One_Sec("Test", "One Second in mSec", One_Sec, 8);

  //add all your parameters here
  wm.addParameter(&custom_GOOGLE_SCRIPT_ID);
  wm.addParameter(&custom_LAST_RECORD);
  wm.addParameter(&custom_Log_Interval);
  wm.addParameter(&custom_One_Sec);

  // set static ip
  IPAddress _ip,_gw,_sn;
    _ip.fromString(static_ip);
    _gw.fromString(static_gw);
    _sn.fromString(static_sn);
    wm.setAPStaticIPConfig(_ip, _gw, _sn);//
  //set custom ip for portal
  //wm.setAPStaticIPConfig(IPAddress(10, 0, 1, 1), IPAddress(10, 0, 1, 1), IPAddress(255, 255, 255, 0));

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  //here  "AutoConnectAP" if empty will auto generate basedcon chipid, if password is blank it will be anonymous
  //and goes into a blocking loop awaiting configuration
  if (!wm.autoConnect("FE-IOTR", "@password")) {
    DPRINTLN2("failed to connect and hit timeout");
    delay(3000);
    // if we still have not connected restart and try all over again
    ESP.restart();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
 
  ticker.detach();
  //keep LED on
  digitalWrite(LED, HIGH);

  //read updated parameters
  strcpy(GOOGLE_SCRIPT_ID, custom_GOOGLE_SCRIPT_ID.getValue());
  strcpy(LAST_RECORD, custom_LAST_RECORD.getValue());
  strcpy(Log_Interval, custom_Log_Interval.getValue());
  strcpy(One_Sec, custom_One_Sec.getValue());
  //save the custom parameters to FS
  SaveConfig();` 

### Basic Infos

#### ESP32 DIO-DEV
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

antonio-emilio, 2020-12-14:
I'm having the same trouble... sometimes it works just fine, but 90% of the time I get "Connection result: WL_DISCONNECTED".

tablatronix, 2020-12-18:
This is not asyncwifimanager..

closed 2020-12-18 by tablatronix

Behzad-6-1, 2020-12-18:
My issue was my resistor value on EN pin. faulty and instead of 10 k it was some random value in ohms. it is working fine.