tzapu/WiFiManager #1173: memory failure

sebosfato, 2020-12-15:
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40100a9c: umm_malloc_core(size_t) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 433
EXCVADDR: 0x40012121

Decoding stack results
0x4021de8a: String::changeBuffer(unsigned int) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\WString.cpp line 182
0x40100c84: malloc(size_t) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\umm_malloc\umm_malloc.cpp line 552
0x402213c1: uart_write(uart_t*, char const*, size_t) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\uart.cpp line 509
0x401005a0: pvPortMalloc(size_t, char const*, int) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\heap.cpp line 271
0x4021c644: HardwareSerial::write(unsigned char) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 161
0x4021c650: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 162
0x4020bdc9: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 441
0x4021ccb0: Print::println() at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Print.cpp line 184
0x4020c2c4: ESP8266WiFiGenericClass::mode(WiFiMode, WiFiState*) at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 415
0x40205a3c: setup() at C:\Users\Fabio\Documents\Arduino\GardenToy\pro4s/pro4s.ino line 487
0x4021f468: loop_wrapper() at C:\Users\Fabio\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 192

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] ESP8266
**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
**ESP Core Version:	2.4.0, staging**
 [ ] 2.4.0
### Description
after some time on if reboot unexpectedly it will ran into a sequence of autoreboot... printing the above stack trace... 
the last lines say is a problem in setup were i declared WiFi.mode(WIFI_STA); line 487 the other indication i dont understand much but i think is related to the wifi manager because if i reset the module with a button i set, it will erase the wifimanager credentials and the module run again but only in ap mode... if i set the password again it keeps reboting and giving this stack

so i think is a memory related issue and is conflicting with the wifimanager...

im using rtavars
spiffs 
eppron 

is this a known issue?

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

#include <FS.h>
#include <ESP8266WiFi.h> // ja tem no wifi manager 
#include <WiFiClient.h> // ja tem na libraria abaixo http client
#include <ESP8266HTTPClient.h> // httprequest  possui arduino.h <memory>  
#include <ESP8266httpUpdate.h>
#include <ESP_EEPROM.h>
#include <WiFiUdp.h>  //ota
#include <WiFiManager.h>   // contem esp8266webserver e esp8266WiFi.h
#include <ESP8266mDNS.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>// wifi manager
#include <NTPClient.h>
#include <TaskScheduler.h>
#include <RTCVars.h>
RTCVars state; 
WiFiClient client;

#define TRIGGER_PIN D3
const char* host = "GardenToy";

void RESTART() {
  wait = millis();
}
void t1Callback();
void t2Callback();
[... 226 lines cut ...]
//if (iplocal == "(IP unset)")iplocal="";

  //dns was here 
 dnsServer.setErrorReplyCode(DNSReplyCode::NoError);
  dnsServer.start(DNS_PORT, "*", apIP);
/* Setup web pages: root, wifi config pages, SO captive portal detectors and not found. */
 // server.on("/", handleRoot);
 //  server.on("/wifi", handleWifi);
  //server.on("/wifisave", handleWifiSave);
  server.on("/generate_204", handleRoot);
//  servidor.on("/generate_204", handleRooter);//Android captive portal. Maybe not needed. Might be handled by notFound handler.
  server.on("/fwlink", handleRoot);  //Microsoft captive portal. Maybe not needed. Might be handled by notFound handler.
  server.onNotFound(handleNotFound);
    server.begin();
delay(500);
  SPIFFS.begin();
  {
    Dir dir = SPIFFS.openDir("/");
    while (dir.next()) {
      String fileName = dir.fileName();
      size_t fileSize = dir.fileSize(); //     ////Serial.printf("FS File: %s, size: %s\n", fileName.c_str(), formatBytes(fileSize).c_str());
    } //Serial.printf("\n");
  }
  delay(500);
  MDNS.begin(host);

sebosfato, 2020-12-15:
relevant lines on unmm maloc ccp line 433
  while( cf ) {
    blockSize = (UMM_NBLOCK(cf) & UMM_BLOCKNO_MASK) - cf;

    DBGLOG_TRACE( "Looking at block %6d size %6d\n", cf, blockSize );
line  552

 /* Allocate the memory withing a protected critical section */

  UMM_CRITICAL_ENTRY(id_malloc);

  ptr = umm_malloc_core( size );

  UMM_CRITICAL_EXIT(id_malloc);

  return( ptr );
}

wstring line 182 
  uint16_t oldLen = len();
    char *newbuffer = (char *) realloc(isSSO() ? nullptr : wbuffer(), newSize);
    if (newbuffer) {
        size_t oldSize = capacity() + 1; // include NULL.
        if (isSSO()) {
            // Copy the SSO buffer into allocated space
            memmove_P(newbuffer, sso.buff, sizeof(sso.buff));
        }
        if (newSize > oldSize)
        {
            memset(newbuffer + oldSize, 0, newSize - oldSize);
        }
        setSSO(false);
        setCapacity(newSize - 1);
        setLen(oldLen); // Needed in case of SSO where len() never existed
        setBuffer(newbuffer);
        return 1;
    }
    return 0;
}

theres many lines... but seem to be something in the memory and wifi or hardware serial.. 

the problem is keep happening since some weeks ago... never  had it before...

sebosfato, 2020-12-15:
i noticed that when i connect my phone uncharged to the usb of the computer it kind of bugs the usb and the problem is reproduced again... 

so if i reset it fisicaly the wifimanager setings it work but when i set the password it keeps reboting over and over with that trace

sebosfato, 2020-12-15:
i tried to raise the eppron size to 210 was 202 and took that wifi sta mode out to see if the problem happen again 

the last saved value is on position 198 is a string of the version

any ideas please?

something that was not used before in this project when there was not problems is 

wificlient 
rtcvars 

i even thought the problem was related to updates but it seems last time i didnt updated and it happened again...

sebosfato, 2020-12-15:
i found a small error the version that is a string was being recorded on every start to 102 of epprom that was reserved for an int 
could it be the error?

sebosfato, 2020-12-16:
more info about it... i replicated the error  today... just disconected from usb and connected again... now its giving a watchdog reset 
after this print of i have a saved network.. i get the parameters from the wifimanager  seems to get stuck there 

 if (wm.getWiFiIsSaved()==1){
    
Serial.println("I have a saved network");
wm.setTimeout(180);
wm.setConfigPortalBlocking(false);
wm.autoConnect();
delay(500);
}

mac address xxxxxxx
04:35:15.194 -> xxxxxxx
04:35:15.194 -> I have a saved network
04:35:22.191 -> 
04:35:22.191 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
04:35:22.191 -> 
04:35:22.191 -> wdt reset
04:35:22.191 -> load 0x4010f000, len 3584, room 16 
04:35:22.191 -> tail 0
04:35:22.191 -> chksum 0xb0
04:35:22.191 -> csum 0xb0
04:35:22.225 -> v2843a5ac
04:35:22.225 -> ~ld

tablatronix, 2020-12-18:
REASON_SOFT_RESTART, means you are doing something in your code to restart

sebosfato, 2020-12-18:
actualy i dont have any while loop.... and was not calling restart.. thats why i thought maybe could be the wifimanager lib since the prints stoped there... i was diging in the net and  found somewhere that wificlient should be declared localy instead of globaly i changed this and i also catch an error of my code where i was trying to save a string to eeprom and i think this could be also the cause of the strange results... the last time it happen i could put just the sketch over it and it became alive again so the actuall credentials was not the problem... perhaps trying to save a string to eeprom was ruining it.. or the wificlient.. i call it many times in diferent functions... im trying to replicate the error after last changes... 

another thing is i took away the first declaration of ap ip mask etc...  i tried to look all the exception was giving me 
i never had such problems with memory before

sebosfato, 2020-12-18:
i will be trying the whole week to replicate the error.. basically take it out of usb and plug again if blinks problem if not .. for more than 2 days on probably the problem is solved .. lets hope so... 

i was trying to save the mac adress but i realised i didnt needed to

EEPROM only gets floats ints bool etc so far as i know how to implement and is all that is needed most of the times

tablatronix, 2020-12-18:
you will want to enable the esp debug from more detail

tablatronix, 2020-12-18:
You have to much code to debugm also `while (dir.next()) {` this is a loop ?

sebosfato, 2020-12-19:
> You have to much code to debug also `while (dir.next()) {` this is a loop ?

im not sure if this is usefull for the program it was from the fsbrowser example.... just left untouched ... 

i will try to debug more... but i think it was the eeprom the problem or the wificlient declaration being global... so far i could not replicate the error since i changes this two things