tzapu/WiFiManager #1174: Error: 'esp_err_t' was not declared in this scope

egrekov, 2020-12-16:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [X ] Master
- [X ] Development

**Esp8266/Esp32:**
- [X ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [X ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [X ] 2.3.0 ```PLATFORM: Espressif 8266 (2.3.3) > NodeMCU 1.0 (ESP-12E Module)```
- [ ] 2.4.0
- [ ] staging (master/dev)

### Settings in IDE
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps = 
    ArduinoJson
    ESP Async WebServer
    AsyncMqttClient
    https://github.com/tzapu/WiFiManager.git#development
build_flags = -fexceptions
build_unflags = -fno-exceptions
```

### Description

When compiling, an error occurs, tried both the master and the development branch

```c++
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp: In member function 'bool WiFiManager::WiFiSetCountry()':
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3170:3: error: 'esp_err_t' was not declared in this scope
   esp_err_t ret = false;
   ^
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3170:13: error: expected ';' before 'ret'
   esp_err_t ret = false;
             ^
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3191:33: error: 'ret' was not declared in this scope
Compiling .pio\build\nodemcuv2\libe40\LittleFS\lfs.c.o
        if(_wificountry == "US") ret = wifi_set_country((wifi_country_t*)&WM_COUNTRY_US);
                                 ^
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3192:33: error: 'ret' was not declared in this scope
   else if(_wificountry == "JP") ret = wifi_set_country((wifi_country_t*)&WM_COUNTRY_JP);
                                 ^
Archiving .pio\build\nodemcuv2\lib482\libESP8266mDNS.a
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3193:33: error: 'ret' was not declared in this scope
   else if(_wificountry == "CN") ret = wifi_set_country((wifi_country_t*)&WM_COUNTRY_CN);
                                 ^
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3200:6: error: 'ret' was not declared in this scope
   if(ret) DEBUG_WM(DEBUG_VERBOSE,F("[OK] esp_wifi_set_country: "),_wificountry);
      ^
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3205:10: error: 'ret' was not declared in this scope
   return ret;
          ^
.pio\libdeps\nodemcuv2\WiFiManager\WiFiManager.cpp:3206:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
*** [.pio\build\nodemcuv2\libe2f\WiFiManager\WiFiManager.cpp.o] Error 1
```

egrekov, 2020-12-16:
Did a little investigation and found a [commit](https://github.com/tzapu/WiFiManager/commit/e071e3ad8fb799044589643b1f355ef750485599) that broke compilation of the library under esp8266
-3155 +3170

egrekov, 2020-12-16:
Added [pull request](https://github.com/tzapu/WiFiManager/pull/1175)

tablatronix, 2020-12-16:
Thanks this should have not been committed sorry about that

closed 2020-12-16 by tablatronix