tzapu/WiFiManager #1177: Basic Dev Branch not compiling anymore

M4XXX, 2020-12-16:
#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x ] ESP32

### Description

Project with platformio and i get errors with my basic project...

### Settings in IDE / Platformio:
```
[env:ttgo-t-watch-test3]
platform = espressif32
board = ttgo-t-watch
framework = arduino
upload_speed = 2000000
monitor_speed = 115200
lib_deps =
  https://github.com/tzapu/WiFiManager.git#development
```

### Sketch

```
#include <Arduino.h>
#include <WiFiManager.h>

void setup() {    
    //Setting up Serial
    Serial.begin(115200);
    Serial.println("Serial initialized");

    WiFiManager wifiManager;
}

void loop(){

}
```

### Debug Messages

```
Dependency Graph
|-- <WiFiManager> 2.0.4-beta+sha.4d2a6b2
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <WiFi> 1.0

.pio\libdeps\ttgo-t-watch-test3\WiFiManager\WiFiManager.cpp: In member function 'bool WiFiManager::WiFiSetCountry()':
.pio\libdeps\ttgo-t-watch-test3\WiFiManager\WiFiManager.cpp:3184:10: warning: comparison of constant '12289' with boolean expression is always false [-Wbool-compare]
   if(ret == ESP_ERR_WIFI_NOT_INIT) DEBUG_WM(DEBUG_ERROR,"[ERROR] ESP_ERR_WIFI_NOT_INIT");
          ^
.pio\libdeps\ttgo-t-watch-test3\WiFiManager\WiFiManager.cpp:3185:15: warning: comparison of constant '258' with boolean expression is always false [-Wbool-compare]
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\esp32-hal-uart.c.o
   else if(ret == ESP_ERR_INVALID_ARG) DEBUG_WM(DEBUG_ERROR,"[ERROR] ESP_ERR_WIFI_ARG");
               ^
.pio\libdeps\ttgo-t-watch-test3\WiFiManager\WiFiManager.cpp:3186:19: error: 'ERR_OK' was not declared in this scope
   else if (ret != ERR_OK)DEBUG_WM(DEBUG_ERROR,"[ERROR] unknown error",(String)ret);
                   ^
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\libb64\cdecode.c.o
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\libb64\cencode.c.o
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\main.cpp.o
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\stdlib_noniso.c.o
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\wiring_pulse.c.o
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
Compiling .pio\build\ttgo-t-watch-test3\FrameworkArduino\wiring_shift.c.o
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
*** [.pio\build\ttgo-t-watch-test3\lib19b\WiFiManager\WiFiManager.cpp.o] Error 1
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
<command-line>:0:10: warning: ISO C99 requires whitespace after the macro name
```

tablatronix, 2020-12-16:
Did you pull ?

tablatronix, 2020-12-16:
Thanks, my bad

commit 41cb00ec (2020-12-16): fixes #1177 oops
closed 2020-12-16 by tablatronix