tzapu/WiFiManager #1178: Show assigned local IP address

litamin, 2020-12-17:
This is one of those long going questions that I simply have not found a solid answer for. It has been mentioned in #159, #265 and #872.

--> How to show the assigned local IP after adding your WiFi credentials? There are many real-world "out in the field" cases where you do not have access to the serial monitor to read this. And you don't want to add an OLED display or whatever either. How do you guys do this?

I understand the underlying problem of not being able to show it on the /wifisave page when still in AP mode, since I guess the IP is only known when one tries to enter STA mode. But then again, I know it is possible *somehow* since libraries like [Kentaylor's fork](https://github.com/kentaylor/WiFiManager) and [IoTAppStory](https://youtu.be/_4NRqIa-cvo?t=290) are in fact showing the assigned IP right there in the browser! 

Unfortunately I am not skilled enough to dig into those libraries to extract the inner workings. Can somebody help here - how do go about solving this?

sebosfato, 2020-12-21:
you can use mdns... but the new wifimanager lib show the ip if you conect to the config portal again... you can also send it to your email with webhooks for example

litamin, 2020-12-22:
Thanks @sebosfato, do you have a link to some instructions on a working mdns implementation? I've heard it is one of those "sometimes it works, sometimes it doesn't" type of things..

And regarding Wifimanager showing the IP if you connect to the config portal again: how do you mean exactly? Where do I go to see this? I am running dev branch on a ESP32, and once I have entered the new Wifi network credentials I am disconnected from the AP and it goes directly to STA mode and starts running the webserver code I have.

tablatronix, 2020-12-22:
You would have to use ondemand via a button or something. MDNS usually works well assuming you are on the same subnet or ap. Some routers will block mdns across wifi to lan etc.

litamin, 2020-12-23:
@tablatronix I'm afraid I'm not proficient enough in the workings of networking/routing stuff to understand what you mean.

I am looking for something that just works - in all cases and on any home wifi network. You enter your wifi credentials and get connected and then you just go to a custom http://HELLO.local or something to access my webserver.

Is that possible and can you guide me (or link to a guide) how to set that up?

jmgk77, 2021-07-31:
Hi,

I had the same issue, and **partially** fixed it adding `<script>setTimeout(function (){document.location.href = 'http://myapp.local';}, 30000);</script>` to `HTTP_SAVED[]`

That way, after credentials are saved, a timer is started, and after 30 seconds, the page location is changed from 192.168.4.1 to myapp.local. These 30s are enough to esp8266 boot my code, set a mdns, computer loose connection due disappearing hotspot, connect to previous wifi, and then load my page.

The problem is that it only work if you access wifimanager throught browser. If you use android or linux "login helpers" the page is closed before the URL change happen.

Hope it helps.

tablatronix, 2021-07-31:
Yeah this remains 1 of 2 issues with keeping the client in the loop on connects. CP closes as soon as we try to change wifi modes to test the connect.

EzeKoren, 2021-12-31:
As cool as it is to add a custom mDNS domain, it's not entirely adopted in the industry, for neither Windows nor Android devices support mDNS so it's important to be able to display the assigned IP

dagi0627, 2024-05-21:
Show assigned local IP address on browser is still not possible?

tablatronix, 2024-05-22:
Tasmota handles this well, no idea how, maybe they keep the ap active, but what if the channel is not the same. I would have to see how they are doing this and keeping the cp alive