tzapu/WiFiManager #1181: problems with ota update

droidprova, 2020-12-28:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !
.ERROR[4]: Not Enough Space
*WM: [3] -> 192.168.4.1 
.ERROR[4]: Not Enough Space
*WM: [3] -> 192.168.4.1 
.ERROR[4]: Not Enough Space
*WM: [3] -> 192.168.4.1 
ERROR[4]: Not Enough Space
*WM: [2] <- Handle update done 
*WM: [3] -> 192.168.4.1 
*WM: [1] [OTA] update failed 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
Hi everyone, I need help:

I am using the new example for the OTA update. To test this function I am trying to flash the same sketch that is in use in my wemos d1 lite (1MB) through the UPDATE function of the captive portal. When I try to update a 546 kb file.bin the operation fails and I get the following error on the serial monitor:

.ERROR [4]: ​​Not Enough Space
* WM: [3] -> 192.168.4.1
.ERROR [4]: ​​Not Enough Space
* WM: [3] -> 192.168.4.1
.ERROR [4]: ​​Not Enough Space
* WM: [3] -> 192.168.4.1
ERROR [4]: ​​Not Enough Space
* WM: [2] <- Handle update done
* WM: [3] -> 192.168.4.1
* WM: [1] [OTA] update failed
* WM: [3] -> www.msftconnecttest.com
* WM: [2] <- Request redirected to captive portal
* WM: [3] -> www.msftconnecttest.com
* WM: [2] <- Request redirected to captive portal

but then if I load another 256kb .bin file the operation is successful and the captive portal returns: Update OK! Device Rebooting now ...

Why does this happen? How is it possible that there are space problems if the sketch in question is normally loaded into the wemos through the arduino IDE? Can someone help me ?
```

tablatronix, 2020-12-28:
1mb limit? You need 2x memory

droidprova, 2020-12-29:
Thank you so much Tabla, so a card like this should solve my problems:

[](https://it.aliexpress.com/item/1005001792693766.html?spm=a2g0o.productlist.0.0.3a575491uhqcu4&algo_pvid=a81d9378-4db0-4aa6-9de7-c9aaebd3d0ee&algo_expid=a81d9378-4db0-4aa6-9de7-c9aaebd3d0ee-24&btsid=2100bdf016092305702604169e23fd&ws_ab_test= searchweb0_0, searchweb201602_, searchweb201603_)

true?

tablatronix, 2020-12-29:
What are you using now?

droidprova, 2020-12-29:
Wemos D1 mini Lite

tablatronix, 2020-12-29:
ohh yeah , get a 4mb chip, esp12, nodemcu, or get your sketch down below 512k for ota

droidprova, 2020-12-29:
my file .bin is 546 kb, if i don't cut it, what i do?

tablatronix, 2020-12-29:
you can also upgrade the flash ic on your board, i would just buy some nodemcu modules, they are really good

droidprova, 2020-12-29:
ok thanks a lot for your support. What I did not understand is: but regardless of the board you have, the maximum limit of a sketch that can be flashed via ota is always 512kb?

tablatronix, 2020-12-29:
No, that is your limit, you need to have 2x the amount of storage for OTA, since you are storing 2 copies at the same time, thats how ota works.

droidprova, 2020-12-30:
thanks a lot for your support.