tzapu/WiFiManager #1183: wifi_country_t does not name a type error

caster1, 2021-01-03:
hi im using code template as below and I'm getting this error
 `#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <PubSubClient.h>

WiFiClient espClient;
PubSubClient client(espClient);

void mqttCallback(char* topic, byte* payload, unsigned int length) {
       // message received
}

void mqttReconnect() {
       // reconnect code from PubSubClient example
}

void WiFiManager::setCountry(String IN){
  _wificountry = IN;
}

void setup() {
     WiFiManager wifiManager;
     wifiManager.setTimeout(180);

     if(!wifiManager.autoConnect("AutoConnectAP")) {
       Serial.println("failed to connect and hit timeout");
       delay(3000);
       ESP.reset();
       delay(5000);
     } 

     Serial.println("connected...yeey :)");

     client.setServer(mqtt_server, 1883);
     client.setCallback(mqttCallback);
}

void loop() {
     if (!client.connected()) {
       mqttReconnect();
     }
     client.loop();
     yield();
}`

my error is

 Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from C:\Users\sai\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:96:0,

                 from C:\Users\sai\Documents\Arduino\wifi_manager_mqtt_v2\wifi_manager_mqtt_v2.ino:8:

C:\Users\sai\Documents\Arduino\libraries\WiFiManager/strings_en.h:358:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_US{"US",1,11,WIFI_COUNTRY_POLICY_AUTO};

C:\Users\sai\Documents\Arduino\libraries\WiFiManager/strings_en.h:359:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,WIFI_COUNTRY_POLICY_AUTO};

C:\Users\sai\Documents\Arduino\libraries\WiFiManager/strings_en.h:360:7: error: 'wifi_country_t' does not name a type

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,WIFI_COUNTRY_POLICY_AUTO};

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

tablatronix, 2021-01-03:
What version are you using ?

caster1, 2021-01-06:
tried everything from 2.0.3 alphato 2.0.1 alpha

luisesn, 2021-02-02:
I have just had this problem and after losing about 1 hour it seems the problem is that you have to add

`#include <ESP8266WiFi.h>`

before the wifimanager include, the examples provided with library doesn't work for me without this.

tablatronix, 2021-02-02:
What is your board definition ?

tablatronix, 2021-02-02:
I mean it makes no sense, the library only works with esp8266 or esp32 and

```
#ifdef ESP8266

    extern "C" {
      #include "user_interface.h"
    }
    #include <ESP8266WiFi.h>
    #include <ESP8266WebServer.h>
```

shrug

luisesn, 2021-02-02:
I am using nodemcu 1.0 and tried the generic esp8266, both gave the same problem.

Esp version is 2.7.4
WifiManager is master .zip file just downloaded today

luisesn, 2021-02-02:
The board.txt definition for the nodemcu (and several others) lack this "-DESP8266" in their options.

I don't know how arduino works at that level but maybe its related

tablatronix, 2021-02-02:
I use nodemcu on platformio never had an issue
Do you have verbose logging on  arduino, can you copy the compile logs?

luisesn, 2021-02-02:
Yep, here it goes:

```
d:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware d:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Luis E\AppData\Local\Arduino15\packages -hardware C:\Users\Luis E\Documents\Arduino\hardware -tools d:\Program Files (x86)\Arduino\tools-builder -tools d:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Luis E\AppData\Local\Arduino15\packages -built-in-libraries d:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Luis E\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10812 -build-path C:\Users\LUISE~1\AppData\Local\Temp\arduino_build_687439 -warnings=none -build-cache C:\Users\LUISE~1\AppData\Local\Temp\arduino_cache_991171 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose D:\nextcloud\proyectos\pantallitaLed\sketch_feb02a\sketch_feb02a.ino
d:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware d:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Luis E\AppData\Local\Arduino15\packages -hardware C:\Users\Luis E\Documents\Arduino\hardware -tools d:\Program Files (x86)\Arduino\tools-builder -tools d:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Luis E\AppData\Local\Arduino15\packages -built-in-libraries d:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Luis E\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=legacy,ssl=all,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10812 -build-path C:\Users\LUISE~1\AppData\Local\Temp\arduino_build_687439 -warnings=none -build-cache C:\Users\LUISE~1\AppData\Local\Temp\arduino_cache_991171 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose D:\nextcloud\proyectos\pantallitaLed\sketch_feb02a\sketch_feb02a.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4
Using core 'esp8266' from platform in folder: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4
Detecting libraries used...
"C:\\Users\\Luis E\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Luis E\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Luis E\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Luis E\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\LUISE~1\\AppData\\Local\\Temp\\arduino_build_687439/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Luis E\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4 [... cut ...]
Usando librería DNSServer con versión 1.1.1 en la carpeta: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\DNSServer 
Utilizando biblioteca MAX7219LedMatrix-master en carpeta: C:\Users\Luis E\Documents\Arduino\libraries\MAX7219LedMatrix-master (legacy)
Usando librería SPI con versión 1.0 en la carpeta: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SPI 
exit status 1
Error compilando para la tarjeta NodeMCU 1.0 (ESP-12E Module).

```

tablatronix, 2021-02-02:
hmm

```
Alternatives for ESP8266WiFi.h: [ESP8266WiFi@1.0]
ResolveLibrary(ESP8266WiFi.h)
  -> candidates: [ESP8266WiFi@1.0]
```

What is this?

luisesn, 2021-02-02:
If I use the #define t uses the same libs (just the relevant part) and it compiles ok

```
Usando librería ESP8266WiFi con versión 1.0 en la carpeta: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi 
Usando librería Ticker con versión 1.0 en la carpeta: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\Ticker 
Usando librería WiFiManager-master con versión 2.0.4-beta en la carpeta: C:\Users\Luis E\Documents\Arduino\libraries\WiFiManager-master 
Usando librería ESP8266WebServer con versión 1.0 en la carpeta: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer 
Usando librería DNSServer con versión 1.1.1 en la carpeta: C:\Users\Luis E\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\DNSServer 
Utilizando biblioteca MAX7219LedMatrix-master en carpeta: C:\Users\Luis E\Documents\Arduino\libraries\MAX7219LedMatrix-master (legacy)
```

tablatronix, 2021-02-02:
Works for me

```
Users/alverson/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3 /Users/alverson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/sizes.py --elf /var/folders/6g/ljwft31n7r5g593rqjywgvmh0000gn/T/arduino_build_860858/OnDemandConfigPortal.ino.elf --path /Users/alverson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin
Executable segment sizes:
IROM   : 344676          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27884   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1348  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 3908  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25968 )         - zeroed variables      (global, static) in RAM/HEAP 
Using library WiFiManager at version 2.0.3-alpha in folder: /Users/alverson/Documents/Arduino/libraries/WiFiManager 
Using library ESP8266WiFi at version 1.0 in folder: /Users/alverson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: /Users/alverson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266WebServer 
Using library DNSServer at version 1.1.1 in folder: /Users/alverson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/DNSServer 
Using library ArduinoOTA at version 1.0 in folder: /Users/alverson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ArduinoOTA 
Using library ESP8266mDNS at version 1.2 in folder: /Users/alverson/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS 
/Users/alverson/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-size -A /var/folders/6g/ljwft31n7r5g593rqjywgvmh0000gn/T/arduino_build_860858/OnDemandConfigPortal.ino.elf
Sketch uses 377816 bytes (36%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31224 bytes (38%) of dynamic memory, leaving 50696 bytes for local variables. Maximum is 81920 bytes.
```

tablatronix, 2021-02-02:
What does your sketch look like? Are you using setCountry() ?

tablatronix, 2021-02-02:
I am not using cutting edge, I am using the lib manager one, maybe that has something to do with it. Ill have to install my extra boards and all that

luisesn, 2021-02-02:
Just some copy-paste boilerplate: 

```
#include <ESP8266WiFi.h>
#include <Ticker.h>

#include <strings_en.h>
#include <WiFiManager.h>

#include <LedMatrix.h>

Ticker ticker;

#define NUMBER_OF_DEVICES 1
#define CS_PIN 2
LedMatrix ledMatrix = LedMatrix(NUMBER_OF_DEVICES, CS_PIN);

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run once:
  Serial.begin(115200);
  
  //set led pin as output
  pinMode(LED_BUILTIN, OUTPUT);
  // start ticker with 0.5 because we start in AP mode and try to connect
  ticker.attach(0.6, tick);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wm;
  //reset settings - for testing
  // wm.resetSettings();

  //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wm.setAPCallback(configModeCallback);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wm.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.restart();
    delay(1000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  //keep LED on
  digitalWrite(LED_BUILTIN, LOW);
}

void loop() {
  // put your main code here, to run repeatedly:

}

void tick()
{
  //toggle state
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));     // set pin to the opposite state
}

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.2, tick);
}
```

tablatronix, 2021-02-05:
Very strange, I will try to install the github on arduino and see if I can get the same with that

caster1, 2021-02-08:
everyone sorry for the late reply the issue was with not including adrduino json library include that and you'll be good

tablatronix, 2021-02-08:
So strange how compilers do that sometime, throw unrelated errors

caster1, 2021-02-15:
so every one its back again

my code
`#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <PubSubClient.h>
#include <ArduinoJson.h>

WiFiClient espClient;
PubSubClient client(espClient);
char mqtt_server[40];
void mqttCallback(char* topic, byte* payload, unsigned int length) {
       // message received
}

void mqttReconnect() {
       // reconnect code from PubSubClient example
}

void setup() {
  Serial.begin(9600);
     WiFiManager wifiManager;
     wifiManager.setTimeout(180);

     if(!wifiManager.autoConnect("AutoConnectAP")) {
       Serial.println("failed to connect and hit timeout");
       delay(3000);
       ESP.reset();
       delay(5000);
     } 
     // id/name, placeholder/prompt, default, length
WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
wifiManager.addParameter(&custom_mqtt_server);

const char* mqtt_server = custom_mqtt_server.getValue();

     Serial.println("connected...yeey :)");

     client.setServer(mqtt_server, 1883);
     client.setCallback(mqttCallback);
}

void loop() {
     if (!client.connected()) {
       mqttReconnect();
     }
     client.loop();
     yield();
}`

my error : 

> n file included from C:\Users\sai\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:96:0,
                 from C:\Users\sai\Documents\Arduino\mqtt_wifi_manager\mqtt_wifi_manager.ino:2:
C:\Users\sai\Documents\Arduino\libraries\WiFiManager/strings_en.h:358:7: error: 'wifi_country_t' does not name a type
 const wifi_country_t WM_COUNTRY_US{"US",1,11,WIFI_COUNTRY_POLICY_AUTO};
       ^
C:\Users\sai\Documents\Arduino\libraries\WiFiManager/strings_en.h:359:7: error: 'wifi_country_t' does not name a type
 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,WIFI_COUNTRY_POLICY_AUTO};
       ^
C:\Users\sai\Documents\Arduino\libraries\WiFiManager/strings_en.h:360:7: error: 'wifi_country_t' does not name a type
 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,WIFI_COUNTRY_POLICY_AUTO};
       ^
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

tablatronix, 2021-02-15:
I would almost suspect you have duplicate libraries or something

caster1, 2021-02-16:
will check that once I'm at my desktop, can you please compile my code?