tzapu/WiFiManager #1184: Using both WebServer.h and esp_http_server.h at the same time???

jameszah, 2021-01-03:
Hi, quick question about using this on ESP32 with Arduino IDE and with an http server already running in my code:

Has anyone tried to use this library which includes the WebServer.h server object, and at the same time use direct server code with esp_http_server.h software.  The two packages fight over who controls symbols (like HTTP_GET) during the compile, and after solving that, then the WebServer.h object and my own object created with esp_http_server.h fight over who controls the events -- and only one can get the events.

I have a work-around to do the WiFiManager stuff, and then close the object, and then start a new server with all my own http server stuff.
I think the solution would be to re-do this library without the WebServer.h object, and directly grab the http events, and include WiFiManger events in the mix of other http events that my program already handles.

So just looking for a little advice if anyone has faced this issue?

I was using the esp_http_server.h calls, such as:

```
httpd_handle_t camera_httpd = NULL;
httpd_start(&camera_httpd, &config), 
httpd_register_uri_handler(camera_httpd, &index_uri), 
httpd_resp_send
```
While WebServer.h calls such as:
```
 WebServer server(80);
 server.on("/wifi", []{ handleConfig(); });
  this->_server->sendHeader("Content-Length", String(message.length()));
  this->_server->send(404, "text/plain", message);

```

tablatronix, 2021-01-04:
Not sure what that is there are a few options

WM does not allow passing in an httpd yet, but ours is a shared pointer so you can use it in your own code.

You can run it on alternate ports, you cannot close and restart same port ( there is a bug in esp with port binding it seems )

I have an experimental async webserver branch also

jameszah, 2021-01-04:
Very interesting -- I think that "port binding" bug was my problem.

So for the record here is my solution:
1. include esp_http_server.h
2. define _HTTP_Method_H_ to prevent the WebServer.h system from trying to define HTTP_Method, and then define HTTP_Method using different names for HTTP_GET, etc.
3. Then we can include WiFIManager.h which will bring in WebServer.h and there will be no compile issues.
4. And to get around the "port binding" bug, which prevents the esp_http_server.h code from binding, just do a reboot of the esp32, WiFIManager will establish the ssid connection without starting its server, and my code will start the server successfully

I haven't tried it away from my wifi yet.  I assume it will fail to connect, then start the server to get the ssid, which will also fail, and then all the non-wifi aspects of my program will start running normally.

https://github.com/jameszah/ESP32-CAM-Video-Recorder-junior

```
#include "esp_http_server.h"

// Workaround for the WebServer.h vs esp_http_server.h problem 

#define _HTTP_Method_H_ 

typedef enum {
  jHTTP_GET     = 0b00000001,
  jHTTP_POST    = 0b00000010,
  jHTTP_DELETE  = 0b00000100,
  jHTTP_PUT     = 0b00001000,
  jHTTP_PATCH   = 0b00010000,
  jHTTP_HEAD    = 0b00100000,
  jHTTP_OPTIONS = 0b01000000,
  jHTTP_ANY     = 0b01111111,
} HTTPMethod;

#include <WiFi.h>
#include <WiFiManager.h> 

// done in the setup()

    WiFiManager wm;
    bool res;
    //wm.resetSettings();  // for debugging

    // wm.setConnectTimeout(20); // how long to try to connect for before continuing
    // wm.setConfigPortalTimeout(30); // auto close configportal after n seconds
    // res = wm.autoConnect(); // auto generated AP name from chipid
    
    res = wm.autoConnect(devname); // use the devname defined above, with no password

    //res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if (res) {
      Serial.println("Succesful Connection using WiFiManager");
    } else {
      InternetFailed = true;
      Serial.println("Internet failed using WiFiManager - not starting Web services");
    }
```

closed 2021-01-05 by jameszah