tzapu/WiFiManager #1187: Can start the AP

Mesihas, 2021-01-07:
### Basic Infos
basic example does not create the AP

#### Hardware
WiFimanager Branch/Release:  Development

Esp32: 

### Description

basic example does not create the AP

using example skrtch (simple) for the first time (nothing stored on FS, no wifi data at all:
this is what I get on serial monitor
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
W (253) wifi:Haven't to connect to a suitable AP now!
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.7
connected...yeey :)

```
`#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    // WiFi.mode(WiFi_STA); // it is a good practice to make sure your code sets wifi mode how you want it.

    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    //reset settings - wipe credentials for testing
    //wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","12345678"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:
    
}`

```

tablatronix, 2021-01-08:
That is not development the logs do not have log levels shown

Mesihas, 2021-01-08:
How can I get the dev logs?

tablatronix, 2021-01-08:
I mean you are NOT actually using development

Mesihas, 2021-01-09:
ohh ok, I think that I installed the lib from Arduino IDE, that's what you mean?

Do I need to install the lib from the repository and then do all my tests?

rhalaly, 2021-03-04:
@tablatronix same issue in 0.16.0 and 2.0.3-alpha (from Arduino IDE Library Manager)

tablatronix, 2021-03-04:
If you can , there has been some updates. If not I should be posting a beta soon to arduino

Luke-KH, 2021-04-11:
I have tried below code from ESP32 Arduino boards manager v1.0.6 library. I found that similar problem that password protected AP will disappear to become an OPEN softeAP.  It will happen if softAP restarted, even reburn program (no change flash setting area). 

It is working fine if using ESP32 Arduino boards manager v1.0.4 library.
Any guys know what is it going on?

```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WiFiAP.h>

#define LED_BUILTIN 2   // Set the GPIO pin where you connected your test LED or comment this line out if your dev board has a built-in LED

// Set these to your desired credentials.
const char *ssid = "yourAP";
const char *password = "yourPassword";
bool wifi_onoff_toggle = true;
WiFiServer server(80);

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");

  // You can remove the password parameter if you want the AP to be open.
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  server.begin();

  Serial.println("Server started");
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn ON the LED.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn OFF the LED.<br>");
            client.print("Click <a href=\"/T\">here</a> to turn OFF the WiFi in 10 second & resume .<br>");
            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LED_BUILTIN, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LED_BUILTIN, LOW);                // GET /L turns the LED off
        }
        if (currentLine.endsWith("GET /T")) {
            WiFi.softAPdisconnect(false);
            delay(10000);
            if(WiFi.enableAP(true)) Serial.println("WiFi.enableAP(true) == TRUE");
            else Serial.println("WiFi.enableAP(true) == false");
        }
      
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

tablatronix, 2021-04-11:
interesting, same ssid?