tzapu/WiFiManager #1191: How to connect and disconnect to wifi cyclically

jordan1982, 2021-01-17:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```
Hi,
I would need to connect and disconnect cyclically. I give an example:
The user enters the data via the web interface (ssid, password, mqtt server, mqtt user, mqtt password etc), the system waits 5 minutes and then connects to mqtt sends the message and then disconnects from mqtt and wifi. After another 5 minutes it reconnects to the previous wifi and repeats the loop.
How can I connect and disconnect from wifi through your library?

svdrummer, 2021-02-04:
Google ESP.Sleep() that's how we do it.

tablatronix, 2021-02-04:
Same way you would with any esp code.. nothing special here

closed 2021-02-04 by tablatronix

jordan1982, 2021-03-13:
With esp.sleep  the data entered in the configuration page are lost?
Thanks for the help you can give me

tablatronix, 2021-03-13:
data such as wifi ? or custom params?

jordan1982, 2021-03-14:
wifi and custom params

tablatronix, 2021-03-14:
well custom params have to be saved somewhere, esp wont remember then sleep is really a reboot