tzapu/WiFiManager #1192: Can´t configure SSID and password for AP configuration page

donsergiobejarano, 2021-01-18:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  2.7.4

Esp8266

Hardware:			ESP-12e

Core Version:      	2.7.4 also tested 2.4.0 

### Description

I configure AP name and password to access to config portal and always create an ap called "ESP-XXXXXX" without password where XXXXXX is a random number depending of ESP module.
Default configuration, with fresh install (all updated) and deleted libraries, folders and cache folders.

Arduino IDE 1.8.13

### Settings in IDE

Board: Generic ESP8266 Module
builtin led: 2
upload speed 115200
cpu freq 80 mhz
crystal freq 26mhz
flashsize 1mb (fs 64k ota 470) (tried with anothers options)
flash mode (tried dout and dio)
flash frequency 40mhz
reset method dtr (aka nodemcu)
debug port disabled
debug level nothing
iwip variant v2 lower memory
vtables flash
exceptions: legacy (new can return nullptr)
erase flash: only sketch (tried with the other two options)
expressif fw nonos-sdk 2.2.1+100 (190703)
ssl support all ssl ciphers
port (the com port)

Module:  			Generic ESP8266 Module

Additional libraries: 
UniversalTelegramBot
The other libraries are default on Arduino IDE install

### Sketch (minimal test sketch)

#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

void setup() {
  Serial.begin(115200);
  WiFiManager wifiManager;
  wifiManager.autoConnect("Access Configuration Page", "12345678");
  Serial.println("You are connected");
}

void loop() {
  }

charitha95, 2021-01-18:
I ran your sketch and it worked as expected for me. The only difference in my environment is the WiFimanager Branch. I have a clone of development.

tablatronix, 2021-01-18:
`*WM: [0] [ERROR] There was a problem starting the AP`

This is coming from ESP lib, try upgrading or downgrading, or full erase your device

donsergiobejarano, 2021-01-19:
Hi, it´s something wrong with the spaces.... I use underscore and works fine... 

wifiManager.autoConnect("Access_Configuration_Page", "12345678");
or
wifiManager.autoConnect("Access-Configuration-Page", "12345678");

Thanks!!

tablatronix, 2021-01-19:
interesting

tablatronix, 2021-02-02:
- [ ] check ssid with spaces