tzapu/WiFiManager #1194: Custom fields are not displayed

openedhardware, 2021-01-25:
### Basic Infos
I have added a couple of custom fields, but those are not displayed on the captive portal.

#### Hardware
WiFimanager Branch/Release:  2.0.4-beta

Esp8266/Esp32: ESP8266(NodeMCU)

Hardware:			ESP-12e

Core Version:      	2.4.0

### Settings in IDE

Module:  			NodeMcu

### Sketch

```cpp

  // wm.setClass("invert");
  WiFiManagerParameter header_text("<p>Please input token:</p>");
  wm.addParameter(&header_text);
  WiFiManagerParameter tb_token_field("token", "Token", conf.tb_token, 21);
  wm.addParameter(&tb_token_field);

```

### Screenshot

[image]

tablatronix, 2021-01-25:
not sure works for me, do you have a custom menu set? Are they in setup ?

openedhardware, 2021-01-25:
No, didn't use the custom menu.

And these codes are in `setup()` function.

Thanks!

tablatronix, 2021-01-25:
Have you tried an example ?

EgHubs, 2021-01-29:
Sometimes it happens for me and I think it's related to if these parameters are already saved to the device or not, you can try to clear all the parameters and see what happens.
I just use a link to add these parameters.
`                'http://192.168.4.1/wifisaves=$SSID&p=$PASS&ip=$ipv4&gw=$defaultGateway&sn=$defaultSubnetMask&token=$token'
`

robertopuhan777, 2022-07-22:
Try this code. I was having the same problem so I took the liberty to tweek a bit the random nerd tutorial on this subject using wifi manager to assign pin number to relay. I added username and password protection for website when settings on http://192.168.4.1 are made you simply add encoded username and password created on website https://www.base64encode.net

//THE CODE//
  //************************************************ For NodeMcu 1.0 ESP 8266 *************************************************//
  //****************** Thanks to Rui Sanstos from randomnerd and many many talented people with great ideas *******************//
  //********************* I just tweeked a bit his code an added username password protection for website *********************//
  //******************* The button I added to program is there for easier usage. To avoid double uploading. *******************//
  //********************* When saving settings in wifi manager you must add user:pass encoded text. ***************************//
  //**************************** The easiest way is to use this site https://www.base64encode.net *****************************//
  //********* I recommend creating your username and password before you start saving settings in Wifi Manager. ***************//
  //*** Just send yourself an email with encoded username password to be able to copy paste when connected to Wifi manager ****//
  //************** Encoded text tends to be quite long so I strongly recommend not to input that text manually =) *************//
  //************************************** One can always delete the comments or credentials =) *******************************//
  //***************** When connected for the first time Use QR Code or simply go to website http://192.168.4.1 =) *************//
  //*****************************************************************************************//
  // LIBRARIES & RESET BUTTON defined as TRIGGER_PIN on pin D0
  //*****************************************************************************************//
#include <FS.h>
#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>
#include "strings_en.h"
#define TRIGGER_PIN D0
  //*****************************************************************************************//
  // Set web server port number to port 80 or any other free port or better not already assigned or designated port for accessing device from anywhere in the world. //
  // Just forward that port in your router =) //
  WiFiServer server(80);
  //*****************************************************************************************//
  // Store HTTP request
  String header;
  //*****************************************************************************************//
  // Set relay states and designate relay pins - if Relays are "reversed" - check BUILD WEBSITE at the bottom of code & change to your needs =)
  //*****************************************************************************************//
  String outputState = "off";
  String output5State = "off";
  String output4State = "off";
  const int output5 = D5;
  const int output4 = D4;
  //     PINOUT NODE MCU  & GPIO*s pin names to avoid assigning designated pins:)
  //
  //
  //       A0  -   |-------|   -  D0   16
  //       G   -   |       |   -  D1    5
  //       VV  -   |       |   -  D2    4
  //  10   S3  -   |       |   -  D3    0
  //   9   S2  -   |       |   -  D4    2
  //       S1  -   |-------|   -  3V
  //       SC  -               -  G
  //       S0  -               -  D5   14
  //       SK  -               -  D6   12
  //       G   -               -  D7   13
  //       3V  -               -  D8   15
  //       EN  -               -  RX    3
  //       RDT -               -  TX    1
  //       G   -               -  G
  //       VIN -               -  3V
  //
  
  
// Try not to asign to already designated pinout number when running WifiManager =)... =) ... =)
// custom output variable output pin
   char output[2] = "5";
//************************************************************************************************************//
//* select your username:password on website https://www.base64encode.net encode it and paste down as output1*//
//******************************************** default user:pass *********************************************//
//********** later you will use this username and password to access you webpage to controll relays **********//
//*************** Having username and password for website that controlls relays is a must =) ****************//
char output1[] = "dXNlcjpwYXNz";
[... 235 lines cut ...]
            }
            else {
              client.println("HTTP/1.1 401 Unauthorized");                 // website checks for username password and gives response if unauthorized
              client.println("WWW-Authenticate: Basic realm=\"Secure\"");  // all body content is in that loop so all content off website is secured with username password
              client.println("Content-Type: text/html");
              client.println();
              client.println("<html>Authentication not correct!</html>");
              break;
            }
          }

          else {
            currentLine = "";
          }
        } else if (c != '\r') {
          currentLine += c;
        }
      }
    }
    header = "";
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}

robertopuhan777, 2022-07-22:
Here is the link to file located on google drive
https://drive.google.com/file/d/1Qqbypqu2c3bnD9tMeZplsEWjyC3EyJAH/view?usp=sharing