tzapu/WiFiManager #1198: DNS starts but my linux and Android never show the captive portal unless I manually type in the IPV4 address.

dennisMe2, 2021-01-26:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development / master 

Esp8266

Hardware:			ESP-12

Core Version:      	master snapshot from 25-01-2020

DNS starts but my linux and Android never show the captive portal unless I manually type in the IPV4 address. Linux shows the DNS registered on the interface when the portal is active (but not in /etc/resolv.conf). An nslookup with the ESP as specific server address returns the soft-AP address for any random name I try. 
Nonetheless, no requests for random (non cached) site names end up at the ESP ip address. Debug statements I inserted do not show any requests entering the unit unless I explicitly use the IP address.

Arduino IDE 1.8.13

Module:  			NodeMcu 0.9 ESP12

Additional libraries: (in order)
Wire.h 
ESP8266WiFi.h 
Adafruit_Sensor.h 
DNSServer.h 
ESP8266WebServer.h 
WiFiManager.h 
ESP_EEPROM.h

#include <Wire.h>
#include <ESP8266WiFi.h>
#include <Adafruit_Sensor.h>

// New Libraries needed for captive portal:
#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
#include <ESP_EEPROM.h>

#define TRIGGER_PIN D9
const int buttonPin = TRIGGER_PIN;

char temperatureString[6];

WiFiClient client;

//measure VCC
ADC_MODE(ADC_VCC);

//WiFiManagerParameter::WiFiManagerParameter(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom)
WiFiManagerParameter p_server("Server", "Server", "api.thingspeak.com", 50);
WiFiManagerParameter p_port("Port", "Port", "80", 5);
WiFiManagerParameter p_apiKey("APIKey", "API Key", "FHH61AF956PDW0ME", 50);

struct settings_t {
  char ssid[51];
  char password[26];
  uint8_t bssid[6];
  uint8_t channel;
  uint8_t ap_mac[16];
  char server[51];
  uint16_t port;
  char apiKey[51];
} settings;

bool doConfigPortal() {
  if ( digitalRead(TRIGGER_PIN) == LOW ) {
    Serial.println("BUTTON PRESSED");
    delay(100);
    if ( digitalRead(TRIGGER_PIN) == LOW ) {
      WiFiManager wifiManager;
      char hex[6];
      Serial.println("BUTTON DEBOUNCED");
      
      p_server.setValue(settings.server, 50);
      wifiManager.addParameter(&p_server);
      
      sprintf(hex, "%x", settings.port);
      hex[5] = 0x0;
      p_port.setValue(hex, 5);
      wifiManager.addParameter(&p_port);
      
      p_apiKey.setValue(settings.apiKey, 50);
      wifiManager.addParameter(&p_apiKey);
      
      wifiManager.setSaveConfigCallback(saveConfigCallback);
      wifiManager.setConfigPortalTimeout(300);
      wifiManager.setCaptivePortalEnable(true);
      if (!wifiManager.startConfigPortal("Config ESP Portal")) {
        Serial.println("failed to create portal for some reason");
        return false;
      } else {
        strncpy(settings.password, WiFi.psk().c_str(), 25);
        settings.password[25] = 0x0;
        saveSettings();
        Serial.print("ssid: ");
        Serial.println(settings.ssid);
        Serial.print("pwd: ");
        Serial.println(settings.password);
        Serial.println("created Portal...:)");
        return true;
      }
    }
  }
  return false;
}

bool tryWifiConnect() {
  int retries = 0;
  int wifiStatus = WiFi.status();
  while ( wifiStatus != WL_CONNECTED ) {
    retries++;
    if (doConfigPortal()) {
      Serial.println("DoConfigPortal returned true");
      return true;
    }
    if ( retries == 100 ) {
      Serial.println(" Quick connect is not working, reset WiFi and try regular connection");
      WiFi.disconnect();
      delay( 10 );
      WiFi.forceSleepBegin();
      delay( 10 );
      WiFi.forceSleepWake();
      delay( 10 );
      WiFi.begin( settings.ssid, settings.password );
    }
    if ( retries == 600 ) {
      Serial.println("No luck with normal Slow connection either!");
      return false;
    }
    delay( 50 );
    wifiStatus = WiFi.status();
  }
  return true;
}

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Callback was called");
  strncpy(settings.server, p_server.getValue(), 50);
  settings.port = atoi(p_port.getValue());
  strncpy(settings.apiKey, p_apiKey.getValue(), 50);
  saveSettings();
}

void saveSettings() {
  memcpy( settings.ssid, WiFi.SSID().c_str(), 50); // Copy 50 bytes of BSSID (AP's MAC address)
  memcpy( settings.bssid, WiFi.BSSID(), 6); // Copy 6 bytes of BSSID (AP's MAC address)
  settings.channel = WiFi.channel();
  EEPROM.put(0, settings);
  boolean ok2 = EEPROM.commit();
  Serial.println((ok2) ? "Commit OK" : "Commit failed");
}

/**************************
     S E T U P
 **************************/
// only runs once on boot
void setup() {

  // start with wifi off to save power
  WiFi.mode( WIFI_OFF );
  WiFi.forceSleepBegin();
  delay( 1 );

  EEPROM.begin(sizeof(settings));
  EEPROM.get(0, settings);

  // Initializing serial port for debugging purposes
  Serial.begin(115200);
  delay(10);
  Serial.println("ESP8266 just woke up!");

  char hex[25];
  Serial.print("ssid: ");
  Serial.println(settings.ssid);
  Serial.print("pwd: ");
  Serial.println(settings.password);
  Serial.print("channel: ");
  Serial.println(settings.channel);
  Serial.print("bssid: ");
  sprintf(hex, "%x", settings.bssid);
  Serial.print(hex);
  Serial.println("ap_mac: ");
  sprintf(hex, "%x", settings.ap_mac);
  Serial.print("Server: ");
  Serial.println(settings.server);
  Serial.print("Port: %n");
  Serial.println( settings.port);
  Serial.print("Api Write Key: ");
  Serial.println(settings.apiKey);

  pinMode(buttonPin, INPUT_PULLUP);
  if (digitalRead(buttonPin) == LOW) {
    Serial.println("Button pressed!");
  }

  //added for powersaving!
  WiFi.forceSleepWake();
  delay( 1 );

  //quick connect attempt with cached settings
  WiFi.begin( settings.ssid, settings.password, settings.channel, settings.bssid, true );
  if (!tryWifiConnect()) {
    Serial.println("Giving up connecting, its no use!");
    WiFi.disconnect( true );
    delay( 1 );
    WiFi.mode( WIFI_OFF );
    ESP.deepSleep( 600e6, WAKE_RF_DISABLED ); //300e6 = 5 minutes
    delay(1);
  } else {
    Serial.println("Quick Connect Success");
    saveSettings();
  }

}

/**************************
   L O O P
**************************/
void loop() {

  Serial.print("Local IP = ");
  Serial.println(WiFi.localIP());

  //    dtostrf(ESP.getVcc(), 5, 1, temperatureString);
  //    Serial.println(temperatureString);

  //    if (client.connect(server,80)) { // "184.106.153.149" or api.thingspeak.com
  //        String postStr = apiKey;
  //        postStr +="&field1=";
  //        postStr += String(temperatureString);
  //
  //        postStr += "\r\n\r\n";
  //
  //        client.print("POST /update HTTP/1.1\n");
  //        client.print("Host: api.thingspeak.com\n");
  //        client.print("Connection: close\n");
  //        client.print("X-THINGSPEAKAPIKEY: "+apiKey+"\n");
  //        client.print("Content-Type: application/x-www-form-urlencoded\n");
  //        client.print("Content-Length: ");
  //        client.print(postStr.length());
  //        client.print("\n\n");
  //        client.print(postStr);
  //        Serial.println("posted " +postStr + " to thingspeak!");
  //    }
  //    delay(10);
  //    client.stop();
  //    delay(1);

  //every 5 Min
  //delay(5000);
  // stop wifi so it does not spike on wake
  WiFi.disconnect( true );
  delay( 1 );

  Serial.println("ESP8266 Going to sleep!");
  delay(1000);

  ESP.deepSleep(10e6, WAKE_RF_DISABLED ); //20 microAmperes FTW!
  delay(1);
}

tablatronix, 2021-01-26:
Does the arduino captive portal example work ?

dennisMe2, 2021-01-27:
I tested it using OnDemandConfigPortal.ino and that works brilliantly. Looks like I'll have to try and refactor my code step by step to identify the culprit.
Thanks for the tip and,
Please close this non-issue!

closed 2021-01-27 by dennisMe2

dennisMe2, 2021-01-27:
FYI:
WiFiManager instance must be global. I had moved it to where I needed it but the instance really has to be global in order to work.

tablatronix, 2021-01-27:
That is true for params, not sure why it would break your captive portal